avrdudes/avrdude #458: [bug #51497] avrdude not writing high byte using linuxgpio when data byte count is less than 0x10

avrdude-admin, 2021-12-11:
Mike Brandon <None>
Tue 18 Jul 2017 03:07:29 PM UTC
Programmer hardware: linuxgpio
Device type: m2560, m12834

Using linuxgpio and tested on m2560 and m1284 chips:

When a line in the hex file has less than 0x10 data bytes, avrdude does not write the high byte at the specified memory location as 0xFF. This in turn results in a verification error.

Attached is a sample hex file that was used to produce this result as well as logic captures when using avrdude and Atmel Studio 7.

Avrdude results in a verification error while Atmel Studio does not. It appears that the difference is that Atmel Studio will write 0xFF to the high byte while avrdude moves on to the next memory location and does not write a high byte.

[file #41233: blink-mod-bitbang-read.PNG](https://savannah.nongnu.org/bugs/download.php?file_id=41233)
[file #41232: blink-2560-mod.hex](https://savannah.nongnu.org/bugs/download.php?file_id=41232)
[file #41231: blink-2560-mod-logic-avrisp-write.PNG](https://savannah.nongnu.org/bugs/download.php?file_id=41231)
[file #41230: blink-2560-mod-logic-avrdude-gpio-write.PNG](https://savannah.nongnu.org/bugs/download.php?file_id=41230)

This issue was migrated from https://savannah.nongnu.org/bugs/?51497

avrdude-admin, 2021-12-11:
Mike <smbrandonjr>
Wed 01 Nov 2017 01:30:37 PM UTC

Is this a valid issue that I have described or, due to my general inexperience, am I trying to do something that should not work?

avrdude-admin, 2021-12-11:
Joerg Wunsch <joerg_wunsch>
Wed 01 Nov 2017 02:06:54 PM UTC

Quite possible this might be valid.

My guess is that linuxgpio simply isn't used very widely by many people.

mcuee, 2022-06-20:
I will give this a try once I set up my Raspberry Pi 3B+ to test linux-gpio and linux-spi.

mcuee, 2022-07-10:
This is probably the same as #455 and there is a patch in #455.

I will mark this as a duplicate of #455.

closed 2022-07-10 by mcuee