avrdudes/avrdude #459: [bug #51614] BusPirate: Paged Read command returned zero. With m1284p flash.

avrdude-admin, 2021-12-11:
Aaron <avrdude@fgthou.se>
Sat 29 Jul 2017 01:35:58 PM UTC
Programmer hardware: BusPirate

Verbose output: [https://gist.github.com/mcd1992/c9bb57071a8db5bf77d07c8a2e60c8df](https://savannah.nongnu.orghttps://gist.github.com/mcd1992/c9bb57071a8db5bf77d07c8a2e60c8df)

Let me know if anymore information is needed, thanks.

This issue was migrated from https://savannah.nongnu.org/bugs/?51614

avrdude-admin, 2021-12-11:
Kyoungkyu Park <yous87uu2>
Sat 04 Nov 2017 02:47:46 PM UTC

Same here. ver 6.3 at Windows, Bus Pirate 3.6a, m128.

My verbose log is here: [https://gist.github.com/choryuidentify/6b14f5d940f0c959ada35715967f354f](https://savannah.nongnu.orghttps://gist.github.com/choryuidentify/6b14f5d940f0c959ada35715967f354f)

Tested on Bus Pirate firmware version 7.0 and v6.3-r2151 both.

mcuee, 2022-06-19:
The verbose output for the first report has gone.
https://gist.github.com/mcd1992/c9bb57071a8db5bf77d07c8a2e60c8df (no longer exist).

The verbos log for the second report is still there.
https://gist.github.com/choryuidentify/6b14f5d940f0c959ada35715967f354f

```
C:\Users\Choryu\Desktop\avrdude-6.3-mingw32>avrdude.exe -c buspirate -P COM10 -p m128 -U flash:r:flash.bin:r -v > buspirate_log.log

avrdude.exe: Version 6.3, compiled on Feb 17 2016 at 09:25:53
             Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
             Copyright (c) 2007-2014 Joerg Wunsch

             System wide configuration file is "C:\Users\Choryu\Desktop\avrdude-6.3-mingw32\avrdude.conf"

             Using Port                    : COM10
             Using Programmer              : buspirate
             AVR Part                      : ATmega128
             Chip Erase delay              : 9000 us
             PAGEL                         : PD7
             BS2                           : PA0
             RESET disposition             : dedicated
             RETRY pulse                   : SCK
             serial program mode           : yes
             parallel program mode         : yes
             Timeout                       : 200
             StabDelay                     : 100
             CmdexeDelay                   : 25
             SyncLoops                     : 32
             ByteDelay                     : 0
             PollIndex                     : 3
             PollValue                     : 0x53
             Memory Detail                 :

                                      Block Poll               Page                       Polled
               Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
               ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
               eeprom         4    12    64    0 no       4096    8      0  9000  9000 0xff 0xff
               flash         33     6   128    0 yes    131072  256    512  4500  4500 0xff 0xff
               lfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
               hfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
               efuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
               lock           0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
               calibration    0     0     0    0 no          4    0      0     0     0 0x00 0x00
               signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

             Programmer Type : BusPirate
             Description     : The Bus Pirate

Attempting to initiate BusPirate binary mode...
BusPirate binmode version: 1
BusPirate SPI version: 1
avrdude.exe: Paged flash write enabled.
AVR Extended Commands version 1
avrdude.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.10s

avrdude.exe: Device signature = 0x1e9702 (probably m128)
avrdude.exe: safemode: hfuse reads as DF
avrdude.exe: safemode: efuse reads as FF
avrdude.exe: reading flash memory:

Reading |                                                    | 0% 0.00sBusPirate: buspirate_paged_load(..,flash,256,0,256)
BusPirate: buspirate_paged_load(..,flash,256,256,256)
BusPirate: buspirate_paged_load(..,flash,256,512,256)
BusPirate: buspirate_paged_load(..,flash,256,768,256)
BusPirate: buspirate_paged_load(..,flash,256,1024,256)
BusPirate: buspirate_paged_load(..,flash,256,1280,256)
Reading | #                                                  | 1% 1.82sBusPirate: buspirate_paged_load(..,flash,256,1536,256)
[... 85 lines cut ...]
BusPirate: buspirate_paged_load(..,flash,256,23552,256)

/* Loop... */

Reading | ################################################   | 96% 151.66sBusPirate: buspirate_paged_load(..,flash,256,125952,256)
BusPirate: buspirate_paged_load(..,flash,256,126208,256)
BusPirate: buspirate_paged_load(..,flash,256,126464,256)
BusPirate: buspirate_paged_load(..,flash,256,126720,256)
BusPirate: buspirate_paged_load(..,flash,256,126976,256)
Reading | #################################################  | 97% 153.16sBusPirate: buspirate_paged_load(..,flash,256,127232,256)
BusPirate: buspirate_paged_load(..,flash,256,127488,256)
BusPirate: buspirate_paged_load(..,flash,256,127744,256)
BusPirate: buspirate_paged_load(..,flash,256,128000,256)
BusPirate: buspirate_paged_load(..,flash,256,128256,256)
Reading | #################################################  | 98% 154.70sBusPirate: buspirate_paged_load(..,flash,256,128512,256)
BusPirate: buspirate_paged_load(..,flash,256,128768,256)
BusPirate: buspirate_paged_load(..,flash,256,129024,256)
BusPirate: buspirate_paged_load(..,flash,256,129280,256)
BusPirate: buspirate_paged_load(..,flash,256,129536,256)
Reading | ################################################## | 99% 156.22sBusPirate: buspirate_paged_load(..,flash,256,129792,256)
BusPirate: buspirate_paged_load(..,flash,256,130048,256)
BusPirate: buspirate_paged_load(..,flash,256,130304,256)
BusPirate: buspirate_paged_load(..,flash,256,130560,256)
BusPirate: Paged Read command returned zero.
```

mcuee, 2022-06-19:
From a quick glance, the code does not seem to deal with load extended address byte command which is needed by m1284p and it should have the same problem for ATmega2560 and similar parts.

Someone more familiar with the codes may need to confirm.

Ref: https://github.com/avrdudes/avrdude/blob/main/src/buspirate.c

MCUdude, 2022-06-19:
I have a BusPirate I can test with later tonight. Does writing extended addresses work?

mcuee, 2022-06-19:
> I have a BusPirate I can test with later tonight. Does writing extended addresses work?

Not so sure. It might work, just like USBtinyISP or the FT232R. Please test both read and write.

MCUdude, 2022-06-19:
For some reason, it doesn't allow me to write the chaucker256k hex file:

```
$ ./avrdude -patmega2560 -cbuspirate -P /dev/cu.usbmodem000000011 -Uflash:w:chaucer256k.hex:i

Attempting to initiate BusPirate binary mode...
avrdude: Paged flash write enabled.
avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.03s

avrdude: Device signature = 0x1e9801 (probably m2560)
avrdude: NOTE: "flash" memory has been specified, an erase cycle will be performed
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: reading input file "chaucer256k.hex"
avrdude: writing flash (176758 bytes):

Writing |                                                    | 0% 0.00savrdude: ser_send(): write error: Resource temporarily unavailable
BusPirate: Fatal error: Write Then Read did not succeed.
^C
```

I have a BusPirate v4.0, but I don't know if it runs the latest FW or not. IMO the BP documentation is a big mess, and Dangerous Prototypes _still_ states that the v4.0 is not "proven" like the v3.6 is, even though it has been out for over 6 years. The BP firmware hasn't seen a release since 2016 either...

MCUdude, 2022-06-19:
I was able to upload a chaucer32k hex file to an ATmega32, but a chaucer64k failed to upload on an ATmega644P:

```
$ ./avrdude -patmega644p -cbuspirate -P /dev/cu.usbmodem000000011 -Uflash:w:/var/folders/6l/ypg6qbw172v1s4vtt6g990tw0000gn/T/arduino_build_54198/Blink.ino.hex:i

Attempting to initiate BusPirate binary mode...
avrdude: Paged flash write enabled.
avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.03s

avrdude: Device signature = 0x1e960a (probably m644p)
avrdude: NOTE: "flash" memory has been specified, an erase cycle will be performed
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: reading input file "/var/folders/6l/ypg6qbw172v1s4vtt6g990tw0000gn/T/arduino_build_54198/Blink.ino.hex"
avrdude: writing flash (62386 bytes):

Writing | #                                                  | 1% 0.96savrdude: ser_send(): write error: Resource temporarily unavailable
BusPirate: Fatal error: Write Then Read did not succeed.
Writing | #                                                  | 2% 6.28savrdude: ser_recv(): read error: Device not configured
^C
```

mcuee, 2022-06-19:
> I have a BusPirate v4.0, but I don't know if it runs the latest FW or not. IMO the BP documentation is a big mess, and Dangerous Prototypes _still_ states that the v4.0 is not "proven" like the v3.6 [...]

@MCUdude 
I do not have the BusPirate but I think the latest FW is here.
https://github.com/BusPirate/Bus_Pirate

Indeed they also do not have any official release after 14 Aug 2016. But there are quite some fixes in the git master. You may want to give the Community 7.0 release a try, or even build the latest git master FW.
https://github.com/BusPirate/Bus_Pirate/releases/tag/v7.0

mcuee, 2022-07-28:
Ref: https://www.rapid7.com/blog/post/2019/04/16/extracting-firmware-from-microcontrollers-onboard-flash-memory-part-1-atmel-microcontrollers/

> The Bus Pirate firmware can have issues extracting memory above 128K on the ATmega 2560-2561. To resolve that issue, you will need to have the STK500v2 firmware installed on the Bus Pirate. The STK500v2 firmware will turn the Bus Pirate into an AVR debugger clone.

> The firmware to this can be [downloaded here](https://code.google.com/archive/p/the-bus-pirate/downloads), and the directions for upgrading the Bus Pirate firmware can be found at [this website](http://dangerousprototypes.com/blog/2009/08/06/bus-pirate-firmware-upgrades-on-linux-osx/).

mcuee, 2022-07-28:
So this is a FW issue of the BusPirate. Not so sure if it can be enhanced like the USBtinyISP (#992) to be able to program the ATmega1284 and the ATmega2560/2561.

mcuee, 2023-06-03:
Close this for now. 

We can re-open this if there is FW update or if someone is working on this issue.

closed 2023-06-03 by mcuee