avrdudes/avrdude #462: [bug #52318] Fuse progamming error

avrdude-admin, 2021-12-11:
Art <a.s.barulin@hotmail.com>
Tue 31 Oct 2017 05:50:42 PM UTC
Programmer hardware: usbasp
Device type: attiny10

As per the title. I wrote the fuse value of 0xfe which disables the reset pin on this MCU. I then put 12V on the reset pin, and tried to write the value 0xff, which should enable it again. AVR dude says the value was written, but then spits out this:
avrdude: verification error, first mismatch at byte 0x0000
0xfe != 0xff
I tried a few other fuse values, but none of them worked. Not a big deal as I could still reprogram the unit as normal, so long as I had 12V on the reset pin. I couldn't leave well enough alone though, so I went googling. I came across a german forum post which I misunderstood. Consequently, I tried writing the fuse value of 0x00, and it worked. Now that seems to have done something strange to the MCU.
I still seem to be able to write to it, but the output doesn't match what I'd expect. For example, if I set it to toggle a pin every 500ms, it seems to do it every 50ms.
When I supply 12V to the reset pin & then try to write the fuse, I get this:
C:\Program Files (x86)\MHV AVR Tools\bin>avrdude -p attiny10 -P usb -c usbasp -B 10 -v -u -U fuse:w:0xFF:m

avrdude: Version 6.2, compiled on Nov 20 2015 at 23:33:37
Copyright (c) 2000-2005 Brian Dean, [http://www.bdmicro.com/](https://savannah.nongnu.orghttp://www.bdmicro.com/)
Copyright (c) 2007-2014 Joerg Wunsch

System wide configuration file is "C:\Program Files (x86)\MHV AVR Tools\bin\avrdude.conf"

Using Port                    : usb
Using Programmer              : usbasp
Setting bit clk period        : 10.0
AVR Part                      : ATtiny10
Chip Erase delay              : 0 us
PAGEL                         : P00
BS2                           : P00
RESET disposition             : dedicated
RETRY pulse                   : SCK
serial program mode           : yes
parallel program mode         : yes
Timeout                       : 0
StabDelay                     : 0
CmdexeDelay                   : 0
SyncLoops                     : 0
ByteDelay                     : 0
PollIndex                     : 0
PollValue                     : 0x00
Memory Detail                 :

Block Poll               Page
Polled
Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
signature      0     0     0    0 no          3   16      0     0     0 0x00 0x00
fuse           0     0     4    0 no          1   16      0     0     0 0x00 0x00
calibration    0     0     0    0 no          1   16      0     0     0 0x00 0x00
lockbits       0     0     0    0 no          1   16      0     0     0 0x00 0x00
flash          0     0   128    0 no       1024   16      0     0     0 0x00 0x00

Programmer Type : usbasp
Description     : USBasp, [http://www.fischl.de/usbasp/](https://savannah.nongnu.orghttp://www.fischl.de/usbasp/)

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.03s

avrdude: Device signature = 0x1e9003 (probably t10)
avrdude: reading input file "0xFF"
avrdude: writing fuse (1 bytes):

Writing | ################################################## | 100% 0.11s

avrdude: 1 bytes of fuse written
avrdude: verifying fuse memory against 0xFF:
avrdude: load data fuse data from input file 0xFF:
avrdude: input file 0xFF contains 1 bytes
avrdude: reading on-chip fuse data:

Reading | ################################################## | 100% 0.03s

avrdude: verifying ...
avrdude: verification error, first mismatch at byte 0x0000
0x00 != 0xff
avrdude: verification error; content mismatch

avrdude done.  Thank you.

C:\Program Files (x86)\MHV AVR Tools\bin>

This issue was migrated from https://savannah.nongnu.org/bugs/?52318

avrdude-admin, 2021-12-11:
Joerg Wunsch <joerg_wunsch>
Tue 31 Oct 2017 10:18:47 PM UTC

Question is whether this is actually related to the USBasp firmware or to AVRDUDE itself.

Do you have a reference to the German forum post you mentioned?

Configuration byte 0 bits 7:3 are "reserved", so if you program them away from their default value, there are no guarantees whatsoever. Quite possible they might contain fuses that have never been released to the public, for whatever reasons.

avrdude-admin, 2021-12-11:
Artem <artbar>
Wed 01 Nov 2017 08:17:32 AM UTC

OK. Fuse of this controller is always written down. If you write 0xfe then you do not write 0xff, but you can write 0xfc. And so on down.
Links unfortunately I do not have since I found this post on the network when I was looking for a solution. 0x00 I wrote down just like that. And so it happened that coincided. It is not necessary to write 0x00 the problem is that after recording even 0xfe (0xfc etc.) can not all be returned back.

avrdude-admin, 2021-12-11:
Artem <artbar>
Wed 01 Nov 2017 08:20:55 AM UTC

One of these days comes the programmer avr mk !!. I'll try with it and write the result.

avrdude-admin, 2021-12-11:
Joerg Wunsch <joerg_wunsch>
Wed 01 Nov 2017 08:54:57 AM UTC

Thanks for the response. I'll see to try it myself, too. I do have some ad-hoc USBasp somewhere, but I'm afraid I have to find a recent firmware for it that handles the ATtiny10.

avrdude-admin, 2021-12-11:
Artem <artbar>
Thu 02 Nov 2017 05:06:31 AM UTC

The flush is not correctly stitched.
Maybe this is of course also the compiler is to blame. And on fuse addition, despite the fact that it was stitched 0xFE activated WDT and does not turn off .... only #asm (wdr) helps.

MCUdude, 2022-04-14:
I can confim that this is not an issue anymore. I'm able to recover an ATtiny10 just fine by applying 12V to the RESET pin while programming. Please open a new issue if this somehow still is a problem.

```
$ ./avrdude -Cavrdude.conf -pattiny10 -cusbasp -Ufuse:w:0xfe:m

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e9003 (probably t10)
avrdude: reading input file "0xfe"
avrdude: writing fuse (1 bytes):

Writing | ################################################## | 100% 0.07s

avrdude: 1 bytes of fuse written
avrdude: verifying fuse memory against 0xfe:

Reading | ################################################## | 100% 0.00s

avrdude: 1 bytes of fuse verified

avrdude done.  Thank you.

$ ./avrdude -Cavrdude.conf -pattiny10 -cusbasp -Ufuse:w:0xff:m

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e9003 (probably t10)
avrdude: reading input file "0xff"
avrdude: writing fuse (1 bytes):

Writing | ################################################## | 100% 0.07s

avrdude: 1 bytes of fuse written
avrdude: verifying fuse memory against 0xff:

Reading | ################################################## | 100% 0.00s

avrdude: 1 bytes of fuse verified

avrdude done.  Thank you.
```

closed 2022-04-14 by MCUdude