avrdudes/avrdude #463: [bug #52645] (0xc000007b) with last windows update

avrdude-admin, 2021-12-11:
Patrick <support@emoshape.com>
Tue 12 Dec 2017 01:33:29 AM UTC

Hi,
Windows just did a major update (Windows 10) Since then the exe crash instantly at start.
(0xc000007b)
We have 64bits and use version 6.3 of avrdude.exe, we tried to set the compatibility to windows 8 or XP but nothing helps
Thanks

[file #42613: Screenshot 2017-12-11 20.30.42.png](https://savannah.nongnu.org/bugs/download.php?file_id=42613)

This issue was migrated from https://savannah.nongnu.org/bugs/?52645

avrdude-admin, 2021-12-11:
Joerg Wunsch <joerg_wunsch>
Wed 13 Dec 2017 09:40:31 PM UTC

Here's the reply Bob Paddock sent to the mailinglist:

```
This can happen if a 32-bit DLL is still lingering in the PATH for some reason.

Open a DOS box, create an empty directory.
Copy avrdude.exe and config file to the empty directory.
Set PATH to nothing ie. PATH=C:\YourEmptyDirectory
Run avrdude

Does it still crash?

No: Update dotNET and DirectX.  Make sure ALL of the Windows10 updates
have been done, it is a recursive process.
Check path for old tools.  I've found it best to have  each
project setup its own PATH for this reason.
Don't allow installs to update the global path.

Yes: The program may need rebased with the MinGW rbase tool after
recent Windows10 updates.
As I'm running Win10-64bit here with freshly downloaded
avrdude 6.3  which ran fine this is unlikely.
This usually happens with old programs like WinAVR20100110.
For completeness, doesn't really  apply to your problem, might
help someone else coming across this:
$ rebase.exe -b 0x50000000 msys-1.0.dll
https://nurikabe.blog/2014/02/14/couldnt-reserve-space-for-cygwins-heap-win32-error-0/

```

mariusgreuel, 2022-01-12:
0xC000007B => STATUS_INVALID_IMAGE_FORMAT

Op probably attempted to load a 32-bit DLL into a 64-bit exe, or vice versa => not an avrdude issue.

closed 2022-01-12 by mariusgreuel