avrdudes/avrdude #464: [bug #52653] device initialization often fails in FT232R syncbb mode, due to glitchy start sequence

avrdude-admin, 2021-12-11:
rai42 <None>
Tue 12 Dec 2017 11:57:30 PM UTC
Programmer hardware: FTDI FT232RL
Device type: ATmega8

SETUP:

Using a custom board with ATmega8 and FT232RL and the following config/connections:

programmer
id    = "ftdi_uni_uC";
desc  = "FT232R Synchronous BitBang";
type  = "ftdi_syncbb";
connection_type = usb;
baudrate = 38400;
reset = 7; #RI
sck   = 3; #CTS
mosi  = 6; #DCD
miso  = 5; #DSR
;

I used avrdude 6.2 (but the relevant file ft245r.c is identical with 6.3).

THE PROBLEM: I find that device initialization often (about 50% of tries) fails as follows:

$ sudo ./avrdude -p m8 -c ftdi_uni_uC -P ft0

avrdude: Device is not responding to program enable. Check connection.
avrdude: initialization failed, rc=-1
Double check connections and try again, or use -F to override
this check.

---
Investigating with an oscilloscope, I found that during ft245r_open() the signals behave as follows:

1) start with RESET=1, SCK=0
2) calling ftdi_set_bitmode (line 614) => sets RESET=0
3) calling ft245r_drain (line 638) =>
shortly pulses RESET=1 and SCK=1 simultaneously, then back to 0
This seems to happen due to the switching of bitmodes back&forth in ft245r_drain().
4) calling ft245r_send (line 640) - set output to ft245r_out => sets RESET=1

This is followed by the actual init sequence in ft245r_initialize().

It appears that the simultaneous pulsing of RESET and SCK (step 3 above) puts the ATmega (already powered up since VCC is not controlled) in an undefined state. This is not (always) fixed by the folloiwing init sequence, despite pulsing RESET high in accorance with the datasheet.

I assume that programmers that control VCC will not be affected by this problem, since the AVR is not powered up before ft245r_initialize() runs.

PROPOSED PATCH (see attached ft245r.patch):

- change ft245r_drain() to use ftdi_read_data until buffer is empty, instead of bitmode switching- call ft245r_drain() before setting up the reader thread, rather than after (make sure reader thread starts with empty buffer).This fixed the issue for me (i.e. initialization and further operations now work reliably).

[file #42625: ft245r.patch](https://savannah.nongnu.org/bugs/download.php?file_id=42625)

This issue was migrated from https://savannah.nongnu.org/bugs/?52653

mcuee, 2022-05-29:
The code has changed quite a bit, I am not so sure wether this patch is still valid or not.

mcuee, 2022-06-12:
Just tested with ATmega328p on the Arduino Uno and there is no issue, using a FT232R USB to Serial Converter with AVR ISP header (similar to https://i.stack.imgur.com/lmG5D.png ) from the following discussion.
https://arduino.stackexchange.com/questions/30564/ftdi-breakout-with-additional-isp-connector

```
PS C:\work\avr\avrdude_test\avrdude-v7.0-windows-x64> .\avrdude -p atmega328p -c arduino-ft232r -P ft0 -v -D -U flash:v:.\Blink.ino.standard.hex:i

avrdude.exe: Version 7.0
             Copyright (c) Brian Dean, http://www.bdmicro.com/
             Copyright (c) Joerg Wunsch

             System wide configuration file is "C:/work/avr/avrdude_test/avrdude-v7.0-windows-x64/avrdude.conf"

             Using Port                    : ft0
             Using Programmer              : arduino-ft232r
avrdude.exe: ft245r_open(): no device identifier in portname, using default
             AVR Part                      : ATmega328P
             Chip Erase delay              : 9000 us
             PAGEL                         : PD7
             BS2                           : PC2
             RESET disposition             : dedicated
             RETRY pulse                   : SCK
             Serial program mode           : yes
             Parallel program mode         : yes
             Timeout                       : 200
             StabDelay                     : 100
             CmdexeDelay                   : 25
             SyncLoops                     : 32
             PollIndex                     : 3
             PollValue                     : 0x53
             Memory Detail                 :

                                               Block Poll               Page                       Polled
               Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
               ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
               eeprom                 65    20     4    0 no       1024    4      0  3600  3600 0xff 0xff
               flash                  65     6   128    0 yes     32768  128    256  4500  4500 0xff 0xff
               lfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
               hfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
               efuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
               lock                    0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
               calibration             0     0     0    0 no          1    1      0     0     0 0x00 0x00
               signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00

             Programmer Type : ftdi_syncbb
             Description     : Arduino: FT232R connected to ISP
             Pin assignment  : 0..7 = DBUS0..7
               VCC     =  (not used)
               BUFF    =  (not used)
               RESET   =  7
               SCK     =  5
               MOSI    =  6
               MISO    =  3
               ERR LED =  (not used)
               RDY LED =  (not used)
               PGM LED =  (not used)
               VFY LED =  (not used)

avrdude.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.01s

avrdude.exe: Device signature = 0x1e950f (probably m328p)
avrdude.exe: verifying flash memory against .\Blink.ino.standard.hex:

Reading | ################################################## | 100% 54.76s

avrdude.exe: 924 bytes of flash verified

avrdude.exe done.  Thank you.
```

mcuee, 2022-06-12:
Using faster baudrate of 38400 is also okay.

```
PS C:\work\avr\avrdude_test\avrdude-v7.0-windows-x64> .\avrdude -p atmega328p -c arduino-ft232r -P ft0 -v -b 38400 -D -U flash:w:.\Blink.ino.standard.hex:i

avrdude.exe: Version 7.0
             Copyright (c) Brian Dean, http://www.bdmicro.com/
             Copyright (c) Joerg Wunsch

             System wide configuration file is "C:/work/avr/avrdude_test/avrdude-v7.0-windows-x64/avrdude.conf"

             Using Port                    : ft0
             Using Programmer              : arduino-ft232r
             Overriding Baud Rate          : 38400
avrdude.exe: ft245r_open(): no device identifier in portname, using default
             AVR Part                      : ATmega328P
             Chip Erase delay              : 9000 us
             PAGEL                         : PD7
             BS2                           : PC2
             RESET disposition             : dedicated
             RETRY pulse                   : SCK
             Serial program mode           : yes
             Parallel program mode         : yes
             Timeout                       : 200
             StabDelay                     : 100
             CmdexeDelay                   : 25
             SyncLoops                     : 32
             PollIndex                     : 3
             PollValue                     : 0x53
             Memory Detail                 :

                                               Block Poll               Page                       Polled
               Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
               ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
               eeprom                 65    20     4    0 no       1024    4      0  3600  3600 0xff 0xff
               flash                  65     6   128    0 yes     32768  128    256  4500  4500 0xff 0xff
               lfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
               hfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
               efuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
               lock                    0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
               calibration             0     0     0    0 no          1    1      0     0     0 0x00 0x00
               signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00

             Programmer Type : ftdi_syncbb
             Description     : Arduino: FT232R connected to ISP
             Pin assignment  : 0..7 = DBUS0..7
               VCC     =  (not used)
               BUFF    =  (not used)
               RESET   =  7
               SCK     =  5
               MOSI    =  6
               MISO    =  3
               ERR LED =  (not used)
               RDY LED =  (not used)
               PGM LED =  (not used)
               VFY LED =  (not used)

avrdude.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.04s

avrdude.exe: Device signature = 0x1e950f (probably m328p)
avrdude.exe: reading input file ".\Blink.ino.standard.hex"
avrdude.exe: writing flash (924 bytes):

Writing | ################################################## | 100% 10.06s

avrdude.exe: 924 bytes of flash written
avrdude.exe: verifying flash memory against .\Blink.ino.standard.hex:

Reading | ################################################## | 100% 10.96s

avrdude.exe: 924 bytes of flash verified

avrdude.exe done.  Thank you.
```

mcuee, 2022-06-12:
I tested with both cases: the Uno is powered by the USB cable; or it is not connected to the USB cable and powered by the FT232R USB to Serial converter. avrdude 7.0 works fine in both cases.

mcuee, 2022-06-12:
@dl8dtl and @mariusgreuel 
Please take a look at the patch as well. If the patch is no longer relevant, this issue can be closed.

mcuee, 2022-06-20:
I think I encountered a few times of this issue during my testing of FT232R. I will try a bit more.

mcuee, 2022-06-20:
My issue is probably different as it was with a buggy Arduini Mega2560 CH340 clone board which may have HW reset issues. So I will change the label.

mcuee, 2022-06-25:
Most likely the issue has been fixed and no longer valid.

mcuee, 2022-07-10:
I will close this as `invalid` for now. Please re-open when the issue occurs.

closed 2022-07-10 by mcuee