avrdudes/avrdude #465: [bug #52687] avrdude eeprom uploader always uploads 0xff if line terminator is LF not CRLF

avrdude-admin, 2021-12-11:
James Collier <james.collier@awelina.uk>
Mon 18 Dec 2017 10:22:42 AM UTC
Programmer hardware: USBTiny
Device type: atmaga328p

It is useful to create eeprom .hex files from a program rather than from avr-gcc, for instance in a production environment to set calibration constants or a unique ID.

Uploading the resulting .hex file - in the correct form - using

>avrdude -p atmega328p -c -usbtiny -U eprom:w:myfile.hex

works fine: Writing and Reading  get the progress lines of # characters, then the message
avrdude: verifying...
avrdude: 80 bytes of eeprom verified
etc.

BUT it loads every byte as 0xff if the .hex file has  \n line terminators. If the hex file has \r\n terminators avrdude loads the correct data. There is no difference in the messages from avrdude in either case.

I consider this a bug as (a) it's poor practice to make parsers unnessarily strict, and (b) apparent success should not be reported when the operation has, in fact, failed.

This issue was migrated from https://savannah.nongnu.org/bugs/?52687

avrdude-admin, 2021-12-11:
Joerg Wunsch <joerg_wunsch>
Fri 29 Dec 2017 11:29:37 PM UTC

I'd have replied that Intel Hex files are by definition to be
terminated with CR-LF, but it seems Intel indeed never
specified this.

The behaviour you are mentioning does by no means to be
intended by the way the code is written.  Curious, are you
working on Windows?  The code uses fgets() to read the file
line by line, but on Windows, that probably requires CR-LF
line termination.  (On Unix-like systems, it just needs LF,
and any possible CR is later ignored.)  Replacing fgets()
is, unfortunately, quite cumbersome as it is very likely that
any manual reimplementation is going to be slower than the
library-provided fgets().

avrdude-admin, 2021-12-11:
Ryniker <ryniker>
Sat 30 Dec 2017 01:21:32 AM UTC

Windows and line termination characters may or may not be the problem, but removal of white space from the end of a line is not likely to have any noticeable impact on performance.

avrdude-admin, 2021-12-11:
Joerg Wunsch <joerg_wunsch>
Sat 30 Dec 2017 07:23:47 AM UTC

The question is not about removing whitespace but reimplementing
fgets() rather than using the system/library version.

mariusgreuel, 2022-01-10:
Both the Intel HEX and S-Record reader are insensitive to line endings, in fact arbitrary characters may be added to the end of each line. Without any further information it is not possible to solve this issue.

closed 2022-01-10 by mariusgreuel