avrdudes/avrdude #467: [bug #52886] Incorrectly hardcoded fuse addresses in STK500v2 ISP mode

avrdude-admin, 2021-12-11:
Joao Rosa <jmr@ff.ul.pt>
Sun 14 Jan 2018 03:41:54 PM UTC

On file "stk500v2.c", functions stk500isp_write_byte() and stk500isp_read_byte() incorrectly hardcode the address of "fuse"/"lfuse", "hfuse" and "efuse".

This is only needed for HVSP and PP modes, not ISP (probably copy/paste error?). Though this is a "silent" bug that currently doesn't have any effect, it may conceivably manifest itself in the future if future chips access the fuses using some kind of addressing scheme, that is, using a0, a1, a2, etc. in the avrdude.conf definitions.

In fact I found this bug because I'm writing a programmer for UPDI chips and want to use the STK500v2 interface.

I'll copy a few lines of code from "stk500v2.c" as example:

1942 	  } else if (strcmp(mem->desc, "hfuse") == 0) {
1943 	    buf[0] = CMD_READ_FUSE_ISP;
1944 	    addr = 1;   /* This line is not needed/bug */

I believe lines like this one can be safely deleted.

This issue was migrated from https://savannah.nongnu.org/bugs/?52886

mcuee, 2022-06-03:
Just wondering if the existing codes were written due to concerns on backwards compatibility? Is it really safe to remove them for existing chips?
https://github.com/avrdudes/avrdude/blob/main/src/stk500v2.c#L1942

```
  if (strcmp(mem->desc, "lfuse") == 0 ||
	     strcmp(mem->desc, "fuse") == 0) {
    buf[0] = CMD_READ_FUSE_ISP;
    addr = 0;
  } else if (strcmp(mem->desc, "hfuse") == 0) {
    buf[0] = CMD_READ_FUSE_ISP;
    addr = 1;
  } else if (strcmp(mem->desc, "efuse") == 0) {
    buf[0] = CMD_READ_FUSE_ISP;
    addr = 2;
  } else if (strcmp(mem->desc, "lock") == 0) {
    buf[0] = CMD_READ_LOCK_ISP;
  } else if (strcmp(mem->desc, "calibration") == 0) {
    buf[0] = CMD_READ_OSCCAL_ISP;
  } else if (strcmp(mem->desc, "signature") == 0) {
    buf[0] = CMD_READ_SIGNATURE_ISP;
  }

```

mcuee, 2022-06-20:
Closing this issue for now as it does not affect avrdude as of now.