avrdudes/avrdude #468: [bug #52963] -U option with flash memory doesn't perform chip erase with arduino programmer

avrdude-admin, 2021-12-11:
Britton Kerin <britton.kerin@gmail.com>
Tue 23 Jan 2018 11:13:18 PM UTC
Programmer hardware: arduino

It can't because that would blow away the bootlaoder.
The -D option description currently says this:

-D      Disable auto erase for flash.  When the -U option with flash
memory is specified, avrdude will perform a chip erase before
starting any of the programming operations, since it generally is a
mistake to program the flash without performing an erase first.  This
option disables that.  Auto erase is not used for ATxmega devices as
these devices can use page erase before writing each page so no
explicit chip erase is required.  Note however that any page not
affected by the current operation will retain its previous contents.

If exceptions for ATxmega deserve a mention that the arduino programmer
exception certainly does too.  It can matter because e.g. EEPROM doesn't end up
getting cleared like documentation in chip datasheet says it does for chip
erase (assuming EESAVE not programmed).

This issue was migrated from https://savannah.nongnu.org/bugs/?52963

avrdude-admin, 2021-12-11:
Joerg Wunsch <joerg_wunsch>
Tue 23 Jan 2018 11:30:15 PM UTC

IMHO this is entirely an issue of the bootloader but not
AVRDUDE. AVRDUDE requests the programmer to issue a chip
erase, and obviously, a bootloader would be stupid to execute
it in a way to erase itself. ;-)  Anyway, simply ignoring the
erase request is certainly **not** the correct way to handle it.
It could either erase the entire application memory region,
or it could just remember it was asked to perform an erase,
and then erase each page just before reprogramming it.

The Xmega way cannot be used by AVRDUDE on other devices as
there are simply no commands to perform page-wise erases on
them.

This is deja-vu, I seem to remember we've been discussing
this bootloader bug before.  Please complain to Arduino about
it.

MCUdude, 2022-01-07:
Issue can probably be closed

closed 2022-01-08 by dl8dtl