avrdudes/avrdude #469: [bug #53008] avrdude fails to read more than 440 bytes of EEPROM

avrdude-admin, 2021-12-11:
The Lightning Stalker <None>
Mon 29 Jan 2018 11:33:56 PM UTC
Programmer hardware: USBtinyISP
Device type: m8

avrdude is unable to read the EEPROM from this device. Oddly enough writing and verifying seem to work. It seems more likely to be a problem with avrdude than USBtinyISP because otherwise how would it be able to verify?

Regardless of ihex or any other format it will read several kB of data then segfault at 85%.

$ avrdude -v -pm8 -cusbtiny -Ueeprom:r:eep:i

avrdude: Version 6.3
Copyright (c) 2000-2005 Brian Dean, [http://www.bdmicro.com/](https://savannah.nongnu.orghttp://www.bdmicro.com/)
Copyright (c) 2007-2014 Joerg Wunsch

System wide configuration file is "/etc/avrdude.conf"
User configuration file is "/home/robert/.avrduderc"
User configuration file does not exist or is not a regular file, skipping

Using Port                    : usb
Using Programmer              : usbtiny
Setting bit clk period        : 1.0
avrdude: usbdev_open(): Found USBtinyISP, bus:device: 002:032
AVR Part                      : ATmega8
Chip Erase delay              : 10000 us
PAGEL                         : PD7
BS2                           : PC2
RESET disposition             : dedicated
RETRY pulse                   : SCK
serial program mode           : yes
parallel program mode         : yes
Timeout                       : 200
StabDelay                     : 100
CmdexeDelay                   : 25
SyncLoops                     : 32
ByteDelay                     : 0
PollIndex                     : 3
PollValue                     : 0x53
Memory Detail                 :

Block Poll               Page                       Polled
Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
eeprom         4    20   128    0 no        512    4      0  9000  9000 0xff 0xff
flash         33    10    64    0 yes      8192   64    128  4500  4500 0xff 0x00
lfuse          0     0     0    0 no          1    0      0  2000  2000 0x00 0x00
hfuse          0     0     0    0 no          1    0      0  2000  2000 0x00 0x00
lock           0     0     0    0 no          1    0      0  2000  2000 0x00 0x00
calibration    0     0     0    0 no          4    0      0     0     0 0x00 0x00
signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

Programmer Type : USBtiny
Description     : USBtiny simple USB programmer, [http://www.ladyada.net/make/usbtinyisp/](https://savannah.nongnu.orghttp://www.ladyada.net/make/usbtinyisp/)
avrdude: programmer operation not supported

avrdude: Setting SCK period to 1 usec
avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e9307 (probably m8)
avrdude: safemode: hfuse reads as CC
avrdude: reading eeprom memory:

Reading | ###########################################        | 85% 5.23sSegmentation fault

This issue was migrated from https://savannah.nongnu.org/bugs/?53008

avrdude-admin, 2021-12-11:
The Lightning Stalker <lightningstalker>
Tue 30 Jan 2018 01:34:40 AM UTC

By changing line 6448 in avrdude.conf I can read up to 440 bytes of EEPROM. Attempting to read 441 or greater results in the segfault. Luckily I only need the first 89B but anyone who needs >440B is SOL.

avrdude-admin, 2021-12-11:
The Lightning Stalker <lightningstalker>
Tue 30 Jan 2018 01:37:17 AM UTC

I thought it was reading large amounts of memory but turns out reading EEPROM is just slower than flash. The name of this bug should really be "avrdude cannot read more than 440 bytes of EEPROM".

avrdude-admin, 2021-12-11:
Joerg Wunsch <joerg_wunsch>
Tue 30 Jan 2018 07:37:59 PM UTC

Well, I got occasional retries on reading the EEPROM (heavily
depending on the clock value), sometimes the firmware appears
to fail, but if it passes, I get full 512 bytes of EEPROM from
an ATmega16 (don't have an ATmega8 handy):

```
./avrdude -C avrdude.conf -p m16  -c usbtiny -U ee:r:foo.hex:i

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.02s

avrdude: Device signature = 0x1e9403 (probably m16)
avrdude: reading eeprom memory:

Reading | #                                                  | 2% 0.40savrdude: 1 retries during read
Reading | #####                                              | 9% 1.81s
avrdude: error: usbtiny_receive: Unknown error (expected 128, got -1)
Reading | #####                                              | 10% 2.58savrdude: 10 retries during SPI command
Reading | ##########################                         | 52% 4.19savrdude: 3 retries during SPI command
Reading | ################################################## | 100% 5.85s

avrdude: writing output file "foo.hex"

avrdude: safemode: Fuses OK (E:FF, H:99, L:E4)

avrdude done.  Thank you.

```
The target is sitting in an STK500.  If I use the STK500
itself to read the EEPROM, everything is fine (and fast).

I also tried with the junk fill option of malloc enabled
(this is FreeBSD), to imcrease the probability of hitting
a segfault, but it passes just the same as above.

Can you analyze the segfault with a debugger, to see where
it crashes for you?

avrdude-admin, 2021-12-11:
The Lightning Stalker <lightningstalker>
Thu 01 Feb 2018 02:15:52 AM UTC

Well I'm not sure why but rebooting seems to have fixed it. idk computers are strange

MCUdude, 2022-03-28:
I've tested the UsbTinyISP programmer on an ATmega8 and am able to read and write to the EEPROM without any issues. I've even tried to run the ATmega8 and 1 MHz and the USBtinyiSP with clock speed `-B3` which is the fastest I can run without the programmer being too fast for the target. Note that I'm running the latest upstream Avrdude version on a mac.

I'm closing this issue because I'm not able to reproduce the issue on actual hardware, the issue resolved itself for "The Lightning Stalker". Feel free to open a new issue if this turns out to still be a problem!

closed 2022-03-28 by MCUdude