avrdudes/avrdude #472: [bug #53279] Permission denied accessing /sys/class/gpio/gpioX/direction would leave pin exported

avrdude-admin, 2021-12-11:
Michael Cain <loud.mouth.braggart@gmail.com>
Sat 03 Mar 2018 10:10:16 PM UTC
Programmer hardware: linuxgpio
Device type: RaspberryPi

When using a Raspberry Pi to program an atmega2560 using linuxgpio and you do not have the permission to access /sys/class/gpio/gpioX/direction it would leave the gpio pin exported.  The only way to unexport the pin would be to reboot the Rasbperry Pi.

The problem lies in the function linuxgpio_open, change line 280 from:
if (r < 0)
return r;

TO:
if (r < 0) {
linuxgpio_unexport(pin);
return r;
}

This issue was migrated from https://savannah.nongnu.org/bugs/?53279

mcuee, 2022-06-06:
Yes this is indeed similar to #917 and #372.
* #917 
* #372

mcuee, 2022-07-10:
I will mark this as duplicate of #386 and the fix is in #917.

closed 2022-07-10 by mcuee