avrdudes/avrdude #474: [bug #53703] FT232H based programmer returns all 0xFF when reading flash from atmega2560

avrdude-admin, 2021-12-11:
Christoph <ceaich@gmx.de>
Thu 19 Apr 2018 10:07:59 PM UTC
Programmer hardware: FT232H
Device type: m2560

avrftdi.c does not send load extended address byte command when address is lower than 64k word but using a device which need this command. See attached patch for bug-fix.

[file #43979: avrftdi.c.patch](https://savannah.nongnu.org/bugs/download.php?file_id=43979)

This issue was migrated from https://savannah.nongnu.org/bugs/?53703

mcuee, 2022-06-05:
This may be a valid one. I will try to set up my FT2232H based programmer to carry out this test.

mcuee, 2022-06-08:
Reading also has some problem with content mis-match but not really returns all 0xFF.

```
PS C:\work\avr\avrdude_test\avrdude-v7.0-windows-x64>  .\avrdude.exe -p m2560 -c jtagkey -U flash:r:.\mega2560_test_ftdi2232h_read.hex:i

avrdude.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude.exe: Device signature = 0x1e9801 (probably m2560)
avrdude.exe: reading flash memory:

Reading | ################################################## | 100% 57.58s

avrdude.exe: writing output file ".\mega2560_test_ftdi2232h_read.hex"

avrdude.exe done.  Thank you.

PS C:\work\avr\avrdude_test\avrdude-v7.0-windows-x64>  .\avrdude.exe -p m2560 -c jtagkey -U flash:v:.\mega2560_test_ftdi2232h_read.hex:i

avrdude.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude.exe: Device signature = 0x1e9801 (probably m2560)
avrdude.exe: verifying flash memory against .\mega2560_test_ftdi2232h_read.hex:

Reading | ################################################## | 100% 57.44s

avrdude.exe: 261414 bytes of flash verified

avrdude.exe done.  Thank you.

PS C:\work\avr\avrdude_test\avrdude-v7.0-windows-x64>  .\avrdude.exe -p m2560 -c wiring -P COM6 -U flash:v:.\mega2560_test_ftdi2232h_read.hex:i

avrdude.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.01s

avrdude.exe: Device signature = 0x1e9801 (probably m2560)
avrdude.exe: verifying flash memory against .\mega2560_test_ftdi2232h_read.hex:

Reading | ################################################## | 100% 30.95s

avrdude.exe: verification error, first mismatch at byte 0x1e000
             0xff != 0x0d
avrdude.exe: verification error; content mismatch

avrdude.exe done.  Thank you.

PS C:\work\avr\avrdude_test\avrdude-v7.0-windows-x64>  .\avrdude.exe -p m2560 -c dragon_isp -U flash:v:.\mega2560_test_ftdi2232h_read.hex:i

avrdude.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.15s

avrdude.exe: Device signature = 0x1e9801 (probably m2560)
avrdude.exe: verifying flash memory against .\mega2560_test_ftdi2232h_read.hex:

Reading | ################################################## | 100% 120.73s

avrdude.exe: verification error, first mismatch at byte 0x1e000
             0xff != 0x0d
avrdude.exe: verification error; content mismatch

avrdude.exe done.  Thank you.

```

mcuee, 2022-06-08:
Readback differences using FT2232H and AVR Dragon.

Using the wiring bootloader to verify -- this shows that AVR Dragon is correct.

```
PS C:\work\avr\avrdude_test\avrdude-v7.0-windows-x64>  .\avrdude.exe -p m2560 -c wiring -P COM6 -U flash:v:.\mega2560_test_dragonisp_read.hex:i

avrdude.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.01s

avrdude.exe: Device signature = 0x1e9801 (probably m2560)
avrdude.exe: verifying flash memory against .\mega2560_test_dragonisp_read.hex:

Reading | ################################################## | 100% 30.92s

avrdude.exe: 261414 bytes of flash verified

avrdude.exe done.  Thank you.

PS C:\work\avr\avrdude_test\avrdude-v7.0-windows-x64>  .\avrdude.exe -p m2560 -c wiring -P COM6 -U flash:v:.\mega2560_test_ftdi2232h_read.hex:i

avrdude.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.01s

avrdude.exe: Device signature = 0x1e9801 (probably m2560)
avrdude.exe: verifying flash memory against .\mega2560_test_ftdi2232h_read.hex:

Reading | ################################################## | 100% 30.92s

avrdude.exe: verification error, first mismatch at byte 0x1e000
             0xff != 0x0d
avrdude.exe: verification error; content mismatch

avrdude.exe done.  Thank you.
```

[mega2560_test_dragonisp_read.hex.txt](https://github.com/avrdudes/avrdude/files/8861927/mega2560_test_dragonisp_read.hex.txt)
[mega2560_test_ftdi2232h_read.hex.txt](https://github.com/avrdudes/avrdude/files/8861928/mega2560_test_ftdi2232h_read.hex.txt)

mcuee, 2022-06-08:
I can confirm the patch here fixed the reading issue for me.

```
PS C:\work\avr\avrdude_test\avrdude-7.0_bin64>  .\avrdude_issue474.exe -p m2560 -c jtagkey -U flash:r:.\mega2560_test_ftdi2232h_read474.hex:i

avrdude_issue474.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.01s

avrdude_issue474.exe: Device signature = 0x1e9801 (probably m2560)
avrdude_issue474.exe: reading flash memory:

Reading | ################################################## | 100% 57.90s

avrdude_issue474.exe: writing output file ".\mega2560_test_ftdi2232h_read474.hex"

avrdude_issue474.exe done.  Thank you.

PS C:\work\avr\avrdude_test\avrdude-7.0_bin64>  .\avrdude.exe -p m2560 -c wiring -P COM6 -U flash:v:.\mega2560_test_ftdi2232h_read474.hex:i

avrdude.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.02s

avrdude.exe: Device signature = 0x1e9801 (probably m2560)
avrdude.exe: verifying flash memory against .\mega2560_test_ftdi2232h_read474.hex:

Reading | ################################################## | 100% 31.77s

avrdude.exe: 261414 bytes of flash verified

avrdude.exe done.  Thank you.

PS C:\work\avr\avrdude_test\avrdude-7.0_bin64>  .\avrdude.exe -p m2560 -c dragon_isp -U flash:v:.\mega2560_test_ftdi2232h_read474.hex:i

avrdude.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.17s

avrdude.exe: Device signature = 0x1e9801 (probably m2560)
avrdude.exe: verifying flash memory against .\mega2560_test_ftdi2232h_read474.hex:

Reading | ################################################## | 100% 122.06s

avrdude.exe: 261414 bytes of flash verified

avrdude.exe done.  Thank you.

```

mcuee, 2022-06-08:
@dl8dtl and @mariusgreuel 

My test shows that the patch (I applied the patch on git head) is good. Please apply. Thanks.

```
PS C:\work\avr\avrdude> git diff
diff --git a/src/avrftdi.c b/src/avrftdi.c
index e41d775..1a186ed 100644
--- a/src/avrftdi.c
+++ b/src/avrftdi.c
@@ -1021,7 +1021,7 @@ static int avrftdi_flash_write(PROGRAMMER * pgm, AVRPART * p, AVRMEM * m,
         * bytes.
         * write the command only once.
         */
-       if(use_lext_address && (((addr/2) & 0xffff0000))) {
+       if (use_lext_address && (addr == 0 || ((addr/2) % /*ext_address_boundary*/ 65536) == 0)) {
                if (0 > avrftdi_lext(pgm, p, m, addr/2))
                        return -1;
        }
@@ -1128,7 +1128,7 @@ static int avrftdi_flash_read(PROGRAMMER * pgm, AVRPART * p, AVRMEM * m,
                return -1;
        }

-       if(use_lext_address && ((address & 0xffff0000))) {
+       if (use_lext_address && (address == 0 || (address % /*ext_address_boundary*/ 65536) == 0)) {
                if (0 > avrftdi_lext(pgm, p, m, address))
                        return -1;
        }

PS C:\work\avr\avrdude_test\avrdude-7.0_bin64> .\avrdude_issue474.exe -v

avrdude_issue474.exe: Version 7.0-20220530 (feaa1c6)
                      Copyright (c) Brian Dean, http://www.bdmicro.com/
                      Copyright (c) Joerg Wunsch

                      System wide configuration file is "C:/work/avr/avrdude_test/avrdude-7.0_bin64/avrdude.conf"

avrdude_issue474.exe: no programmer has been specified on the command line or the config file
                      Specify a programmer using the -c option and try again
```

MCUdude, 2022-06-08:
Thanks for spending your time and testing this patch! As I don't have any 2232 hardware, @dl8dtl or @mariusgreuel should take a look at this. If they approve, someone should create a PR so we get this fix into the main branch.

mariusgreuel, 2022-06-08:
Both the old and patched code look fishy. Isn't the purpose of a lext command to set the upper 8 bits of a 24-bit address? To me, it makes little sense to trigger this based on the state of the lower 16 address bits. We need to send an lext when the upper 8 address bits do not match the CPUs upper 8 bits, i.e. the first time we start programming and every time the upper 8 address bits change.

Could you try programming a hex file that starts at at address 0x1000? I would assume that breaks the patch.

mcuee, 2022-06-09:
> Could you try programming a hex file that starts at at address 0x1000? I would assume that breaks the patch.

Yes I will try. 

If you can come out with a more proper fix, I can test it out as well.

mariusgreuel, 2022-06-09:
@mcuee Thanks for all your help! Thinking about this, 0x1000 should probably be 0x11000.

mcuee, 2022-06-09:
@mariusgreuel 

Not so sure if the attached hex file (stk500v2 bootloader with EEPROM fix) is good to the test condition (starts at 0xE000). Using the hex file it is okay (the bootloader is verified with AVR Dragon and also tested with Arduino).

```
PS C:\work\avr\avrdude_test\avrdude-7.0_bin64>  .\avrdude_issue474.exe -p m2560 -c jtagkey -U .\stk500boot_v2_mega2560_eeprom_fix.hex -v

avrdude_issue474.exe: Version 7.0-20220530 (feaa1c6)
                      Copyright (c) Brian Dean, http://www.bdmicro.com/
                      Copyright (c) Joerg Wunsch

                      System wide configuration file is "C:/work/avr/avrdude_test/avrdude-7.0_bin64/avrdude.conf"

                      Using Port                    : usb
                      Using Programmer              : jtagkey
                      AVR Part                      : ATmega2560
                      Chip Erase delay              : 9000 us
                      PAGEL                         : PD7
                      BS2                           : PA0
                      RESET disposition             : dedicated
                      RETRY pulse                   : SCK
                      Serial program mode           : yes
                      Parallel program mode         : yes
                      Timeout                       : 200
                      StabDelay                     : 100
                      CmdexeDelay                   : 25
                      SyncLoops                     : 32
                      PollIndex                     : 3
                      PollValue                     : 0x53
                      Memory Detail                 :

                                                        Block Poll               Page                       Polled
                        Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
                        ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
                        eeprom                 65    10     8    0 no       4096    8      0  9000  9000 0x00 0x00
                        flash                  65    10   256    0 yes    262144  256   1024  4500  4500 0x00 0x00
                        lfuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
                        hfuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
                        efuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
                        lock                    0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
                        calibration             0     0     0    0 no          1    1      0     0     0 0x00 0x00
                        signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00

                      Programmer Type : avrftdi
                      Description     : Amontec JTAGKey, JTAGKey-Tiny and JTAGKey2

avrdude_issue474.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.02s

avrdude_issue474.exe: Device signature = 0x1e9801 (probably m2560)
avrdude_issue474.exe: NOTE: "flash" memory has been specified, an erase cycle will be performed
                      To disable this feature, specify the -D option.
avrdude_issue474.exe: erasing chip
avrdude_issue474.exe: reading input file ".\stk500boot_v2_mega2560_eeprom_fix.hex"
avrdude_issue474.exe: input file .\stk500boot_v2_mega2560_eeprom_fix.hex auto detected as Intel Hex
avrdude_issue474.exe: writing flash (261414 bytes):

Writing | ################################################## | 100% 0.07s

avrdude_issue474.exe: 261414 bytes of flash written
avrdude_issue474.exe: verifying flash memory against .\stk500boot_v2_mega2560_eeprom_fix.hex:
avrdude_issue474.exe: input file .\stk500boot_v2_mega2560_eeprom_fix.hex auto detected as Intel Hex

Reading | ################################################## | 100% 0.07s

avrdude_issue474.exe: 261414 bytes of flash verified

avrdude_issue474.exe done.  Thank you.

.PS C:\work\avr\avrdude_test\avrdude-7.0_bin64>  .\avrdude_issue474.exe -p m2560 -c dragon_isp -U flash:v:.\stk500boot_v2_mega2560_eeprom_fix.hex

avrdude_issue474.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.18s

avrdude_issue474.exe: Device signature = 0x1e9801 (probably m2560)
avrdude_issue474.exe: verifying flash memory against .\stk500boot_v2_mega2560_eeprom_fix.hex:
avrdude_issue474.exe: input file .\stk500boot_v2_mega2560_eeprom_fix.hex auto detected as Intel Hex

Reading | ################################################## | 100% 0.13s

avrdude_issue474.exe: 261414 bytes of flash verified

avrdude_issue474.exe done.  Thank you.
```
[stk500boot_v2_mega2560_eeprom_fix.hex.txt](https://github.com/avrdudes/avrdude/files/8870497/stk500boot_v2_mega2560_eeprom_fix.hex.txt)

mcuee, 2022-06-09:
> @mariusgreuel
> 
> Not so sure if the attached hex file (stk500v2 bootloader with EEPROM fix) is good to the test condition or not (it starts at 0xE000). Using the hex file it is okay (the bootloader is verified with AVR Dragon and also tested with Arduino).

Sorry it is probably not good enough as you mentioned that the hex file needs to start at 0x11000. I need to learn how to generate a hex file for the testing.

mcuee, 2022-06-09:
@mariusgreuel 
Hopefully the attached hex file meets the testing condition (got the above bootloader, data at 0x10000 and data at 0x11000). It still works.

```
PS C:\work\avr\avrdude_test\avrdude-7.0_bin64>  .\avrdude_issue474.exe -p m2560 -c jtagkey -U flash:w:.\atmega2560_0x11000_offset.hex:i -v

avrdude_issue474.exe: Version 7.0-20220530 (feaa1c6)
                      Copyright (c) Brian Dean, http://www.bdmicro.com/
                      Copyright (c) Joerg Wunsch

                      System wide configuration file is "C:/work/avr/avrdude_test/avrdude-7.0_bin64/avrdude.conf"

                      Using Port                    : usb
                      Using Programmer              : jtagkey
                      AVR Part                      : ATmega2560
                      Chip Erase delay              : 9000 us
                      PAGEL                         : PD7
                      BS2                           : PA0
                      RESET disposition             : dedicated
                      RETRY pulse                   : SCK
                      Serial program mode           : yes
                      Parallel program mode         : yes
                      Timeout                       : 200
                      StabDelay                     : 100
                      CmdexeDelay                   : 25
                      SyncLoops                     : 32
                      PollIndex                     : 3
                      PollValue                     : 0x53
                      Memory Detail                 :

                                                        Block Poll               Page                       Polled
                        Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
                        ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
                        eeprom                 65    10     8    0 no       4096    8      0  9000  9000 0x00 0x00
                        flash                  65    10   256    0 yes    262144  256   1024  4500  4500 0x00 0x00
                        lfuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
                        hfuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
                        efuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
                        lock                    0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
                        calibration             0     0     0    0 no          1    1      0     0     0 0x00 0x00
                        signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00

                      Programmer Type : avrftdi
                      Description     : Amontec JTAGKey, JTAGKey-Tiny and JTAGKey2

avrdude_issue474.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.02s

avrdude_issue474.exe: Device signature = 0x1e9801 (probably m2560)
avrdude_issue474.exe: NOTE: "flash" memory has been specified, an erase cycle will be performed
                      To disable this feature, specify the -D option.
avrdude_issue474.exe: erasing chip
avrdude_issue474.exe: reading input file ".\atmega2560_0x11000_offset.hex"
avrdude_issue474.exe: writing flash (261414 bytes):

Writing | ################################################## | 100% 0.07s

avrdude_issue474.exe: 261414 bytes of flash written
avrdude_issue474.exe: verifying flash memory against .\atmega2560_0x11000_offset.hex:

Reading | ################################################## | 100% 0.07s

avrdude_issue474.exe: 261414 bytes of flash verified

avrdude_issue474.exe done.  Thank you.

PS C:\work\avr\avrdude_test\avrdude-7.0_bin64>  .\avrdude_issue474.exe -p m2560 -c dragon_isp -U flash:v:.\atmega2560_0x11000_offset.hex:i -v

avrdude_issue474.exe: Version 7.0-20220530 (feaa1c6)
[... 43 lines cut ...]
                        efuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
                        lock                    0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
                        calibration             0     0     0    0 no          1    1      0     0     0 0x00 0x00
                        signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00

                      Programmer Type : DRAGON_ISP
                      Description     : Atmel AVR Dragon in ISP mode
                      Vtarget         : 4.9 V
                      SCK period      : 1.00 us

avrdude_issue474.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.18s

avrdude_issue474.exe: Device signature = 0x1e9801 (probably m2560)
avrdude_issue474.exe: verifying flash memory against .\atmega2560_0x11000_offset.hex:

Reading | ################################################## | 100% 0.07s

avrdude_issue474.exe: 261414 bytes of flash verified

avrdude_issue474.exe done.  Thank you.

```
[atmega2560_0x11000_offset.hex.txt](https://github.com/avrdudes/avrdude/files/8870650/atmega2560_0x11000_offset.hex.txt)

mcuee, 2022-06-09:
The bootloader contents were not changed. The extra contents were generated by srec_cat.

```
$ srec_cat eepromtest.bin -offset 0x011000 -o eepromtest.srec
srec_cat: eepromtest.bin: 19: warning: no execution start address record

$ srec_cat eepromtest.srec  -o eepromtest_intel.hex -intel
srec_cat: eepromtest.srec: 19: warning: no execution start address record

$ cat eepromtest_intel.hex
:020000040001F9
:2010000054686520717569636B2062726F776E20666F78206A756D7073206F7665722074FE
:201020006865206C617A7920646F670A54686520717569636B2062726F776E20666F78206C
:201040006A756D7073206F76657220746865206C617A7920646F670A5468652071756963ED
:201060006B2062726F776E20666F78206A756D7073206F76657220746865206C617A7920C4
:20108000646F670A54686520717569636B2062726F776E20666F78206A756D7073206F76A5
:2010A000657220746865206C617A7920646F670A54686520717569636B2062726F776E20EE
:2010C000666F78206A756D7073206F76657220746865206C617A7920646F670A54686520B2
:2010E000717569636B2062726F776E20666F78206A756D7073206F76657220746865206C06
:20110000617A7920646F670A54686520717569636B2062726F776E20666F78206A756D7028
:2011200073206F76657220746865206C617A7920646F670A54686520717569636B20627269
:201140006F776E20666F78206A756D7073206F76657220746865206C617A7920646F670AFE
:2011600054686520717569636B2062726F776E20666F78206A756D7073206F76657220749D
:201180006865206C617A7920646F670A54686520717569636B2062726F776E20666F78200B
:2011A0006A756D7073206F76657220746865206C617A7920646F670A54686520717569638C
:2011C0006B2062726F776E20666F78206A756D7073206F76657220746865206C617A792063
:2011E000646F670A54686520717569636B2062726F776E20666F78206A756D7073206F7644
:00000001FF
```

mcuee, 2022-06-16:
> Both the old and patched code look fishy. Isn't the purpose of a lext command to set the upper 8 bits of a 24-bit address? To me, it makes little sense to trigger this based on the state of the [...]

Is the fix something like this? Or probably the command set is different.
https://github.com/avrdudes/avrdude/blob/main/src/stk500v2.c#L2281
```
// determine which command is to be used
  if (strcmp(m->desc, "flash") == 0) {
    addrshift = 1;
    commandbuf[0] = CMD_PROGRAM_FLASH_ISP;
    /*
     * If bit 31 is set, this indicates that the following read/write
     * operation will be performed on a memory that is larger than
     * 64KBytes. This is an indication to STK500 that a load extended
     * address must be executed.
     */
    if (m->op[AVR_OP_LOAD_EXT_ADDR] != NULL) {
      use_ext_addr = (1U << 31);
    }
  }
```

mcuee, 2022-06-17:
Apparently the test file exceeds 256KB and I need to redo the test with a proper hex file.
Ref: https://github.com/avrdudes/avrdude/issues/994#issuecomment-1158655821

mcuee, 2022-06-18:
The correct way to combine the two hex file.

```
mcuee@mcuees-Mac-mini bin % srec_cat entest_0x11000_offset.hex -intel stk500boot_v2_mega2560_eeprom_fix.hex -intel -o MergedHexFile.hex -intel
mcuee@mcuees-Mac-mini bin % cat MergedHexFile.hex 
:020000040001F9
:201000005468650A5468650A5468650A5468650A5468650A5468650A5468650A5468650A78
:201020005468650A5468650A5468650A5468650A5468650A5468650A5468650A5468650A58
:201040005468650A5468650A5468650A5468650A5468650A5468650A5468650A5468650A38
:201060005468650A5468650A5468650A5468650A5468650A5468650A5468650A5468650A18
:201080005468650A5468650A5468650A5468650A5468650A5468650A5468650A5468650AF8
:2010A0005468650A5468650A5468650A5468650A5468650A5468650A5468650A5468650AD8
:2010C0005468650A5468650A5468650A5468650A5468650A5468650A5468650A5468650AB8
:2010E0005468650A5468650A5468650A5468650A5468650A5468650A5468650A5468650A98
:201100005468650A5468650A5468650A5468650A5468650A5468650A5468650A5468650A77
:201120005468650A5468650A5468650A5468650A5468650A5468650A5468650A5468650A57
:201140005468650A5468650A5468650A5468650A5468650A5468650A5468650A5468650A37
:201160005468650A5468650A5468650A5468650A5468650A5468650A5468650A5468650A17
:201180005468650A5468650A5468650A5468650A5468650A5468650A5468650A5468650AF7
:2011A0005468650A5468650A5468650A5468650A5468650A5468650A5468650A5468650AD7
:2011C0005468650A5468650A5468650A5468650A5468650A5468650A5468650A5468650AB7
:2011E0005468650A5468650A5468650A5468650A5468650A5468650A5468650A5468650A97
:020000040003F7
:20E000000D9489F10D94B2F10D94B2F10D94B2F10D94B2F10D94B2F10D94B2F10D94B2F109
:20E020000D94B2F10D94B2F10D94B2F10D94B2F10D94B2F10D94B2F10D94B2F10D94B2F1C0
:20E040000D94B2F10D94B2F10D94B2F10D94B2F10D94B2F10D94B2F10D94B2F10D94B2F1A0
:20E060000D94B2F10D94B2F10D94B2F10D94B2F10D94B2F10D94B2F10D94B2F10D94B2F180
:20E080000D94B2F10D94B2F10D94B2F10D94B2F10D94B2F10D94B2F10D94B2F10D94B2F160
:20E0A0000D94B2F10D94B2F10D94B2F10D94B2F10D94B2F10D94B2F10D94B2F10D94B2F140
:20E0C0000D94B2F10D94B2F10D94B2F10D94B2F10D94B2F10D94B2F10D94B2F10D94B2F120
:20E0E0000D94B2F141546D656761323536300041726475696E6F206578706C6F726572205D
:20E1000073746B3530305632206279204D4C5300426F6F746C6F616465723E004875683FDC
:20E1200000436F6D70696C6564206F6E203D20004350552054797065202020203D20005F52
:20E140005F4156525F415243485F5F3D2000415652204C696243205665723D200047434305
:20E160002056657273696F6E203D20004350552049442020202020203D20004C6F772066B2
:20E18000757365202020203D20004869676820667573652020203D200045787420667573A6
:20E1A00065202020203D20004C6F636B20667573652020203D20004A756E202035203230E0
:20E1C000323200312E362E3700342E332E33005623202020414444522020206F7020636FC6
:20E1E00064652020202020696E737472756374696F6E2061646472202020496E7465727265
:20E20000757074006E6F20766563746F7200726A6D702020006A6D70200057686174207091
:20E220006F72743A00506F7274206E6F7420737570706F72746564004D7573742062652083
:20E2400061206C65747465720020005772697474696E672045450052656164696E6720456C
:20E26000450045452065727220636E743D00504F525400303D5A65726F2061646472003F78
:20E280003D43505520737461747300403D454550524F4D207465737400423D426C696E6B76
:20E2A000204C454400453D44756D7020454550524F4D00463D44756D7020464C41534800C2
:20E2C000483D48656C70004C3D4C69737420492F4F20506F72747300513D51756974005269
:20E2E0003D44756D702052414D00563D73686F7720696E7465727275707420566563746FC9
:20E30000727300593D506F727420626C696E6B002A0011241FBECFEFD1E2DEBFCDBF01E0F6
:20E320000CBF12E0A0E0B2E0E6E2FDEF03E00BBF02C007900D92A030B107D9F712E0A0E0EB
:20E34000B2E001C01D92AE30B107E1F70F9460F30D9491FE01E20EBF0FEF0DBF11241FBE9B
:20E360000D9460F30D9400F020E030E040ED57E005C0FA013197F1F72F5F3F4F2817390799
:20E38000C0F308959C01260F311DC901A0E0B0E02F5F3F4FABBFFC018791882361F0809389
:20E3A000C6008091C00086FFFCCF8091C00080648093C000EACF08958DE08093C600809141
:20E3C000C00086FFFCCF8091C00080648093C0008AE08093C6008091C00086FFFCCF809130
:20E3E000C00080648093C00008950F94C2F10F94DCF10895FC019081992359F09093C600AA
:20E400008091C00086FFFCCF8091C00080648093C0003196992379F70895282F982F92957E
:20E420009F70892F805D8A3308F0895F8093C6008091C00086FFFCCF8091C00080648093D9
:20E44000C000822F8F70982F905D9A3308F0995F9093C6008091C00086FFFCCF8091C00000
:20E4600080648093C00008959C01FB01853691051CF46330710594F0C90164E670E00F94BA
:20E4800032FE605D7F4F6093C6008091C00086FFFCCF8091C00080648093C0002B303105CE
:20E4A00014F43297B4F0C90164E670E00F9432FE6AE070E00F9432FE605D7F4F6093C600FF
:20E4C0008091C00086FFFCCF8091C00080648093C000C9016AE070E00F9432FEC0968093F3
:20E4E000C6008091C00086FFFCCF8091C00080648093C00008951F93182F8EE692EE60E0E3
:20E500000F94C2F11093C6008091C00086FFFCCF8091C00080648093C0000F94DCF11F9173
:20E5200008952F923F924F925F926F927F928F929F92AF92BF92CF92DF92EF92FF920F935E
:20E540001F93DF93CF93CDB7DEB762970FB6F894DEBF0FBECDBF382E622ECA01DB015C01E2
:20E560006D01772420E2222E2E010894411C511C8BC081E0A81680E0B80681E0C80680E094
:20E58000D80628F0C601AA27BB270F940DF2BB27AD2D9C2D8B2D0F940DF28A2D0F940DF231
:20E5A0002092C6008091C00086FFFCCF8091C00080648093C0009DE29093C6008091C00001
:20E5C00086FFFCCF8091C00080648093C0002092C6008091C00086FFFCCF8091C000806415
[... 166 lines cut ...]
:20FAA0003BC0C05EDE4F88819981AA81BB81C052D140B695A79597958795BC01F90161BDAF
:20FAC00072BD6F5F7F4FF89A80B5DA018D93AD013197A9F721503040C901A0E0B0E02F5F3A
:20FAE0003F4F880F991FAA1FBB1F0296A11DB11DC05EDE4FE880F9800A811B81C052D140EC
:20FB0000E80EF91E0A1F1B1FC05EDE4FE882F9820A831B83C052D1402D5F3F4FDA011C9254
:20FB200004C080EC8A8322E030E08BE18093C6008091C00086FFFCCF8091C0008064809348
[... cut ...]
:20FC60000D941AF497FB092E07260AD077FD04D02ED006D000201AF4709561957F4F089555
:20FC8000F6F7909581959F4F0895A1E21A2EAA1BBB1BFD010DC0AA1FBB1FEE1FFF1FA217F9
:20FCA000B307E407F50720F0A21BB30BE40BF50B661F771F881F991F1A9469F760957095A7
:20FCC000809590959B01AC01BD01CF010895AA1BBB1B51E107C0AA1FBB1FA617B70710F0C4
:20FCE000A61BB70B881F991F5A95A9F780959095BC01CD010895F999FECF92BD81BDF89AB3
:20FD0000992780B50895262FF999FECF1FBA92BD81BD20BD0FB6F894FA9AF99A0FBE01967E
:06FD20000895F894FFCFE6
:040000050003E00014
:00000001FF

```

stefanrueger, 2022-06-18:
> Is this the correct way to combine the two hex file? It does not seem to be correct

Yes it is, but why do you want to combine the files? You can use multiple `-U ...` options with different files on the command line.

@mariusgreuel is correct in his analysis. It is clear from the code that programming a file that starts at 0x11000 will not work. (no tests needed). There are examples how to treat sending the LOAD_EXT_ADDR command only when needed in stk500.c. 

If you want a real quick fix then always send the LOAD_EXT_ADDR command when it exists for the part a la
```
  if(use_lext_address && avrftdi_lext(pgm, p, m, addr/2) < 0)
    return -1;
```
That will do. As it is for paged read/writes and as the ATmega2560 has a page size of 256, programming 65k will draw 254 redundant load extended address commands at some 8(?) bytes traffic each, so programming might take 3% longer - you probably won't even notice.

mcuee, 2022-06-18:
@stefanrueger 
Thanks. I think my original hex file (simply combined the two files using Notepad++) is still valid for this test as it is using programmer and not bootloader. So  261414 bytes size is still okay and within the 256KB range (262144 bytes).

It is not valid for the #994 test since that was for the bootloader (262144 - 768 = 261376).

mcuee, 2022-06-18:
> > Is this the correct way to combine the two hex file? It does not seem to be correct
> 
> Yes it is, but why do you want to combine the files? You can use multiple `-U ...` options with different files on the command line.
> 
> @mariusgreuel is correct in his analysis. It is clear from the code that programming a file that starts at 0x11000 will not work. (no tests needed). There are examples in stk500.c how to treat sending the LOAD_EXT_ADDR command only when needed.
> 
> If you want a real quick fix then always send the LOAD_EXT_ADDR command when it exists for the part a la
> 
> ```
>   if(use_lext_address && avrftdi_lext(pgm, p, m, addr/2) < 0)
>     return -1;
> ```
> 
> That will do. As it is for paged read/writes and as the ATmega2560 has a page size of 256, programming 65k will draw 254 redundant load extended address commands at some 8(?) bytes traffic each, so programming might take 3% longer - you probably won't even notice.

@stefanrueger 
Indeed the quick fix seems to work fine.

```
PS C:\work\avr\avrdude_test\avrdude-7.0_bin64> .\avrdude_sr474.exe -p m2560 -c jtagkey -v -U .\stk500boot_v2_mega2560_arduino_official.hex -U .\entest_offset_0x11000.hex

avrdude_sr474.exe: Version 7.0-20220530 (feaa1c6)
                   Copyright (c) Brian Dean, http://www.bdmicro.com/
                   Copyright (c) Joerg Wunsch

                   System wide configuration file is "C:/work/avr/avrdude_test/avrdude-7.0_bin64/avrdude.conf"

                   Using Port                    : usb
                   Using Programmer              : jtagkey
                   AVR Part                      : ATmega2560
                   Chip Erase delay              : 9000 us
                   PAGEL                         : PD7
                   BS2                           : PA0
                   RESET disposition             : dedicated
                   RETRY pulse                   : SCK
                   Serial program mode           : yes
                   Parallel program mode         : yes
                   Timeout                       : 200
                   StabDelay                     : 100
                   CmdexeDelay                   : 25
                   SyncLoops                     : 32
                   PollIndex                     : 3
                   PollValue                     : 0x53
                   Memory Detail                 :

                                                     Block Poll               Page                       Polled
                     Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
                     ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
                     eeprom                 65    10     8    0 no       4096    8      0  9000  9000 0x00 0x00
                     flash                  65    10   256    0 yes    262144  256   1024  4500  4500 0x00 0x00
                     lfuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
                     hfuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
                     efuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
                     lock                    0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
                     calibration             0     0     0    0 no          1    1      0     0     0 0x00 0x00
                     signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00

                   Programmer Type : avrftdi
                   Description     : Amontec JTAGKey, JTAGKey-Tiny and JTAGKey2

avrdude_sr474.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.01s

avrdude_sr474.exe: Device signature = 0x1e9801 (probably m2560)
avrdude_sr474.exe: NOTE: "flash" memory has been specified, an erase cycle will be performed
                   To disable this feature, specify the -D option.
avrdude_sr474.exe: erasing chip
avrdude_sr474.exe: reading input file ".\stk500boot_v2_mega2560_arduino_official.hex"
avrdude_sr474.exe: input file .\stk500boot_v2_mega2560_arduino_official.hex auto detected as Intel Hex
[... 61 lines cut ...]
                     lock                    0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
                     calibration             0     0     0    0 no          1    1      0     0     0 0x00 0x00
                     signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00

                   Programmer Type : avrftdi
                   Description     : Amontec JTAGKey, JTAGKey-Tiny and JTAGKey2

avrdude_sr474.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.01s

avrdude_sr474.exe: Device signature = 0x1e9801 (probably m2560)
avrdude_sr474.exe: verifying flash memory against .\stk500boot_v2_mega2560_arduino_official.hex:

Reading | ################################################## | 100% 0.06s

avrdude_sr474.exe: 261406 bytes of flash verified
avrdude_sr474.exe: verifying flash memory against .\entest_offset_0x11000.hex:

Reading | ################################################## | 100% 0.00s

avrdude_sr474.exe: 70144 bytes of flash verified

avrdude_sr474.exe done.  Thank you.
```

stefanrueger, 2022-06-18:
Created PR #998 for this Issue #474

mcuee, 2022-06-19:
Test hex file creation:
```
$ srec_cat entest.bin -offset 0x021000 -o entest_offset_0x21000.srec
srec_cat: entest.bin: 19: warning: no execution start address record

$ srec_cat entest_offset_0x21000.srec  -o entest_offset_0x21000.hex -intel
srec_cat: entest_offset_0x21000.srec: 19: warning: no execution start address
    record

$ srec_cat entest_offset_0x11000.hex -intel -o - -c-arra dopp -Output_Word | tail -9 | head -4
const unsigned long dopp_termination = 0x00000000;
const unsigned long dopp_start       = 0x00011000;
const unsigned long dopp_finish      = 0x00011200;
const unsigned long dopp_length      = 0x00000200;

$ srec_cat entest_offset_0x21000.hex -intel -o - -c-arra dopp -Output_Word | tail -9 | head -4
const unsigned long dopp_termination = 0x00000000;
const unsigned long dopp_start       = 0x00021000;
const unsigned long dopp_finish      = 0x00021200;
const unsigned long dopp_length      = 0x00000200;

$ srec_cat entest.hex -intel -o - -c-arra dopp -Output_Word | tail -9 | head -4
const unsigned long dopp_termination = 0x00000000;
const unsigned long dopp_start       = 0x00000000;
const unsigned long dopp_finish      = 0x00000200;
const unsigned long dopp_length      = 0x00000200;
```
[entest_hexfiles.zip](https://github.com/avrdudes/avrdude/files/8934575/entest_hexfiles.zip)

mcuee, 2022-06-19:
> Created PR #998 for this Issue #474

@stefanrueger 

Great. This works perfectly.

```
PS C:\work\avr\avrdude_test\avrdude-7.0_bin64> .\avrdude_sr474v1.exe -c jtagkey -patmega2560 -D 
-U flash:w:.\hex\entest_offset_0x21000.hex:i

avrdude_sr474v1.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.01s

avrdude_sr474v1.exe: Device signature = 0x1e9801 (probably m2560)
avrdude_sr474v1.exe: reading input file ".\hex\entest_offset_0x21000.hex"
avrdude_sr474v1.exe: writing flash (135680 bytes):

Writing | ################################################## | 100% 0.00s

avrdude_sr474v1.exe: 135680 bytes of flash written
avrdude_sr474v1.exe: verifying flash memory against .\hex\entest_offset_0x21000.hex:

Reading | ################################################## | 100% 0.00s

avrdude_sr474v1.exe: 135680 bytes of flash verified

avrdude_sr474v1.exe done.  Thank you.

PS C:\work\avr\avrdude_test\avrdude-7.0_bin64> .\avrdude_sr474v1.exe -c jtagkey -patmega2560 -D 
-U flash:v:.\hex\entest_offset_0x21000.hex:i

avrdude_sr474v1.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.01s

avrdude_sr474v1.exe: Device signature = 0x1e9801 (probably m2560)
avrdude_sr474v1.exe: verifying flash memory against .\hex\entest_offset_0x21000.hex:

Reading | ################################################## | 100% 0.00s

avrdude_sr474v1.exe: 135680 bytes of flash verified

avrdude_sr474v1.exe done.  Thank you.

PS C:\work\avr\avrdude_test\avrdude-7.0_bin64> .\avrdude_sr474v1.exe -c usbasp -patmega2560 -D 
-U flash:v:.\hex\entest_offset_0x21000.hex:i

avrdude_sr474v1.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.01s

avrdude_sr474v1.exe: Device signature = 0x1e9801 (probably m2560)
avrdude_sr474v1.exe: verifying flash memory against .\hex\entest_offset_0x21000.hex:

Reading | ################################################## | 100% 0.00s

avrdude_sr474v1.exe: 135680 bytes of flash verified

avrdude_sr474v1.exe done.  Thank you.
```

stefanrueger, 2022-07-04:
Fixed in PR #998

closed 2022-07-04 by stefanrueger