avrdudes/avrdude #475: [bug #53745] ocdrev undocumented

avrdude-admin, 2021-12-11:
Cameron Tacklind <None>
Tue 24 Apr 2018 11:41:55 PM UTC

I've been looking at modifying avrdude.conf to add support for ATmega328PB. In doing so, I noticed the option `ocdrev` that is set to 0~4 on various chips. I haven't found any primary documentation that directly describes this option. I've seen references to OCD on AVR datasheets which I'm pretty sure refer to On-Chip Debugger but I haven't found an obvious version number there, which is what I'm guessing `ocdrev` controls. It would be nice if this could be documented further.

This issue was migrated from https://savannah.nongnu.org/bugs/?53745

avrdude-admin, 2021-12-11:
Joerg Wunsch <joerg_wunsch>
Wed 25 Apr 2018 05:20:48 AM UTC

This parameter is taken straight from the Atmel (Microchip) XML files. We do not have any further information or documentation about it, we simply pass it on to the Atmel tools.

MCUdude, 2022-01-07:
This issue can probably be closed

dl8dtl, 2022-01-08:
Yep, ocdrev is just an internal parameter in the Atmel tools. Nothing we could document.

closed 2022-01-08 by dl8dtl