avrdudes/avrdude #477: [bug #54159] Buffer overflow in usbtiny.c

avrdude-admin, 2021-12-11:
Marius Greuel <greuelm@mgtek.com>
Wed 20 Jun 2018 08:47:58 PM UTC
Programmer hardware: usbtiny
Device type: t85

There is a bug in usbtiny_paged_load() that causes a read/verify operation to overflow the flash memory buffer and crash the application.

The chunk being read is always 128, regardless of the actual bytes requested or bytes left.

See patch.txt for details.

[file #44398: patch.txt](https://savannah.nongnu.org/bugs/download.php?file_id=44398)

This issue was migrated from https://savannah.nongnu.org/bugs/?54159

avrdude-admin, 2021-12-11:
Kārlis Seņko <karliss>
Sun 16 Sep 2018 09:41:08 AM UTC

Looks like the same problem as [http://savannah.nongnu.org/bugs/?48776](https://savannah.nongnu.orghttp://savannah.nongnu.org/bugs/?48776)

avrdude-admin, 2021-12-11:
Marius Greuel <mariusgreuel>
Mon 03 Aug 2020 09:55:18 PM UTC

**[comment #1:](https://savannah.nongnu.org#comment1)**

> Looks like the same problem as [http://savannah.nongnu.org/bugs/?48776](https://savannah.nongnu.orghttp://savannah.nongnu.org/bugs/?48776)

You are correct, please mark this bug as a duplicate.

yegorich, 2022-01-09:
@mariusgreuel can this one be closed?

mariusgreuel, 2022-01-09:
Yes, thank you!

closed 2022-01-09 by mariusgreuel