avrdudes/avrdude #478: [bug #54289] attiny4/5/9/10 fuses reset option

avrdude-admin, 2021-12-11:
passnet <None>
Wed 11 Jul 2018 01:53:10 PM UTC
Programmer hardware: usbasp
Device type: attiny4/5/9/10

USBasp supports TPI with attiny4/5/9/10. Setting (writing to 0) fuse bits are also supported with avrdude. However resetting (writing 1) individual bits on these devices are not possible. It works but the bits remains unchanged:
avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.01s

avrdude: Device signature = 0x1e9003 (probably t10)
avrdude: reading input file "0xff"
avrdude: writing fuse (1 bytes):

Writing | ################################################## | 100% 0.08s

avrdude: 1 bytes of fuse written
avrdude: verifying fuse memory against 0xff:
avrdude: load data fuse data from input file 0xff:
avrdude: input file 0xff contains 1 bytes
avrdude: reading on-chip fuse data:

Reading | ################################################## | 100% 0.01s

avrdude: verifying ...
avrdude: verification error, first mismatch at byte 0x0000
0xfe != 0xff
avrdude: verification error; content mismatch

Datasheet on these tinys claims that to reset fuses you need to completely erase config section by following this procedure:

16.4.3.4. Erasing the Configuration Section
The algorithm for erasing the Configuration section is as follows:
1. Write the 0x14 (SECTION_ERASE) to the NVMCMD register
2. Start the erase operation by writing a dummy byte to the high byte of any word location inside the
configuration section
3. Wait until the NVMBSY bit has been cleared

Latest usbasp.c from svn trunk lacks this feature. That could be useful to add and pretty easy to implement.

This issue was migrated from https://savannah.nongnu.org/bugs/?54289

avrdude-admin, 2021-12-11:
Passnet <passnet>
Thu 12 Jul 2018 07:27:51 AM UTC

Ok, I was looking at the wrong place. Seems like this function are implemented in trunk (not the latest release), but not documented yet.

The '-x section_config' option with -e (erase) option should erase (reset) config section as I guess.

MCUdude, 2022-04-14:
I can confirm that you can write set/clear individual fuse bits using a USBasp programmer.

closed 2022-04-14 by MCUdude