avrdudes/avrdude #479: [bug #54387] minor compiler warnings

avrdude-admin, 2021-12-11:
sk8nfool <None>
Thu 26 Jul 2018 01:52:55 PM UTC
Programmer hardware: N/A
Device type: N/A

Compiling version 6.3 from source on Ubuntu 18.04 reports a redefinition warning that seems to be currently benign but may cause problems in the future. The error is

gcc -DHAVE_CONFIG_H -I. -I..  -DCONFIG_DIR=\"/home/braun/loca/etc\"  -Wall -Wno-pointer-sign -g -O2 -MT libavrdude_a-fileio.o -MD -MP -MF .deps/libavrdude_a-fileio.Tpo -c -o libavrdude_a-fileio.o `test -f 'fileio.c' || echo '../'`fileio.c
../fileio.c:37:0: warning: "EM_AVR32" redefined
#define EM_AVR32 0x18ad         /* inofficial */

In file included from /usr/include/libelf.h:36:0,
from ../fileio.c:33:
/usr/include/elf.h:323:0: note: this is the location of the previous definition
#define EM_AVR32 185 /* Amtel 32-bit microprocessor */

At least a comment should be added explaining why the redefinition is needed. If the intent was to actually redefine the constant a "#undef" should precede the redefinition.

A second warning occurs stating that misleading indentation is present. The warning is

gcc -DHAVE_CONFIG_H -I. -I..  -DCONFIG_DIR=\"/home/braun/local/etc\"  -Wall -Wno-pointer-sign -g -O2 -MT libavrdude
_a-avr.o -MD -MP -MF .deps/libavrdude_a-avr.Tpo -c -o libavrdude_a-avr.o `test -f 'avr.c' || echo '../'`avr.c
../avr.c: In function ‘avr_tpi_chip_erase’:
../avr.c:83:5: warning: this ‘while’ clause does not guard... [-Wmisleading-indentation]
while (avr_tpi_poll_nvmbsy(pgm));
^~~~~
../avr.c:85:3: note: ...this statement, but the latter is misleadingly indented as if it were guarded by the ‘while
’
err = pgm->cmd_tpi(pgm, cmd, sizeof(cmd), NULL, 0);
^~~

BTW - I created a ./build directory in which I performed the build steps.

[file #44628: excerpt.log](https://savannah.nongnu.org/bugs/download.php?file_id=44628)

This issue was migrated from https://savannah.nongnu.org/bugs/?54387

avrdude-admin, 2021-12-11:
Joerg Wunsch <joerg_wunsch>
Thu 26 Jul 2018 07:48:55 PM UTC

Frankly, I have no idea what is the definition for EM_AVR32 that could actually be found in AVR32 ELF files in the wild ... 0x18ad seems to have been historically used, but obviously never made it into any official state.

If there are any AVR32 users around, maybe they could comment on that.

avrdude-admin, 2021-12-11:
Dave Braun <sk8nfool>
Sat 28 Jul 2018 11:57:33 AM UTC

I'm confused. What is 0x18AD? The ELF package has a bunch of constants define for use in the e_machine field of the elf32 and elf64 headers. These are simple integers ranging from 0 (EM_NONE) to 247 (EM_BPF). These are set by whatever application creates the ELF formatted file. To my knowledge 0x18AD doesn't fit this definition!

There is a comment in the elf.h file suggesting using large random numbers for new architectures. But there is a definition for the 32 bit AVR architecture. Perhaps 0x18AD is left over from ancient history when the 32 bit AVR architecture was new? If that's the case then there is a need for two constants - the official value (185) and the unofficial value (0x18AD).

mariusgreuel, 2022-01-12:
No warnings with -Wall.

closed 2022-01-12 by mariusgreuel