avrdudes/avrdude #483: [bug #55123] Error in flashing / reading Eeprom on ATtiny45 under Windows

avrdude-admin, 2021-12-11:
hfuhrruhrurr <None>
Wed 28 Nov 2018 11:41:42 AM UTC
Programmer hardware: mySmartUsbLight
Device type: ATtiny45

Hi all,

when reading the EEPROM (256 byte) from the tiny, the first 128 byte are repeated twice in the downloaded file.

Trying to flash more than 128 bytes results in a verify error.

Using 5.10 instead, everything is fine.

Best regards,

Rainer

This issue was migrated from https://savannah.nongnu.org/bugs/?55123

avrdude-admin, 2021-12-11:
Joerg Wunsch <joerg_wunsch>
Mon 06 Dec 2021 10:20:20 PM UTC

Which "tiny"?

Best, also provide a full commandline.

Sorry, I have no idea about that mySmartUsbLight programmer, nor do I have access to it.

mcuee, 2022-05-27:
I think this should be closed if the reported never comes back for so long time.

BTW, mySmartUsbLight is probably the following programmer.
https://shop.myavr.com/index.php?sp=article.sp.php&artID=200006
> Supported protocols:
> AVR910/911
> STK500v2

closed 2022-05-29 by MCUdude