avrdudes/avrdude #488: [bug #55884] ft232r communication fails on programmed part if miso line toggles

avrdude-admin, 2021-12-11:
Klaus Syttkus <None>
Mon 11 Mar 2019 11:43:09 AM UTC
Programmer hardware: ft232r
Device type: t85

I use the miso line for UART emulation on my ATtiny, so that I can use the FT232R in ftdi_sio mode for communication.
In this configuration, reprogramming an ATtiny fails if the ATtiny produces continuous traffic right after RESET ("Device is not responding to program enable. Check connection").
This can be fixed by inserting a call to ft245r_drain in ft245r_program_enable:

--- /tmp/avrdude-6.3/ft245r.c   2014-09-15 21:01:09.000000000 +0200
+++ ft245r.c    2019-03-11 11:21:14.798593714 +0100
@@ -387,6 +387,7 @@
avr_set_bits(p->op[AVR_OP_PGM_ENABLE], cmd);

for(i = 0; i < 4; i++) {
+        ft245r_drain(pgm, 0);
ft245r_cmd(pgm, cmd, res);

if (res[p->pollindex-1] == p->pollvalue) return 0;

This issue was migrated from https://savannah.nongnu.org/bugs/?55884

avrdude-admin, 2021-12-11:
Joerg Wunsch <joerg_wunsch>
Sun 28 Nov 2021 10:34:37 PM UTC

Well, I tried to reproduce the problem. I wrote a firmware that toggles MOSI each 100 µs (on an ATmega16 but I don't think that matters much). I can connect to the target without any problems using both, the ft232r as well as the ft245r programmer.

When adding your suggested patch, the ft232r programmer still works, but the ft245r programmer breaks, and cannot connect anymore.

So seems we need some more analysis.

Can you upload a (simplified) firmware that triggers the bug, together with the avrdude commandline used?

mcuee, 2022-05-29:
As the OP never came back, and most likely this is invalid, I think this can be closed.

closed 2022-05-29 by MCUdude