avrdudes/avrdude #490: [bug #56686] Verification error at 0x0000, 0x00 != 0x0c

avrdude-admin, 2021-12-11:
Andrzej Pietrasiewicz <andrzejtp2010@gmail.com>
Sun 28 Jul 2019 11:48:26 AM UTC
Programmer hardware: linuxgpio
Device type: m2560

I'm using linuxgpio programmer with Raspberry Pi Zero acting as the programmer and programming an m2560. There is always the problem mentioned in the summary.

The problem happens after this invocation:

avrdude -c 2560 -p m2560 -U test.hex

But, on the other hand, this works:

avrdude -c 2560 -p m2560 -eu && avrdude -c 2560 -p m2560 -U test.hex -D

It seems that a sufficiently long delay between erasing and programming helps. I tried the below and indeed it helped:

diff --git a/main.c b/main.c
index ad50772..5dbcf30 100644
--- a/main.c
+++ b/main.c
@@ -1241,6 +1241,9 @@ int main(int argc, char * argv [])
}
exitrc = avr_chip_erase(pgm, p);
if(exitrc) goto main_exit;
+      pgm->close(pgm);
+      pgm->open(pgm, port);
+      pgm->initialize(pgm, p);
}
}

but it seems an overkill; ultimately only much smaller change in linuxgpio is enough to fix the problem:

diff --git a/linuxgpio.c b/linuxgpio.c
index b616311..b93142a 100644
--- a/linuxgpio.c
+++ b/linuxgpio.c
@@ -209,6 +209,7 @@ static int linuxgpio_highpulsepin(PROGRAMMER * pgm, int pinfunc)
return -1;

linuxgpio_setpin(pgm, pinfunc, 1);
+  bitbang_delay(20000);
linuxgpio_setpin(pgm, pinfunc, 0);

return 0;

BTW I'm using an SVN clone, but for the purpose of my experiments I also made the trunk/avrdude directory a git repository, hence the format of diffs.

This issue was migrated from https://savannah.nongnu.org/bugs/?56686

mcuee, 2022-06-04:
This is related to linuxgpio. Not so sure if it is still relevant though. Or maybe it is specific to Raspberry Pi Zero.

MCUdude, 2022-06-04:
I done have a pi zero to test with, but I do have a Pi model B with the came SoC.

My experience is that linuxgpio is a bit flakey (see #932), and that linuxspi is more reliable and much faster, but I'll give it a try when I'm back home.

mcuee, 2022-06-09:
I will try out setting up the test setup using linuxspi/linuxgpio as well since I have quite a few Raspberry Pi boards (Raspberry Pi 3B+ and 400) as well as Orange Pi boards.

mcuee, 2022-07-10:
I am not so sure if Pull Request #917 helps here or not -- I can not reproduce the issue using my Raspberry Pi 3B+, with PR #917. I will test out master git repo without #917 as well.

```
pi@raspberrypi:~/Desktop/build/avr/avrdude_pr917/build_linux/src $ sudo ./avrdude -C ./avrdude.conf 
-c linuxgpio -P GPIO -p m2560 -U flash:w:./blinky_m2560.hex:i

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e9801 (probably m2560)
avrdude: NOTE: "flash" memory has been specified, an erase cycle will be performed
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: reading input file "./blinky_m2560.hex"
avrdude: writing flash (261414 bytes):

Writing | ################################################## | 100% 56.31s

avrdude: 261414 bytes of flash written
avrdude: verifying flash memory against ./blinky_m2560.hex:

Reading | ################################################## | 100% 109.93s

avrdude: 261414 bytes of flash verified

avrdude done.  Thank you.

```

mcuee, 2022-07-10:
There is no issue using git repo master as well.

```
pi@raspberrypi:~/Desktop/build/avr/avrdude/build_linux/src $ sudo ./avrdude -C ./avrdude.conf -c linuxgpio 
-P GPIO -p m2560 -U flash:w:./blinky_m2560.hex:i

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e9801 (probably m2560)
avrdude: NOTE: "flash" memory has been specified, an erase cycle will be performed
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: reading input file "./blinky_m2560.hex"
avrdude: writing flash (261414 bytes):

Writing | ################################################## | 100% 57.70s

avrdude: 261414 bytes of flash written
avrdude: verifying flash memory against ./blinky_m2560.hex:

Reading | ################################################## | 100% 113.70s

avrdude: 261414 bytes of flash verified

avrdude done.  Thank you.

```

mcuee, 2022-07-10:
I will close this issue for now based on my testing.

Please re-open if you can reproduce the issue. Thanks.

closed 2022-07-10 by mcuee