avrdudes/avrdude #491: [bug #56970] avrdude exit code -1073741819 on Windows 10

avrdude-admin, 2021-12-11:
Joe <None>
Mon 30 Sep 2019 05:45:23 PM UTC
Programmer hardware: usbasp
Device type: ATtiny9

Hi,

avrdude returns -1073741819 in Windows 10 for ATtiny9.
It uploads the file correctly just that exit value is strange.

I've tested it for ATtiny10,5,4 and it works fine and it returns 0 for them.
Only for ATtiny9 it returns this code which is a problem when I call it from make.

d:\Solutions\Microcontrollers\Architecture\AVR\8-bit\AVRTiny\SimpleBlink>avrdude -c usbasp -p attiny9 -U simple_blink.bin

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.04s

avrdude: Device signature = 0x1e9008 (probably t9)
avrdude: NOTE: "flash" memory has been specified, an erase cycle will be performed
To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: reading input file "simple_blink.bin"
avrdude: input file simple_blink.bin auto detected as ELF
avrdude: writing flash (72 bytes):

Writing | ################################################## | 100% 0.49s

avrdude: 72 bytes of flash written
avrdude: verifying flash memory against simple_blink.bin:
avrdude: load data flash data from input file simple_blink.bin:
avrdude: input file simple_blink.bin auto detected as ELF
avrdude: input file simple_blink.bin contains 72 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 0.21s

avrdude: verifying ...
avrdude: 72 bytes of flash verified

avrdude done.  Thank you.

d:\Solutions\Microcontrollers\Architecture\AVR\8-bit\AVRTiny\SimpleBlink>echo %ERRORLEVEL%
-1073741819

Thank you for your hard work!

This issue was migrated from https://savannah.nongnu.org/bugs/?56970

avrdude-admin, 2021-12-11:
Joerg Wunsch <joerg_wunsch>
Mon 30 Sep 2019 07:48:43 PM UTC

Well, -1073741819 is 0xc0000005.

Microsoft says 0xc0000005 is the error code for an access violation.

Since the entire programming went fine, this probably happens during the final cleanup. It would be great if someone with a debugger under Windows could find out **where** that access violation actually happens.

avrdude-admin, 2021-12-11:
Jozef Chovan <chovanj>
Sat 05 Oct 2019 12:03:48 PM UTC

Hi,

is there any guide how to built it all for Windows or better question: How do you build the binary which is on your site for downloading? It would be very helpful to know for debugging it.

Thank you.

avrdude-admin, 2021-12-11:
Joerg Wunsch <joerg_wunsch>
Sat 05 Oct 2019 06:44:17 PM UTC

The script I am using is in the repository, under

tools/build-mingw32.sh

It's a cross-platform build using MinGW32 on my FreeBSD host.

Unfortunately, I cannot really tell you how to build under Windows natively. Never did that myself.

mariusgreuel, 2022-01-10:
0xC0000374 => A heap has been corrupted

Fixed with #784.

closed 2022-01-10 by mariusgreuel