avrdudes/avrdude #492: [bug #57153] OSX Catalina

avrdude-admin, 2021-12-11:
Gerrit Bosch <None>
Sat 02 Nov 2019 11:49:54 AM UTC

It is a cry for help rather than a bug i guess, as it is a Catalina headache.

In terminal is get this response:

user@mac /  % avrdude help
zsh: bad CPU type in executable: avrdude

What is the problem here?

This issue was migrated from https://savannah.nongnu.org/bugs/?57153

avrdude-admin, 2021-12-11:
Scott Price <scottprice>
Sun 03 Nov 2019 12:05:10 AM UTC

Are you using a 32 bit binary?  As I understand it Catalina is the first MacOS to be 64bit only.

avrdude-admin, 2021-12-11:
Gerrit <i2io>
Sun 03 Nov 2019 04:54:57 PM UTC

You may be right there. I am using CrossPack-AVR-20131216 including Avrdude version 6.0.1. Found no information whether it is 32 or 64 bit, might well be 32 bit.
Investigating the web i found a 6.3 version of this magnificent piece of software. Still have to figure out how to install, but there is nothing indicating conversion to 64bits. Leaves me wondering if it at all exists?

avrdude-admin, 2021-12-11:
Joerg Wunsch <joerg_wunsch>
Sun 03 Nov 2019 07:10:18 PM UTC

If all else fails, get a compiler, and build yourself.

mcuee, 2022-05-22:
This should be closed. There is no issue to build avrdude under macOS.

closed 2022-05-22 by MCUdude