avrdudes/avrdude #495: [bug #57339] [PATCH] improve safemode -V interaction, additional doc improvements

avrdude-admin, 2021-12-11:
Britton Kerin <britton.kerin@gmail.com>
Mon 02 Dec 2019 12:19:01 AM UTC
Programmer hardware: all
Device type: all

I'm attaching a series of patches that mainly address issues originally described in this email:

[https://lists.nongnu.org/archive/html/avrdude-dev/2019-11/msg00006.html](https://savannah.nongnu.orghttps://lists.nongnu.org/archive/html/avrdude-dev/2019-11/msg00006.html)

I fixed the -V interaction of safemode, but just documented the persistent effects of -B (because the originally envisioned fix looked hard and better -B documentation would have prevented the problems I had).

Later patches in this series fix some other small documentation issues, some of which have caused me confusion over the years.

One patch does a very slight code cleanup in main (all the comment typos and indent bugs I noticed while working on other things).

[file #47962: patches.tar.xz](https://savannah.nongnu.org/bugs/download.php?file_id=47962)

This issue was migrated from https://savannah.nongnu.org/bugs/?57339

mcuee, 2022-06-05:
Fixed the mailing list link:
https://lists.nongnu.org/archive/html/avrdude-dev/2019-11/msg00006.html

mcuee, 2022-06-05:
Patches included. 
```
Mode                 LastWriteTime         Length Name
----                 -------------         ------ ----
-a---           2/12/2019  8:13 am          10852 0001-Improve-safemode-V-interaction-and-warnings.patch
-a---           2/12/2019  8:13 am           3391 0002-Improve-B-documentation.patch
-a---           2/12/2019  8:13 am           2935 0003-Iimprove-D-and-e-documentation.patch
-a---           2/12/2019  8:13 am           1583 0004-Improve-description-of-Terminal-mode.patch
-a---           2/12/2019  8:13 am           1006 0005-Improve-description-of-b-option.patch
-a---           2/12/2019  8:13 am          11085 0006-Tiny-comment-and-wrogn-indentation-fixes.patch
-a---           2/12/2019  8:13 am           2229 0007-Document-surprising-arduino-bootloader-behavriod.patch
```

mcuee, 2023-03-26:
It seems to me the patches are no longer needed.

@stefanrueger and @MCUdude 
Please take a look at the patches to see if there are anything of value which we can adopt. Thanks.

MCUdude, 2023-03-27:
Suggested -b, -B, -D and -e documentation improvements in the patch provided by OP:

<details>
<summary>-B documentation</summary>

```diff
From eb6dadba334735b857ea70769b1d1b420ce898e0 Mon Sep 17 00:00:00 2001
From: Britton Leo Kerin <britton.kerin@gmail.com>
Date: Sat, 30 Nov 2019 12:30:27 -0900
Subject: [PATCH 2/7] Improve -B documentation

---
 avrdude.1 | 37 +++++++++++++++++++++++++------------
 1 file changed, 25 insertions(+), 12 deletions(-)

diff --git a/avrdude.1 b/avrdude.1
index 2258160..aa6d926 100644
--- a/avrdude.1
+++ b/avrdude.1
@@ -258,18 +258,28 @@ programmed in high-voltage serial mode.
 Override the RS-232 connection baud rate specified in the respective
 programmer's entry of the configuration file.
 .It Fl B Ar bitclock
-Specify the bit clock period for the JTAG interface or the ISP clock (JTAG ICE only).
-The value is a floating-point number in microseconds.
-Alternatively, the value might be suffixed with "Hz", "kHz", or "MHz",
-in order to specify the bit clock frequency, rather than a period.
-The default value of the JTAG ICE results in about 1 microsecond bit
-clock period, suitable for target MCUs running at 4 MHz clock and
-above.
-Unlike certain parameters in the STK500, the JTAG ICE resets all its
-parameters to default values when the programming software signs
-off from the ICE, so for MCUs running at lower clock speeds, this
-parameter must be specified on the command-line.
-You can use the 'default_bitclock' keyword in your
+WARNING: this option and its equivalent configuration file setting have a
+persistant effect on some programmers (e.g. stk500, avrispmkII): they remember
+the setting and will continue to use it until a different value is provided.
+.Nm avrdude
+will only change the existing setting if a -B option or default_bitclock
+configuration setting is provided.  This can result in persistant programming
+failure if the saved value is too fast for the target.  Other programmers (e.g.
+JTAG ICE) do not save the specified value, so for these programmers a
+non-default value must be specified with avrdude each time.
+.Pp
+Specify the bit clock period for the JTAG interface or the ISP clock for
+programmers which support one of these.  At least these programmer types
+probably correctly support this option: avrftdi, ft245r, jtag3, jtagmkI,
+jtagmkII, pickit2, stk500v2 (and deriviatives, e.g. AVRispmkII), usbasp, and
+usbtiny.  The value is a floating-point number in microseconds.  Alternatively,
+the value may be suffixed with "Hz", "kHz", or "MHz", in order to specify the
+bit clock frequency, rather than a period.  An ISP frequency of f_target/4 is
+supposed to work reliably, but for at least some clone programmers it actually
+doesn't and a lower ISP frequency setting (f_target/8) must be used.    For
+example, the default value of the JTAG ICE results in about 1 microsecond bit
+clock period, suitable for target MCUs running at 4 MHz clock and above.  You
+can use the 'default_bitclock' keyword in your
 .Pa ${HOME}/.avrduderc
 file to assign a default value to keep from having to specify this
 option on every invocation.
@@ -417,6 +427,9 @@ short delays.
 On Unix-style operating systems, the spin loop is initially calibrated
 against a system timer, so the number of microseconds might be rather
 realistic, assuming a constant system load while
+See also
+.Fl B
+which applies to a mostly different, larger set of programmers.
 .Nm
 is running.
 On Win32 operating systems, a preconfigured number of cycles per
[... 65 lines cut ...]
-from within terminal mode as well.
+individual memory cells.  On some programmers some settings of the programmer
+itself can be examined and changed from within terminal mode as well; see
+the 
+.Em Terminal mode
+section.
 .Ss Options
 In order to control all the different operation modi, a number of options
 need to be specified to
@@ -729,6 +730,9 @@ programmer itself.
 See below for a list of programmers accepting extended parameters.
 .El
 .Ss Terminal mode
+WARNING: some of these commands have a persistant effect on the programmer
+itself (e.g. sck period on the STK500 and derivatives).
+.Pp
 In this mode,
 .Nm
 only initializes communication with the MCU, and then awaits user
-- 
2.13.3

```

</details>

closed 2023-04-06 by stefanrueger