avrdudes/avrdude #500: [bug #58251] only odd bytes are programmed / programming only works in terminal mode.

avrdude-admin, 2021-12-11:
funkyluke <None>
Sun 26 Apr 2020 08:27:30 PM UTC
Programmer hardware: avrisp2
Device type: 8535

EEP Programming works:

avrdude -p8535 -cavrisp2 -Pusb -U eeprom:w:eep.hex --> verifies OK

Flash Programming:

root@luke-hp:~# cat short.hex
:200000000102030405060708091011121314151617181920212223242526272829303132E6
:00000001FF

root@luke-hp:~# avrdude -p8535 -cavrisp2 -Pusb -q -q -U flash:w:short.hex
avrdude: verification error, first mismatch at byte 0x0001
0xff != 0x02

root@luke-hp:~# cat short-readback.hex
:2000000001FF03FF05FF07FF09FF11FF13FF15FF17FF19FF21FF23FF25FF27FF29FF31FF84
:00000001F

--> Note that every other (even) byte is 0xFF (not programmed)
But:

root@luke-hp:~# avrdude -p8535 -cavrisp2 -Pusb -t

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.11s

avrdude: Device signature = 0x1e9303 (probably 8535)
avrdude> e

>>> e

avrdude: erasing chip
avrdude> write flash 0x00 0x11 0x22 0x33 0x44 0x55 0x66 0x77 0x88

>>> write flash 0x00 0x11 0x22 0x33 0x44 0x55 0x66 0x77 0x88

avrdude> dump flash 0x00 32

>>> dump flash 0x00 32

- - - - 11 22 33 44 55 66 77 88  ff ff ff ff ff ff ff ff  |."3DUfw.........|0010  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|

--> so writing in general seems to work, just not using -U and a hex file ?!

Thanks for looking into this!

This issue was migrated from https://savannah.nongnu.org/bugs/?58251

avrdude-admin, 2021-12-11:
Lukas Raschendorfer <funkyluke>
Mon 27 Apr 2020 05:49:18 AM UTC

Maybe a duplicate of **[bug #49082](https://savannah.nongnu.org/bugs/?49082)**, as it works with a pre-R1336 Release (6.1).

dl8dtl, 2022-03-30:
With the current code, the AVRISPmkII cannot write to the AT90S8535's flash at all: it runs into USB communication errors / timeouts, both in terminal as well as batch mode.
So we have to keep this one open.

mcuee, 2022-07-28:
Now the question is whether we really care about the old AT90S part like AT90S8535.

mcuee, 2022-09-07:
Looks like the following is the root cause.
https://github.com/avrdudes/avrdude/issues/509#issuecomment-1238655829

> This gets a bit complicated. The older AVRs that don't have paged flash used to have a single "write program memory" low-level ISP command where the low and high bytes of flash memory could be written individually. This was used up to v6.2 with the `SPI_MULTI` command (ISP passthrough). However, the STK500v2 protocol only offers `CMD_PROGRAM_FLASH_ISP` which assumes a prior `CMD_LOAD_ADDRESS` command to describe the address. For word-oriented flash, this address is a 16-bit word address though, so there's no option to individually address single (odd) bytes in flash that way. However, AVRDUDE can right now only distinguish between "paged mode" and "byte mode" - there is no 16-bit word mode at all inside.

mcuee, 2023-06-06:
Help wanted from the community. PR is welcome.

closed 2023-06-15 by stefanrueger