avrdudes/avrdude #507: [bug #59420] No PORT number

avrdude-admin, 2021-12-11:
Zisu Patriche <zisupatriche@yahoo.com>
Thu 05 Nov 2020 05:53:41 PM UTC
Programmer hardware: USBasp

error:
vrdude.exe: Version 6.3, compiled on Feb 17 2016 at 09:25:53
Copyright (c) 2000-2005 Brian Dean, [http://www.bdmicro.com/](https://savannah.nongnu.orghttp://www.bdmicro.com/)
Copyright (c) 2007-2014 Joerg Wunsch

System wide configuration file is "avrdude.conf"

Using Port                    : usb:040
Using Programmer              : usbasp-clone
avrdude.exe: error: could not find USB device with vid=0x16c0 pid=0x5dc

avrdude.exe done.  Thank you.

I tested any possibilities. My port on desktop (z640) and win10 is Port_#0011.Hub_#0003 How to translate in -P usb:???

This issue was migrated from https://savannah.nongnu.org/bugs/?59420

avrdude-admin, 2021-12-11:
Joerg Wunsch <joerg_wunsch>
Thu 05 Nov 2020 08:36:55 PM UTC

There is no option to specify a "port" number. USB works by walking through all attached devices, and look for the VID/PID combination you're seeing in the message. The first one found is then used.

(For some other programmers, an additional serial# could be provided, to disambiguate multiple attached programmers. USBasp doesn't offer a serial number feature though.)

In your case, there's simply no matchinng device found.

Since you're using Windows: make sure you have been setting up libusb correctly. Unlike non-Windows systems (where all devices with appropriate permissions are always found), on Windows, you explicitly have to assign any desired device for libusb usage.

MCUdude, 2022-01-09:
This issue can probably be closed. Either the programmer is faulty or have a different USB vid/pid, or Windows is using WinUSB instead of libusb to communicate with the programmer.

mariusgreuel, 2022-01-09:
The op most likely used libusb0.sys for USBasp, and the Arduino build of avrdude, which are not compatible. Fixed in v7.0

closed 2022-01-09 by mariusgreuel