avrdudes/avrdude #509: [bug #59667] verification error; content mismatch

avrdude-admin, 2021-12-11:
Johan Olin <johanolin@icloud.com>
Sat 12 Dec 2020 08:33:30 PM UTC
Programmer hardware: avrisp MKII
Device type: ATTiny15

avrdude -p t15 -c avrispmkii -U flash:w:file.hex
Results in the dreaded 'verification error; content mismatch' error ...
Seems like only the high byte gets written ?
Reading back gives <valid data><0xFF><valid data><0xFF> ...

Writing to eeprom seems to work fine !

And v6.3 works perfect with a Mega168.

Had to install avrdude 6.1 to program the Tiny15.  Tried 6.2 which also failed ...

This issue was migrated from https://savannah.nongnu.org/bugs/?59667

mcuee, 2022-06-04:
Hopefully someone has an ATtiny15 to test this out as this is reported as a regression (avrdude 6.1 works but not 6.2/6.3).

dl8dtl, 2022-06-20:
I have some exotic stuff like an ATtiny11 (HVSP only device :-o ) and ATtiny22L, but no '15 over here.

dl8dtl, 2022-06-20:
Found a Polish seller on ebay, ordered some.

dl8dtl, 2022-06-28:
I can basically confirm that the current stk500v2 code has issues with these old chips.
I do not get any programming communication at all with either an STK500v2 (+ USB adapter), or an AVRISPmkII – all result in USB receive timeouts.
Using an FT245R bitbang adapter, I can successfully program the ATtiny15 though.

mcuee, 2022-07-28:
This is probably a similar  case from AT90S part as ATtiny15L is obsolete too.

Ref: https://www.avrfreaks.net/forum/many-attinys-discontinued (post in 2007)

> The ATtiny11 and ATtiny12 were both replaced by the more feature-rich, pin-compatible ATtiny13.
> The ATtiny15 was replaced by the ATtiny25 in a special compatibility mode. The ATtiny25 is part of a new, more feature-rich sub-family including the ATtiny45 and ATtiny85.
> The ATtiny26 was replaced by the more feature-rich and pin-compatible ATtiny261/ATtiny461/ATtiny861 sub-family.

dl8dtl, 2022-09-05:
I just spent a bit more time on it: it works up to release 6.3. Starting with 6.4, it broke.

Unfortunately, it's a long period of time between both.

dl8dtl, 2022-09-05:
But the original report was "verification error", and it was for v6.3.

I didn't get any verification errors with v6.3, so we can close this.

(The effect I'm seeing are communication timeouts, that's something completely different.)

dl8dtl, 2022-09-05:
```
% ./avrdude -c stk500v2 -P /dev/cuaU3 -C avrdude.conf -p t15 -v

avrdude: Version 6.3, compiled on Sep  5 2022 at 21:54:20
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

…
         System wide configuration file is "avrdude.conf"
         User configuration file is "/home/joerg/.avrduderc"

         Programmer Type : STK500V2
         Description     : Atmel STK500 Version 2.x firmware
         Programmer Model: STK500
         Hardware Version: 2
         Firmware Version Master : 2.07
         Topcard         : Unknown
         Vtarget         : 3.0 V
         SCK period      : 17.4 us
         Varef           : 3.0 V
         Oscillator      : 3.686 MHz

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.01s

avrdude: Device signature = 0x1e9006 (probably t15)
avrdude: safemode: fuse reads as 5C

avrdude: safemode: fuse reads as 5C
avrdude: safemode: Fuses OK (E:FF, H:FF, L:FF)

avrdude done.  Thank you.

% ./avrdude -c stk500v2 -P /dev/cuaU3 -C avrdude.conf -p t15 -U test.elf

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.01s

avrdude: Device signature = 0x1e9006 (probably t15)
avrdude: NOTE: "flash" memory has been specified, an erase cycle will be performed
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: reading input file "test.elf"
avrdude: input file test.elf auto detected as ELF
avrdude: writing flash (1006 bytes):

Writing | ################################################## | 100% 14.41s

avrdude: 1006 bytes of flash written
avrdude: verifying flash memory against test.elf:
avrdude: load data flash data from input file test.elf:
avrdude: input file test.elf auto detected as ELF
avrdude: input file test.elf contains 1006 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 2.33s

avrdude: verifying ...
avrdude: verification error, first mismatch at byte 0x0001
         0xff != 0xc0
avrdude: verification error; content mismatch

avrdude: safemode: Fuses OK (E:FF, H:FF, L:FF)

avrdude done.  Thank you.
```

dl8dtl, 2022-09-05:
Sorry, I didn't look correctly.

Indeed, v6.3 causes that "verification error".

(With v6.4, the entire communication broke then.)

dl8dtl, 2022-09-05:
v6.2 is also broken.

Given that the major change in v6.2 in the stk500v2 implementation was the demission of the `SPI_MULTI` command, I guess that's where the breakage was introduced.

dl8dtl, 2022-09-06:
This gets a bit complicated.
The older AVRs that don't have paged flash used to have a single "write program memory" low-level ISP command where the low and high bytes of flash memory could be written individually. This was used up to v6.2 with the `SPI_MULTI` command (ISP passthrough).
However, the STK500v2 protocol only offers `CMD_PROGRAM_FLASH_ISP` which assumes a prior `CMD_LOAD_ADDRESS` command to describe the address. For word-oriented flash, this address is a 16-bit word address though, so there's no option to individually address single bytes in flash that way.
However, AVRDUDE can right now only distinguish between "paged mode" and "byte mode" - there is no 16-bit word mode at all inside.

mcuee, 2022-09-07:
Looks like #433 has the same root cause.
* #433

MCUdude, 2023-03-30:
Maybe @stefanrueger can have a look at this? I know I have an ATtiny15 somewhere, so I can help out with testing.

stefanrueger, 2023-03-30:
@MCUdude Is that still a problem? A *lot* has changed since September. I'd hazard a guess that it works now.

MCUdude, 2023-03-30:
I'm occasionally able to read the device signature and flash memory. However, I'm not able to write anything using an AVRISPmkII.

I've double-checked with a USBasp programmer, and the chip isn't defective.

(Here I've written 256 0xAA's using the USBasp programmer before dumping the flash contents).
```
$ ./avrdude -cavrispmkii -pattiny15 -Uflash:r:-:I

avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9006 (probably t15)
avrdude: reading flash memory ...

Reading | ################################################## | 100% 0.56 s 

avrdude: writing output file <stdout>
:20000000AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA0 // 00000> ********************************
:20002000AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA80 // 00020> ********************************
:20004000AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA60 // 00040> ********************************
:20006000AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA40 // 00060> ********************************
:20008000AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA20 // 00080> ********************************
:2000A000AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA00 // 000a0> ********************************
:2000C000AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAE0 // 000c0> ********************************
:2000E000AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAC0 // 000e0> ********************************
:00000001FF

avrdude done.  Thank you.
```

```
$ ./avrdude -cavrispmkii -pattiny15 -Uflash:w:0x55_256b.hex:i

avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9006 (probably t15)
avrdude: Note: flash memory has been specified, an erase cycle will be performed.
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: reading input file 0x55_256b.hex for flash
         with 256 bytes in 1 section within [0, 0xff]
avrdude: writing 256 bytes flash ...

Writing | -------------------------------------------------- | 0% 10.00 s 

avrdude error: unable to receive from USB
^C

```

stefanrueger, 2023-03-31:
So unable to reproduce the original problem right now (only high bytes are written) and unable to check whether numerous updates in the last 6 months have sorted this issue (never mind the *unable to receive from USB* which seems unrelated).

MCUdude, 2023-03-31:
Here are the output logs when uploading using -vvvv.

BTW, the schematic for the test board I'm using is this.
I've removed R7 and R11, and is powering the board via USB.
[MicroCore development board R2 schematic](https://d3s5r33r268y59.cloudfront.net/datasheets/19312/2020-02-10-21-50-27/MicroCore_development_board_R2_Schematic.pdf)

<details>
<summary>ATtiny15 AVRISPmkII upload attempt (-vvvv)</summary>

```
$ ./avrdude -cavrispmkii -pattiny15 -Uflash:w:test_avrdude_info_1k.hex:i -vvvv

avrdude: Version 7.1-20230108
         Copyright the AVRDUDE authors;
         see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

         System wide configuration file is /Users/hans/Downloads/avrdude/src/avrdude.conf
         User configuration file is /Users/hans/.avrduderc

         Using Port                    : usb
         Using Programmer              : avrispmkii
avrdude: usbdev_open(): found AVRISP mkII, serno: 000200124552
avrdude: sent: . [01] 
avrdude: recv: . [01] . [00] . [0a] A [41] V [56] R [52] I [49] S [53] P [50] _ [5f] M [4d] K [4b] 2 [32] 
avrdude: stk500v2_getsync(): found AVRISP mkII programmer
         AVR Part                      : ATtiny15
         Chip Erase delay              : 8200 us
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         Serial program mode           : yes
         Parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom                  4    20    64    0 no         64    1      0  8200  8200 0xff 0xff
                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           flash                   4    10   128    0 no       1024    1      0  4100  4100 0xff 0x00
                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           fuse                    0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           lock                    0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00
                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           calibration             0     0     0    0 no          1    1      0     0     0 0x00 0x00

         Programmer Type : STK500V2
         Description     : Atmel AVR ISP mkII
         Programmer Model: AVRISP mkII
avrdude: sent: . [03] . [90] 
avrdude: recv: . [03] . [00] . [01] 
avrdude: sent: . [03] . [91] 
[... 95 lines cut ...]
avrdude: sent: . [01] 
avrdude: usbdev_recv_frame(): usb_bulk_read(): Operation timed out
avrdude stk500v2_recv_mk2() [stk500v2.c:517] error: unable to receive from USB
avrdude: sent: . [01] 
avrdude: usbdev_recv_frame(): usb_bulk_read(): Operation timed out
avrdude stk500v2_recv_mk2() [stk500v2.c:517] error: unable to receive from USB
avrdude: sent: . [01] 
avrdude: usbdev_recv_frame(): usb_bulk_read(): Operation timed out
avrdude stk500v2_recv_mk2() [stk500v2.c:517] error: unable to receive from USB
avrdude: sent: . [01] 
avrdude: usbdev_recv_frame(): usb_bulk_read(): Operation timed out
avrdude stk500v2_recv_mk2() [stk500v2.c:517] error: unable to receive from USB
avrdude: sent: . [01] 
^C
```

</details>

<details>
<summary>ATtiny15 PICkit4 upload attempt (-vvvv)</summary>

The output log is too long, so I had to paste it over at gist.github.com instead:
https://gist.github.com/MCUdude/cea0bd6ead93485eae5260609f63dc60

</details>

mcuee, 2023-03-31:
@MCUdude 

I have placed an order for ATtiny15L myself but it will only arrive in about a month time (using sea shipment).

Just wondering if you can try avrdude 6.2 and 6.3 version to see if that works or not. As per the earlier comments by Joerg, version 6.2 is supposed to work but not 6.3.

mcuee, 2023-04-02:
> I have placed an order for two ATtiny15L SMD chips but they will only arrive in about a month time (using sea shipment).

@MCUdude 
Unfortunately the vendor cancelled my order as they do not have stock. Sorry I can not test this one.

mcuee, 2023-06-06:
Help wanted from the community. PR is welcome.

mcuee, 2023-06-11:
@MCUdude 

Please check whether terminal mode works or not.

Then please try #1386 using both CLI mode the terminal mode.

In #500, it is said that terminal mode may work.
* #500

closed 2023-06-15 by stefanrueger