avrdudes/avrdude #510: [bug #59801] macOS documentation improvement

avrdude-admin, 2021-12-11:
Jeroen <jeroendevries@runbox.eu>
Sat 02 Jan 2021 02:05:38 PM UTC
Programmer hardware: Arduino Uno

Dear avrdude team,

As a macOS user, I did find it difficult to find out what the port number needs to be when flashing an arduino uno device.

I would like to suggest a improvement in the documentation.

- A header specific for macOS on the page [https://www.nongnu.org/avrdude/user-manual/avrdude_18.html#Unix-Port-Names](https://savannah.nongnu.orghttps://www.nongnu.org/avrdude/user-manual/avrdude_18.html#Unix-Port-Names)

- Along with a instruction how to find the current port name. For example
ls /dev/tty.usb*

If it is too much of a hussle I would like to edit the page myself if the addition is wanted.

Thank you for the project,

Jeroen

This issue was migrated from https://savannah.nongnu.org/bugs/?59801

avrdude-admin, 2021-12-11:
Joerg Wunsch <joerg_wunsch>
Sun 27 Jun 2021 08:31:06 PM UTC

Yes, please submit a concrete wording for the change/addition in documentation. Does not need to be a SVN diff (but could), a paragraph or two in "prosa" are OK, so I can add that to the documentation. (Personally, I've got not much experience with MacOS myself.)

MCUdude, 2022-03-25:
@dl8dtl I can add a paragraph about macOS (as a PR) if you like to get this issue out of the way.

PR #908 (2022-03-31): Add MacOS serial/parallel port note
Closes #510