avrdudes/avrdude #512: [bug #60575] Permission denied on macOS Big Sur

avrdude-admin, 2021-12-11:
vicbarbu <None>
Mon 10 May 2021 06:13:32 PM UTC
Programmer hardware: ATMEL_ICE

Hi everyone!

Recently migrated to macOS Big Sur and can no longer use avrdude.

I am using ATMEL-ICE programmer and made the correct connections to an ATMega1284P (although it's pretty irrelevant in this scenario).

Running the following command:

sudo avrdude -c atmelice_isp -p m1284p -t -v

Gives:

avrdude: Version 6.3, compiled on Mar 13 2021 at 14:40:05
Copyright (c) 2000-2005 Brian Dean, [http://www.bdmicro.com/](https://savannah.nongnu.orghttp://www.bdmicro.com/)
Copyright (c) 2007-2014 Joerg Wunsch

System wide configuration file is "/opt/homebrew/Cellar/avrdude/6.3_1/etc/avrdude.conf"
User configuration file is "/Users/victorbarbu/.avrduderc"
User configuration file does not exist or is not a regular file, skipping

Using Port                    : usb
Using Programmer              : atmelice_isp
avrdude: usbdev_open(): Found Atmel-ICE CMSIS-DAP, serno: Jxxxxxxxxxxx
avrdude: usbdev_open(): error claiming interface 0: Permission denied
avrdude: usbdev_open(): no usable interface found
avrdude: jtag3_open_common(): Did not find any device matching VID 0x03eb and PID list: 0x2141

Spawning a virtual machine with Fedora ARM and connecting the programmer straight to that virtual machine makes avrdude work perfectly (by running the same command).

Is there any workaround to the issue (apart from using the virtual machine)?

Thanks!

This issue was migrated from https://savannah.nongnu.org/bugs/?60575

avrdude-admin, 2021-12-11:
Brian Good <briangood>
Tue 11 May 2021 08:10:23 PM UTC

Victor,
Yes, I spent a lot of time on this as a user and finally got it working (also MacOS Big Sur using the Atmel-ICE).  The fundamental problem is that when MacOS sees a device (like the Atmel-ICE), it claims it for itself.  The new programming model is that Apple wants you to use the Human Interface Device API (a subset of USB).  There was a previous workaround that used a zero code kernel extension file (KEXT) to grab the device before MacOS claimed it, but this no longer works in Big Sur.  I think the HID was is a better solution.  So here is what I did...

First, the avrdude support for HID USB devices requires a package called HIDAPI.  I got the code from GitHub (github.com/libusb/hidapi.git):

a) download with the command below into a directory (mine was ~/code) in Terminal

git clone git://github.com/libusb/hidapi.git

b) change to the newly created directory and run this command to get the development scripts

cd hidapi
./bootstrap

c) run the configure command to check that your system has the tools to compile

./configure

d) compile

make

e) run this command to install the newly compiled hidapi libraries.  This requires access to the target directories (/usr/local/include and /usr/local/bin), so it will require your MacOS password

sudo make install

Next, you need to get the latest avrdude code.  The most recent I could find was also on github:

a) Download - run this back from your (~/code) root code directory

git clone git://github.com/facchinm/avrdude.git

b)  Now comes the fix to the bug.  See my earlier post in the AVRDUDE mailing list for a line of code that needs to be added.  I am not sure if the post mentions it, but the file that needs to be edited is configure.ac

[https://lists.nongnu.org/archive/html/avrdude-dev/2021-03/msg00001.html](https://savannah.nongnu.orghttps://lists.nongnu.org/archive/html/avrdude-dev/2021-03/msg00001.html)

c) Once you are done, switch to the avrdude directory and run bootstrap

cd avrdude
./bootstrap

d)  Since you changed the configure file, you need make sure the configure script is updated, so run

./autoreconf

e)  Then run configure to check your tools again

./configure

f)  When that is done, look at the output from the configure command.  It should mention in there that it located HIDAPI.  If not, you have a problem from the first part.

g) Next, compile avrdude

make

h)  Finally, install it to /usr/local/include and /usr/local/bin

sudo make install

Hope this helps
Brian

avrdude-admin, 2021-12-11:
Victor Barbu <vicbarbu>
Tue 11 May 2021 08:15:41 PM UTC

Brian,

This sounds like an amazing investigation you've done. I will give it a try as soon as possible and update here.

Thank you.

Best regards,
Victor.

avrdude-admin, 2021-12-11:
Joerg Wunsch <joerg_wunsch>
Tue 11 May 2021 08:29:20 PM UTC

**[comment #1:](https://savannah.nongnu.org#comment1)**

> a) Download - run this back from your (~/code) root code directory
>
> git clone git://github.com/facchinm/avrdude.git

Well, this is not an official repository ...

> [https://lists.nongnu.org/archive/html/avrdude-dev/2021-03/msg00001.html](https://savannah.nongnu.orghttps://lists.nongnu.org/archive/html/avrdude-dev/2021-03/msg00001.html)

... and it turns out they use different code than the official AVR (subversion) repository, for whatever reason.

From your mail, it's not entirely clear to me what your change was. If it's the line with AC_DEFINE([HAVE_LIBHIDAPI]), well, this one has been in the official code base since the introduction of libhidapi 5 years ago.

The difference between the version you are quoting and the official repository is that the official version uses AC_SEARCH_LIBS, the version you quoted it wraps their own loop around PKG_CHECK_MODULES.

See line #170 here:

[http://svn.savannah.gnu.org/viewvc/avrdude/trunk/avrdude/configure.ac?revision=1447&view=markup](https://savannah.nongnu.orghttp://svn.savannah.gnu.org/viewvc/avrdude/trunk/avrdude/configure.ac?revision=1447&view=markup)

avrdude-admin, 2021-12-11:
Bernhard Nebel <nebel>
Sun 11 Jul 2021 08:48:50 AM UTC

Hi,

I tried Brian's recipe, and it works!

I was also curious whether it would work with the current codebase of avrdude. So, I checked out the SVN rep and compiled the code in trunk/avrdude. As mentioned by Joerg, the line to be added in the configure.ac file is already there. While this version can access the ICE, it is not able to access the MCU. I did a diff between the two versions, but it is more than just a few lines, so I gave up.

Here is the output:

mac: avrdude> ./avrdude -c atmelice_isp -p m328p -t -v

avrdude: Version 6.3-20171130
Copyright (c) 2000-2005 Brian Dean, [http://www.bdmicro.com/](https://savannah.nongnu.orghttp://www.bdmicro.com/)
Copyright (c) 2007-2014 Joerg Wunsch

System wide configuration file is "/usr/local/etc/avrdude.conf"

Using Port                    : usb
Using Programmer              : atmelice_isp
avrdude: Found CMSIS-DAP compliant device, using EDBG protocol
AVR Part                      : ATmega328P
Chip Erase delay              : 9000 us
PAGEL                         : PD7
BS2                           : PC2
RESET disposition             : dedicated
RETRY pulse                   : SCK
serial program mode           : yes
parallel program mode         : yes
Timeout                       : 200
StabDelay                     : 100
CmdexeDelay                   : 25
SyncLoops                     : 32
ByteDelay                     : 0
PollIndex                     : 3
PollValue                     : 0x53
Memory Detail                 :

Block Poll               Page                       Polled
Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
eeprom        65    20     4    0 no       1024    4      0  3600  3600 0xff 0xff
flash         65     6   128    0 yes     32768  128    256  4500  4500 0xff 0xff
lfuse          0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
hfuse          0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
efuse          0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
lock           0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
calibration    0     0     0    0 no          1    1      0     0     0 0x00 0x00
signature      0     0     0    0 no          3    1      0     0     0 0x00 0x00

Programmer Type : JTAG3_ISP
Description     : Atmel-ICE (ARM/AVR) in ISP mode
Vtarget         : 5.3 V
SCK period      : 8.00 us

avrdude: Short read, read only 0 out of 512 bytes
avrdude: jtag3_edbg_recv(): Unexpected response 0x00
avrdude: stk500v2_jtag3_recv(): error in jtagmkII_recv()
avrdude: AVR device initialized and ready to accept instructions

Reading |                                                    | 0% 0.00savrdude: jtag3_edbg_send(): Unexpected response 0x81, 0x11
avrdude: jtag3_edbg_recv(): Unexpected response 0x80
avrdude: stk500v2_jtag3_recv(): error in jtagmkII_recv()
avrdude: Short read, read only 0 out of 512 bytes
avrdude: jtag3_edbg_send(): Unexpected response 0x40, 0xf2
avrdude: Short read, read only 0 out of 512 bytes
avrdude: jtag3_edbg_recv(): Unexpected response 0x00
avrdude: stk500v2_jtag3_recv(): error in jtagmkII_recv()
Reading | #################                                  | 33% 0.85savrdude: Short read, read only 0 out of 512 bytes
avrdude: jtag3_edbg_send(): Unexpected response 0x00, 0x00
avrdude: Short read, read only 0 out of 512 bytes
avrdude: jtag3_edbg_recv(): Unexpected response 0x00
avrdude: stk500v2_jtag3_recv(): error in jtagmkII_recv()
Reading | ################################################## | 100% 1.45s

avrdude: Device signature = 0x303030
avrdude: Expected signature for ATmega328P is 1E 95 0F
Double check chip, or use -F to override this check.
avrdude: Short read, read only 0 out of 512 bytes
avrdude: jtag3_edbg_send(): Unexpected response 0x0a, 0x0a
avrdude: Short read, read only 0 out of 512 bytes
avrdude: jtag3_edbg_recv(): Unexpected response 0x00
avrdude: stk500v2_jtag3_recv(): error in jtagmkII_recv()
avrdude: Short read, read only 0 out of 512 bytes
avrdude: jtag3_edbg_send(): Unexpected response 0xf9, 0x2e
avrdude: Short read, read only 0 out of 512 bytes
avrdude: jtag3_edbg_recv(): Unexpected response 0x00
avrdude: Short read, read only 0 out of 512 bytes
avrdude: jtag3_edbg_send(): Unexpected response 0x32, 0x00
avrdude: Short read, read only 0 out of 512 bytes
avrdude: jtag3_edbg_recv(): Unexpected response 0x00
avrdude: Short read, read only 0 out of 512 bytes
avrdude: jtag3_edbg_signoff(): failed to read from serial port (0)

avrdude done.  Thank you.

mac: avrdude>

avrdude-admin, 2021-12-11:
Bernhard Nebel <nebel>
Mon 12 Jul 2021 07:21:43 PM UTC

**[comment #4:](https://savannah.nongnu.org#comment4)**

> I tried Brian's recipe, and it works!
>
> I was also curious whether it would work with the current codebase of avrdude. So, I checked out the SVN rep and compiled the code in trunk/avrdude. As [...]

I tried it again today, and now the original code in the trunk works. No idea why I got the strange error yesterday.

Best,
Bernhard

avrdude-admin, 2021-12-11:
Bernhard Nebel <nebel>
Mon 12 Jul 2021 10:45:13 PM UTC

In order to simplify life for everybody, I have set up a brew formula. So, assuming you have already installed homebrew, you only have to type:

brew uninstall avrdude

which will remove your current avrdude homebrew binary. And then type:

brew install --HEAD felias-fogg/arduino/avrdude

This will install the official version from the avrdude trunk (after downloading the hidapi package). In other words, you now should have a avrdude version located in /usr/local/bin that can connect to the Atmel-ICE. You may want to symlink the Arduino version of avrdude to this binary.

Cheers,
Bernhard

avrdude-admin, 2021-12-11:
Joerg Wunsch <joerg_wunsch>
Sat 06 Nov 2021 10:23:00 PM UTC

If I read that correctly, a new official release might remedy this.

avrdude-admin, 2021-12-16:
mcudude <mcudude>
Tue 14 Dec 2021 01:14:58 PM UTC

I have the same issue on my mac, and I'm running MacOS 10.14 (Mojave). I'm using an Xplained Mini UPDI programmer (ATmega32U4 based).

If I use the Avrdude version that's bundled with Arduino IDE ([https://github.com/facchinm/avrdude](https://savannah.nongnu.orghttps://github.com/facchinm/avrdude)) it works fine, but if I build my own Avrdude version from this repo I get the same error as Victor get.

I stumbled across this issue when helping out Dawid with issue #61624. It would be nice if this was fixed in the next release (6.4). Let me know if you need someone to help out with testing!

avrdude-admin, 2021-12-16:
Joerg Wunsch <joerg_wunsch>
Tue 14 Dec 2021 01:39:51 PM UTC

Strange.

I've meanwhile got a small Macbook, it shipped to me with 12.0.1 (Monterey). Just tested the 6.4 release candidate yesterday, and all in all, MacOS is the OS causing the least trouble in all respects (Linux and FreeBSD need device permission tweaking, Windows needs a lot of driver teasing and installation).

I built the prerequisites (autoconf/automake, libraries) with macports rather than brew, but that's not supposed to affect the permissions on a self-built avrdude binary in any way.

For those that can observe it, please try to dtruss it and see on which device it happens.

avrdude-admin, 2021-12-16:
Joerg Wunsch <joerg_wunsch>
Tue 14 Dec 2021 09:51:33 PM UTC

After quite some discussion in the mailing list, the key point is that you need to make sure the hidapi library has been found by the configure script, using the appropriate CPPFLAGS (for -I) and LDFLAGS (for -L) options to ./configure.

If hidapi has not been included correctly, jtag3.c falls back to the standard libusb implementation which can cause the "permission denied" issue that triggered this bug report.

mcuee, 2022-05-22:
In this case, this can probably closed and then documentation can be updated to recommending to use hidapi instead of libusb for HID devices. There is in general no way to detach the kernel HID driver under newer versions of macOS.
Ref: https://github.com/libusb/libusb/wiki/FAQ#Does_libusb_support_USB_HID_devices

mcuee, 2022-05-29:
@mariusgreuel @dl8dtl 
Just wondering if you think this can be closed. I tend to think this can be closed.

mcuee, 2022-06-04:
@MCUdude 
Maybe you can take a look to see if this can be closed or not. Thanks.

MCUdude, 2022-06-04:
Sorry, but I'm currently running an older macOS version (10.14, Mojave). I can give it a try, just for reference, but I doubt I'll have any issues getting it to work with Avrdude.

dl8dtl, 2022-06-04:
Can be closed, I have no troubles running it on MacOS 12.3 (Monterey).

closed 2022-06-04 by dl8dtl