avrdudes/avrdude #515: [bug #61169] Bad EEPROM read on mega2560 R3 Arduino board

avrdude-admin, 2021-12-11:
Shane <santner@gmail.com>
Thu 16 Sep 2021 02:53:37 PM UTC
Programmer hardware: wiring/COM

I have a mega2560 R3 arduino board. I created a simple sketch from the Arduino IDE from file->Examples->EEPROM->eeprom_clear and then modified it to write 0x00112233445566778899aabbccddeeff repeating to the eeprom memory. After writing out these values and then using avrdude to read them back I get 0x00112233445566770011223344556677 instead. I have a custom programmer that read back the expected values correctly.

The avrdude command I am using for reading out the eeprom is:
avrdude -p atmega2560 -c wiring -P COM4 -b115200 -U eeprom:r:dump.bin

This issue was migrated from https://savannah.nongnu.org/bugs/?61169

MCUdude, 2022-03-20:
I can confirm that reading from EEPROM is troublesome using the "Wiring" bootloader the Arduino Mega 2560 ships with.

My tests shows that writing works fine, but it1s not able to read properly:

Original hex file written to EEPROM:
<details>

```
:20000000EFCDAB8967452301EFCDAB8967452301EFCDAB8967452301EFCDAB8967452301E0
:20002000EFCDAB8967452301EFCDAB8967452301EFCDAB8967452301EFCDAB8967452301C0
:20004000EFCDAB8967452301EFCDAB8967452301EFCDAB8967452301EFCDAB8967452301A0
:20006000EFCDAB8967452301EFCDAB8967452301EFCDAB8967452301EFCDAB896745230180
:20008000EFCDAB8967452301EFCDAB8967452301EFCDAB8967452301EFCDAB896745230160
:2000A000EFCDAB8967452301EFCDAB8967452301EFCDAB8967452301EFCDAB896745230140
:2000C000EFCDAB8967452301EFCDAB8967452301EFCDAB8967452301EFCDAB896745230120
:2000E000EFCDAB8967452301EFCDAB8967452301EFCDAB8967452301EFCDAB896745230100
:20010000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
:20012000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFDF
:20014000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFBF
:20016000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF9F
:20018000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF7F
:2001A000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF5F
:2001C000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF3F
:2001E000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF1F
:20020000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFE
:20022000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFDE
:20024000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFBE
:20026000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF9E
:20028000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF7E
:2002A000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF5E
:2002C000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF3E
:2002E000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF1E
:20030000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFD
:20032000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFDD
:20034000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFBD
:20036000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF9D
:20038000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF7D
:2003A000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF5D
:2003C000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF3D
:2003E000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF1D
:20040000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFC
:20042000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFDC
:20044000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFBC
:20046000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF9C
:20048000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF7C
:2004A000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF5C
:2004C000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF3C
:2004E000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF1C
:20050000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFB
:20052000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFDB
:20054000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFBB
:20056000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF9B
:20058000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF7B
:2005A000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF5B
:2005C000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF3B
:2005E000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF1B
:20060000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFA
:20062000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFDA
:20064000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFBA
:20066000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF9A
:20068000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF7A
:2006A000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF5A
:2006C000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF3A
:2006E000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF1A
:20070000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF9
:20072000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFD9
:20074000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFB9
:20076000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF99
:20078000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF79
:2007A000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF59
[... 313 lines cut ...]
:200DE000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF13
:200E0000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF2
:200E2000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFD2
:200E4000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFB2
:200E6000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF92
:200E8000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF72
[... cut ...]
:200FA000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF51
:200FC000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF31
:200FE000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF11
:00000001FF

avrdude done.  Thank you.
```

</details>

I've read the EEPROM contents with a dedicated programmer, and I can confirm that the contents are correct, it's just the bootloader that can't read it properly. Not sure if this is cause by the bootloader or Avrdude.

MCUdude, 2022-05-29:
I'm very confident this is an issue with the stk500v2 Arduino bootloader, and not Avrdude. Please correct me of I'm wrong about this.

**USBasp write to EEPROM:**

<details>

```
$ echo "write eeprom 0x00 1 2 3 4 5 6 7 8 9 10 11 12 13 14 15 16" | ./avrdude -c usbasp -vvv -p atmega2560 -t 
executable_abspath = /Users/hans/Downloads/avrdude/src/avrdude
executable_abspath_len = 41
executable_dirpath = /Users/hans/Downloads/avrdude/src
executable_dirpath_len = 33
sys_config = /Users/hans/Downloads/avrdude/src/avrdude.conf
sys_config_found = true

avrdude: Version 6.99-20211218
         Copyright (c) Brian Dean, http://www.bdmicro.com/
         Copyright (c) Joerg Wunsch

         System wide configuration file is "/Users/hans/Downloads/avrdude/src/avrdude.conf"
         User configuration file is "/Users/hans/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : usb
         Using Programmer              : usbasp
avrdude: usbasp_open("usb")
avrdude: seen device from vendor ->www.fischl.de<-
avrdude: seen product ->USBasp<-
         AVR Part                      : ATmega2560
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PA0
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         Serial program mode           : yes
         Parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom                 65    10     8    0 no       4096    8      0  9000  9000 0x00 0x00
                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           flash                  65    10   256    0 yes    262144  256   1024  4500  4500 0x00 0x00
                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           lfuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           hfuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           efuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           lock                    0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
[... 701 lines cut ...]
Reading | ######################################             | 76% 0.06savrdude: stk500isp_read_byte(.., eeprom, 0x32, ...)
Reading | #######################################            | 78% 0.06savrdude: stk500isp_read_byte(.., eeprom, 0x33, ...)
Reading | ########################################           | 79% 0.06savrdude: stk500isp_read_byte(.., eeprom, 0x34, ...)
Reading | #########################################          | 81% 0.06savrdude: stk500isp_read_byte(.., eeprom, 0x35, ...)
Reading | #########################################          | 82% 0.06savrdude: stk500isp_read_byte(.., eeprom, 0x36, ...)
Reading | ##########################################         | 84% 0.06savrdude: stk500isp_read_byte(.., eeprom, 0x37, ...)
Reading | ###########################################        | 85% 0.06savrdude: stk500isp_read_byte(.., eeprom, 0x38, ...)
Reading | ############################################       | 87% 0.06savrdude: stk500isp_read_byte(.., eeprom, 0x39, ...)
[... cut ...]
Reading | ################################################   | 96% 0.06savrdude: stk500isp_read_byte(.., eeprom, 0x3f, ...)
Reading | ################################################## | 100% 0.06s

0000  01 02 03 04 05 06 07 08  ff ff ff ff ff ff ff ff  |................|
0010  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0020  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0030  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|

avrdude done.  Thank you.
```

</details>

closed 2022-05-29 by MCUdude

mcuee, 2022-05-30:
I tend to agree that this is most likely an issue with the bootloader and not avrdude.

@stefanrueger has raised quite some issues related to various bootloaders with EEPROM.
Some examples:
* https://github.com/avrdudes/avrdude/discussions/940
* https://github.com/avrdudes/avrdude/issues/967
* https://github.com/avrdudes/avrdude/discussions/970

mcuee, 2022-06-03:
I can reproduce the issue as well.

I use AVR Dragon to write the EEPROM as above and then read it back using AVR Dragon to confirm the contents are correct.

Then I use the Arduino Mega2560 (clone version with CH340) bootloader to read it back to reproduce the issue.

```
PS C:\work\avr\avrdude_test\avrdude-v7.0-windows-x64> echo "read eeprom 0 0x40" | ./avrdude -c wiring -P COM9 -vvv -p at
mega2560 -t
executable_abspath = C:/work/avr/avrdude_test/avrdude-v7.0-windows-x64/avrdude.exe
executable_abspath_len = 61
executable_dirpath = C:/work/avr/avrdude_test/avrdude-v7.0-windows-x64
executable_dirpath_len = 49
sys_config = C:/work/avr/avrdude_test/avrdude-v7.0-windows-x64/avrdude.conf
sys_config_found = true

avrdude.exe: Version 7.0
             Copyright (c) Brian Dean, http://www.bdmicro.com/
             Copyright (c) Joerg Wunsch

             System wide configuration file is "C:/work/avr/avrdude_test/avrdude-v7.0-windows-x64/avrdude.conf"

             Using Port                    : COM9
             Using Programmer              : wiring
avrdude.exe: wiring_open(): releasing DTR/RTS
avrdude.exe: wiring_open(): asserting DTR/RTS
avrdude.exe: ser_recv(): programmer is not responding
avrdude.exe: stk500v2_ReceiveMessage(): timeout
PS C:\work\avr\avrdude_test\avrdude-v7.0-windows-x64> echo "read eeprom 0 0x40" | ./avrdude -c wiring -P COM9 -vvv -p atmega2560 -t
executable_abspath = C:/work/avr/avrdude_test/avrdude-v7.0-windows-x64/avrdude.exe
executable_abspath_len = 61
executable_dirpath = C:/work/avr/avrdude_test/avrdude-v7.0-windows-x64
executable_dirpath_len = 49
sys_config = C:/work/avr/avrdude_test/avrdude-v7.0-windows-x64/avrdude.conf
sys_config_found = true

avrdude.exe: Version 7.0
             Copyright (c) Brian Dean, http://www.bdmicro.com/
             Copyright (c) Joerg Wunsch

             System wide configuration file is "C:/work/avr/avrdude_test/avrdude-v7.0-windows-x64/avrdude.conf"

             Using Port                    : COM9
             Using Programmer              : wiring
avrdude.exe: wiring_open(): releasing DTR/RTS
avrdude.exe: wiring_open(): asserting DTR/RTS
avrdude.exe: stk500v2_getsync(): found AVRISP programmer
             AVR Part                      : ATmega2560
             Chip Erase delay              : 9000 us
             PAGEL                         : PD7
             BS2                           : PA0
             RESET disposition             : dedicated
             RETRY pulse                   : SCK
             Serial program mode           : yes
             Parallel program mode         : yes
             Timeout                       : 200
             StabDelay                     : 100
             CmdexeDelay                   : 25
             SyncLoops                     : 32
             PollIndex                     : 3
             PollValue                     : 0x53
             Memory Detail                 :

                                               Block Poll               Page                       Polled
               Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
               ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
               eeprom                 65    10     8    0 no       4096    8      0  9000  9000 0x00 0x00
                                               Block Poll               Page                       Polled
               Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
               ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
[... 94 lines cut ...]
Reading | ######################################             | 76% 0.07savrdude.exe: stk500isp_read_byte(.., eeprom, 0x32, ...)
Reading | #######################################            | 78% 0.07savrdude.exe: stk500isp_read_byte(.., eeprom, 0x33, ...)
Reading | ########################################           | 79% 0.07savrdude.exe: stk500isp_read_byte(.., eeprom, 0x34, ...)
Reading | #########################################          | 81% 0.07savrdude.exe: stk500isp_read_byte(.., eeprom, 0x35, ...)
Reading | #########################################          | 82% 0.07savrdude.exe: stk500isp_read_byte(.., eeprom, 0x36, ...)
Reading | ##########################################         | 84% 0.07savrdude.exe: stk500isp_read_byte(.., eeprom, 0x37, ...)
Reading | ###########################################        | 85% 0.07savrdude.exe: stk500isp_read_byte(.., eeprom, 0x38, ...)
Reading | ############################################       | 87% 0.08savrdude.exe: stk500isp_read_byte(.., eeprom, 0x39, ...)
Reading | #############################################      | 89% 0.08savrdude.exe: stk500isp_read_byte(.., eeprom, 0x3a, ...)
Reading | #############################################      | 90% 0.08savrdude.exe: stk500isp_read_byte(.., eeprom, 0x3b, ...)
Reading | ##############################################     | 92% 0.08savrdude.exe: stk500isp_read_byte(.., eeprom, 0x3c, ...)
Reading | ###############################################    | 93% 0.08savrdude.exe: stk500isp_read_byte(.., eeprom, 0x3d, ...)
Reading | ################################################   | 95% 0.08savrdude.exe: stk500isp_read_byte(.., eeprom, 0x3e, ...)
Reading | ################################################   | 96% 0.08savrdude.exe: stk500isp_read_byte(.., eeprom, 0x3f, ...)
Reading | ################################################## | 100% 0.08s

0000  01 02 03 04 05 06 07 08  ff ff ff ff ff ff ff ff  |................|
0010  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0020  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0030  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|

avrdude>
avrdude.exe done.  Thank you.

```

mcuee, 2022-06-03:
So for now the workaround is to use a programmer and not the bootloader to read EEPROM.

mcuee, 2022-06-03:
Looks like this is a long standing problem of the Arduino Mega2560 bootloader.
* https://github.com/arduino/ArduinoCore-avr/issues/23

Potential fix:
* https://github.com/arduino/ArduinoCore-avr/pull/24

mcuee, 2022-06-04:
https://github.com/arduino/ArduinoCore-avr/pull/24#issuecomment-441559730

Answered by Arduino project's facchinm
> We normally update bootloaders on the repo only when the boards are being factory flashed with that bootloader.
> The validation phase usually requires quite a long time and it's rather unusual to "live patch" a board except for special cases (like Nano's WDT problem).
> Moreover, I'd say that we'd probably switch to optiboot for the 2560 if we change the BL in production.

mcuee, 2022-06-04:
It seems the bootloader has problem with writing EEPROM as well.

```
PS C:\work\avr\avrdude_test\avrdude-v7.0-windows-x64> echo "write eeprom 0x00 1 2 3 4 5 6 7 8 9 10 11 12 13 14 15 16" | ./avrdude -c wiring -P COM9 -vvv -p atmega2560 -t
executable_abspath = C:/work/avr/avrdude_test/avrdude-v7.0-windows-x64/avrdude.exe
executable_abspath_len = 61
executable_dirpath = C:/work/avr/avrdude_test/avrdude-v7.0-windows-x64
executable_dirpath_len = 49
sys_config = C:/work/avr/avrdude_test/avrdude-v7.0-windows-x64/avrdude.conf
sys_config_found = true

avrdude.exe: Version 7.0
             Copyright (c) Brian Dean, http://www.bdmicro.com/
             Copyright (c) Joerg Wunsch

             System wide configuration file is "C:/work/avr/avrdude_test/avrdude-v7.0-windows-x64/avrdude.conf"

             Using Port                    : COM9
             Using Programmer              : wiring
avrdude.exe: wiring_open(): releasing DTR/RTS
avrdude.exe: wiring_open(): asserting DTR/RTS
avrdude.exe: stk500v2_getsync(): found AVRISP programmer
             AVR Part                      : ATmega2560
             Chip Erase delay              : 9000 us
             PAGEL                         : PD7
             BS2                           : PA0
             RESET disposition             : dedicated
             RETRY pulse                   : SCK
             Serial program mode           : yes
             Parallel program mode         : yes
             Timeout                       : 200
             StabDelay                     : 100
             CmdexeDelay                   : 25
             SyncLoops                     : 32
             PollIndex                     : 3
             PollValue                     : 0x53
             Memory Detail                 :

                                               Block Poll               Page                       Polled
               Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
               ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
               eeprom                 65    10     8    0 no       4096    8      0  9000  9000 0x00 0x00
                                               Block Poll               Page                       Polled
               Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
               ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
               flash                  65    10   256    0 yes    262144  256   1024  4500  4500 0x00 0x00
                                               Block Poll               Page                       Polled
               Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
               ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
               lfuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
                                               Block Poll               Page                       Polled
               Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
               ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
               hfuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
                                               Block Poll               Page                       Polled
               Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
               ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
               efuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
                                               Block Poll               Page                       Polled
               Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
               ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
               lock                    0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
                                               Block Poll               Page                       Polled
               Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
               ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
               calibration             0     0     0    0 no          1    1      0     0     0 0x00 0x00
                                               Block Poll               Page                       Polled
               Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
               ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
               signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00
[... 55 lines cut ...]
avrdude>
avrdude.exe done.  Thank you.
```

Using AVR Dragon to read back shows that only the first 8 bytes are written.
```
PS C:\work\avr\avrdude_test\avrdude-v7.0-windows-x64> echo "read eeprom 0 0x40" | ./avrdude -c dragon_isp -p atmega2560 -t

[... cut ...]
avrdude.exe: Device signature = 0x1e9801 (probably m2560)
avrdude> >>> read eeprom 0 0x40

Reading | ################################################## | 100% 0.82s

0000  01 02 03 04 05 06 07 08  ff ff ff ff ff ff ff 10  |................|
0010  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0020  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0030  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|

avrdude>
avrdude.exe done.  Thank you.
```

mcuee, 2022-06-04:
Dragon_ISP is okay.

```
PS C:\work\avr\avrdude_test\avrdude-v7.0-windows-x64> echo "write eeprom 0x00 1 2 3 4 5 6 7 8 9 10 11 12 13 14 15 16" | ./avrdude -c dragon_isp -p atmega2560 -t

avrdude.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.15s

avrdude.exe: Device signature = 0x1e9801 (probably m2560)
avrdude> >>> write eeprom 0x00 1 2 3 4 5 6 7 8 9 10 11 12 13 14 15 16

Writing | ################################################## | 100% 3.23s

avrdude>
avrdude.exe done.  Thank you.

PS C:\work\avr\avrdude_test\avrdude-v7.0-windows-x64> echo "read eeprom 0 0x40" | ./avrdude -c dragon_isp -p atmega2560 -t

avrdude.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.15s

avrdude.exe: Device signature = 0x1e9801 (probably m2560)
avrdude> >>> read eeprom 0 0x40

Reading | ################################################## | 100% 0.82s

0000  01 02 03 04 05 06 07 08  09 0a 0b 0c 0d 0e 0f 10  |........     ...|
0010  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0020  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0030  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|

avrdude>
avrdude.exe done.  Thank you.
```