avrdudes/avrdude #516: [bug #61624] [Feature request] Serial/UART UPDI programmers

avrdude-admin, 2021-12-11:
Hans <hansibull@me.com>
Mon 06 Dec 2021 11:57:08 AM UTC
Programmer hardware: serialupdi

I know this is a lot to ask, and I don't expect anyone to jump onto this immediately because one guy would find this incredibly useful.

Avrdude now supports most UPDI compatible programming hardware, but the "last" thing it lacks is the ability to program UPDI compatible AVRs using a serial port and a resistor like used in pyupdi and pymcuprog.

[https://github.com/mraardvark/pyupdi](https://savannah.nongnu.orghttps://github.com/mraardvark/pyupdi)
 [https://github.com/microchip-pic-avr-tools/pymcuprog](https://savannah.nongnu.orghttps://github.com/microchip-pic-avr-tools/pymcuprog)

A "Serial UPDI" programmer is pretty much the simplest programmer you can make, and by controlling the DTR/RTS line while the UPDI communication is active, one could easily create a neat programmer you can use for both programming and serial debugging. I'm sure this feature would make UPDI compatible AVRs even more appealing, especially since Avrdude is a binary file that "just works" on all platforms, while a Python package isn't as simple.

This issue was migrated from https://savannah.nongnu.org/bugs/?61624

avrdude-admin, 2021-12-11:
Joerg Wunsch <joerg_wunsch>
Mon 06 Dec 2021 12:05:43 PM UTC

Dawid already volunteered to do just this. ;-)

avrdude-admin, 2021-12-11:
mcudude <mcudude>
Mon 06 Dec 2021 02:08:33 PM UTC

Sorry for not creating this issue with my actual account. I totally forgot that I had one! Anyways, I'm very excited about this, but also for the fact that there finally is some work done to Avrdude. It is after all an invaluable tool for SO many AVR developers!

Like I mentioned in the initial post, a DTR/RTS activity "indicator" will make it possible to create versatile programmers, and probably make "-c serialupdi" the preferred way for uploading to UPDI compatible AVRs, since the hardware is cheap and probably OSHW, the speed is high and the Avrdude implementation excellent!

avrdude-admin, 2021-12-16:
Dawid Buchwald <dawidbuchwald>
Sun 12 Dec 2021 08:07:29 PM UTC

Hello, I have some good news for you - I have first alpha version of the SerialUPDI programmer. I hope you can give it a spin and let me know if it works for you. I'm using FT232RL chip with a diode between Rx and Tx, 1K resistor between Tx and Diode and 470 resistor between UPDI pin and the diode, just like the "ideal" setup described in SerialUPDI documentation.

I will be most grateful if you could test drive it and let me know if you run into any issues. Should that happen, please run AVRDUDE with -v -v -v flags to provide full debug trace for me to investigate.

To start playing, you need to fetch it from SVN branch here:

[http://svn.savannah.gnu.org/viewvc/avrdude/branches/serialupdi/](https://savannah.nongnu.orghttp://svn.savannah.gnu.org/viewvc/avrdude/branches/serialupdi/)

Then follow the usual "build from sources" procedure. It should build without any issues, and then you can:

➜  avrdude ./avrdude -C ./avrdude.conf -p avr128db28 -b 230400 -c serialupdi -U flash:w:flash.bin:r -U fuse0:w:0x00:m -U fuse1:w:0x00:m -U fuse5:w:0xc9:m -U fuse6:w:0x14:m -U fuse7:w:0x00:m -U fuse8:w:0x00:m -P /dev/tty.usbserial-AB0JSZFM

avrdude: UPDI link initialization OK
avrdude: NVM type 2: 24-bit, word oriented write
avrdude: Entering NVM programming mode
avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.04s

avrdude: Device signature = 0x1e970e (probably avr128db28)
avrdude: NOTE: "flash" memory has been specified, an erase cycle will be performed
To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: reading input file "flash.bin"
avrdude: writing flash (928 bytes):

Writing | ################################################## | 100% 0.34s

avrdude: 928 bytes of flash written
avrdude: verifying flash memory against flash.bin:
avrdude: load data flash data from input file flash.bin:
avrdude: input file flash.bin contains 928 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 0.24s

avrdude: verifying ...
avrdude: 928 bytes of flash verified
avrdude: reading input file "0x00"
avrdude: writing fuse0 (1 bytes):

Writing | ################################################## | 100% 0.03s

avrdude: 1 bytes of fuse0 written
avrdude: verifying fuse0 memory against 0x00:
avrdude: load data fuse0 data from input file 0x00:
avrdude: input file 0x00 contains 1 bytes
avrdude: reading on-chip fuse0 data:

Reading | ################################################## | 100% 0.02s

avrdude: verifying ...
avrdude: 1 bytes of fuse0 verified
avrdude: reading input file "0x00"
avrdude: writing fuse1 (1 bytes):

Writing | ################################################## | 100% 0.03s

avrdude: 1 bytes of fuse1 written
avrdude: verifying fuse1 memory against 0x00:
avrdude: load data fuse1 data from input file 0x00:
avrdude: input file 0x00 contains 1 bytes
avrdude: reading on-chip fuse1 data:

Reading | ################################################## | 100% 0.02s

avrdude: verifying ...
avrdude: 1 bytes of fuse1 verified
avrdude: reading input file "0xc9"
avrdude: writing fuse5 (1 bytes):

Writing | ################################################## | 100% 0.03s

avrdude: 1 bytes of fuse5 written
avrdude: verifying fuse5 memory against 0xc9:
avrdude: load data fuse5 data from input file 0xc9:
avrdude: input file 0xc9 contains 1 bytes
avrdude: reading on-chip fuse5 data:

Reading | ################################################## | 100% 0.02s

avrdude: verifying ...
avrdude: 1 bytes of fuse5 verified
avrdude: reading input file "0x14"
avrdude: writing fuse6 (1 bytes):

Writing | ################################################## | 100% 0.03s

avrdude: 1 bytes of fuse6 written
avrdude: verifying fuse6 memory against 0x14:
avrdude: load data fuse6 data from input file 0x14:
avrdude: input file 0x14 contains 1 bytes
avrdude: reading on-chip fuse6 data:

Reading | ################################################## | 100% 0.02s

avrdude: verifying ...
avrdude: 1 bytes of fuse6 verified
avrdude: reading input file "0x00"
avrdude: writing fuse7 (1 bytes):

Writing | ################################################## | 100% 0.03s

avrdude: 1 bytes of fuse7 written
avrdude: verifying fuse7 memory against 0x00:
avrdude: load data fuse7 data from input file 0x00:
avrdude: input file 0x00 contains 1 bytes
avrdude: reading on-chip fuse7 data:

Reading | ################################################## | 100% 0.02s

avrdude: verifying ...
avrdude: 1 bytes of fuse7 verified
avrdude: reading input file "0x00"
avrdude: writing fuse8 (1 bytes):

Writing | ################################################## | 100% 0.03s

avrdude: 1 bytes of fuse8 written
avrdude: verifying fuse8 memory against 0x00:
avrdude: load data fuse8 data from input file 0x00:
avrdude: input file 0x00 contains 1 bytes
avrdude: reading on-chip fuse8 data:

Reading | ################################################## | 100% 0.02s

avrdude: verifying ...
avrdude: 1 bytes of fuse8 verified

avrdude: safemode: Fuses OK (E:FF, H:FF, L:FF)
avrdude: Leaving NVM programming mode

avrdude done.  Thank you.

avrdude-admin, 2021-12-16:
mcudude <mcudude>
Sun 12 Dec 2021 09:09:40 PM UTC

Hi David!

I'd love to try it out on various targets. I have lots of different chips I can try it out on.

One issue though. I've been using git for years, but never ever touched svn. Embarrassing question, how do I clone your repo/fork? I've tried various commands, but I always end up with an empty serialupdi folder.

I'm using MacOS, and when I run git svn clone [http://svn.savannah.gnu.org/viewvc/avrdude/branches/serialupdi](https://savannah.nongnu.orghttp://svn.savannah.gnu.org/viewvc/avrdude/branches/serialupdi)

I get the following message and an empty serialupdi folder:

Using existing [svn-remote "svn"]
svn-remote.svn.fetch already set to track :refs/remotes/git-svn

avrdude-admin, 2021-12-16:
Dawid Buchwald <dawidbuchwald>
Sun 12 Dec 2021 10:05:17 PM UTC

Sorry, should have been clearer. You should do the following:

svn co svn://svn.savannah.nongnu.org/avrdude/branches/serialupdi/avrdude

cd avrdude

./bootstrap

./configure

make all

./avrdude -C ./avrdude.conf -p avr128db28 -b 230400 -c serialupdi -U flash:w:flash.bin:r -U fuse0:w:0x00:m -U fuse1:w:0x00:m -U fuse5:w:0xc9:m -U fuse6:w:0x14:m -U fuse7:w:0x00:m -U fuse8:w:0x00:m -P /dev/tty.usbserial-AB0JSZFM

avrdude-admin, 2021-12-16:
mcudude <mcudude>
Sun 12 Dec 2021 11:04:03 PM UTC

No worries! I was able to build after following your instructions. It would be really nice if the DTR/RTS pulse could stay low during the entire Avrdude session. At the moment there seem to be two short (250us) pulses, one at the start and one at the end. This is different from how pymcuprog behaves. As mentioned in an earlier post, this signal could be used to control some simple digital logic that could automatically switch between UPDI and UART mode on a programmer.

I can successfully upload to an ATmega4809, but I'm getting an error when trying to read from the flash:

avrdude: verifying ...
avrdude: verification error, first mismatch at byte 0x0000
0x11 != 0x0c
avrdude: verification error; content mismatch

avrdude: safemode: Fuses OK (E:FF, H:FF, L:FF)
avrdude: Leaving NVM programming mode

avrdude done.  Thank you.

I'm trying to upload a simple blink program compiled with Arduino IDE. When uploading with an Xplained Mini programmer (and a different Avrdude version, the one bundled with Arduino IDE), I'm not getting any errors. the Xplained Mini doesn't work with the Avrdude binary I compiled. But that's a problem for another time.

Here is the hex file I'm loading:

:020000040000FA
:100000000C9450000C9462000C9462000C946200FA
:100010000C9462000C9462000C9462000C946200D8
:100020000C9462000C9462000C9462000C946200C8
:100030000C9462000C9462000C9462000C946200B8
:100040000C9462000C9462000C9462000C946200A8
:100050000C9462000C9462000C9462000C94620098
:100060000C9462000C94D9000C9462000C94620011
:100070000C9462000C9462000C9462000C94620078
:100080000C9462000C9462000C9462000C94620068
:100090000C9462000C9462000C9462000C94620058
:1000A00011241FBECFEFCDBFDFE3DEBF28E2A0E00B
:1000B000B8E201C01D92A430B207E1F70E9401012D
:1000C0000C9488010C9400009091910A9F7E90936B
:1000D000910A9091A00495FF0AC0811104C080E2AA
:1000E0008093A604089580E28093A50408959FB7A5
:1000F000F894811107C08091B504877F8093B5047F
:100100009FBF08958091B5048860F8CF8FB7F894A9
:100110002091002830910128409102285091032815
:10012000E091AA0AF091AB0A9091A60A90FF08C04C
:100130002F5F3F4F4F4F5F4FE091AA0AF091AB0AFC
:100140008FBFA8EEB3E00E946E0124E0F695E7951C
:100150002A95E1F76E0F7F1F811D911D0895CF92A3
:10016000DF92EF92FF920E9486006B017C0120E4F7
:10017000C20E22E4D21E2FE0E21EF11C80E4C8165B
:1001800082E4D8068FE0E806F10438F40E94860085
:10019000C616D706E806F906C8F30E9486006C1555
:1001A0007D058E059F05C8F3FF90EF90DF90CF90FF
:1001B00008951F920F920FB60F9211248F939F9361
:1001C000AF93BF938091002890910128A0910228BD
:1001D000B09103280196A11DB11D80930028909332
:1001E0000128A0930228B093032881E08093A60AF7
:1001F000BF91AF919F918F910F900FBE0F901F9065
:10020000189588ED90E084BF90936100809102067C
:1002100086608093020680910006816080930006CC
:10022000809102068F7C8065809302061092E20521
:1002300082E08093E40581E08093030A8EEF80934F
:10024000270A8093260A80E880932D0A80932B0A40
:100250008093290A80932C0A80932A0A8093280A83
:100260008BE08093000A87E08093E505E0E8FAE000
:1002700037E02EEF90E885E03183248795878083EF
:100280007096E03C4AE0F407B9F71092A10A8FE7B4
:100290009EE38093AC0A9093AD0A8091A50A816099
:1002A0008093A50A1092A00A8091A00A8160809391
:1002B000A00A789480E28093A104C0E0D0E081E0BD
:1002C0000E9464000E94AF0080E00E9464000E94CF
:1002D000AF002097A1F30E940000F1CF0E947901A6
:1002E000A59F900DB49F900DA49F800D911D11248A
:1002F0000895A29FB001B39FC001A39F700D811DFF
[... 517 lines cut ...]
avrdude: Sending 2 bytes [0x55, 0x24]
avrdude: Received 128 bytes [0x11, 0x24, 0x91, 0x1d, 0xb2, 0x9f, 0x70, 0x0d, 0x81, 0x1d, 0x11, 0x24, 0x91, 0x1d, 0x08, 0x95, 0xf8, 0x94, 0xff, 0xcf, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff]
Reading | ################################################## | 100% 0.15s

avrdude: avr_read(): skipping page 7: no interesting data
avrdude: avr_read(): skipping page 8: no interesting data
... (snip)
avrdude: avr_read(): skipping page 383: no interesting data
avrdude: verifying ...
avrdude: verification error, first mismatch at byte 0x0000
0x11 != 0x0c
avrdude: verification error; content mismatch

avrdude: safemode: Fuses OK (E:FF, H:FF, L:FF)
avrdude: Leaving NVM programming mode
avrdude: Sending reset request
avrdude: STCS 0x59 to address 0x08
avrdude: Sending 3 bytes [0x55, 0xc8, 0x59]
avrdude: Sending release reset request
avrdude: STCS 0x00 to address 0x08
avrdude: Sending 3 bytes [0x55, 0xc8, 0x00]
avrdude: STCS 0x0C to address 0x03
avrdude: Sending 3 bytes [0x55, 0xc3, 0x0c]

avrdude done.  Thank you.

avrdude-admin, 2021-12-16:
Dawid Buchwald <dawidbuchwald>
Mon 13 Dec 2021 08:29:25 AM UTC

First of all, thank you A LOT for your time and I'm really sorry it didn't work for you. Unfortunately, I don't own any other UPDI based MCUs, and there are apparently three different programming models for those, so I need to do some shopping today.

That being said, based on the output you provided (thanks again!) it seems that the issue is not very severe - programming of the flash went just fine and it was read back correctly. The only problem I see is that instead of writing to addresses 0x1000, it seems to have written your image (correctly, based on following read!) to 0x4000. I must have messed up integration of my code with AVRDUDE.

Anyway, I will let you know when I fix this. As for the other issue (DTR/RTS) - I will look into that a bit later, for now I just want the basic upload procedure to work fine.

avrdude-admin, 2021-12-16:
mcudude <mcudude>
Mon 13 Dec 2021 08:33:50 AM UTC

> First of all, thank you A LOT for your time and I'm really sorry it didn't

work for you. Unfortunately, I don't own any other UPDI based MCUs, and there
are apparently three different programming models for those, so I need to do
some shopping today.

Well, I'm just happy I can help out with the development! I also own a bunch of tiny0/1 boards as well, so I can also help you out testing these when the time is ready.

After further investigation, it looks reading flash does work, since the correct content are being received in verbose mode. However, Avrdude don't pick it up. When running the following command:

$ /Users/hans/Downloads/SVN/avrdude/avrdude -C/Users/hans/Documents/Arduino/hardware/MegaCoreX/megaavr/avrdude.conf -vvv -patmega4809 -cserialupdi -P /dev/cu.usbserial-1410 -b230400 -Uflash:r:dump.hex:i

I can see that the actual flash content data is received (matched the loaded hex file), but Avrdude concludes that flash is empty, and dump.hex only contains :00000001FF.

Here is a "compressed" output of the flash read command:

$ /Users/hans/Downloads/SVN/avrdude/avrdude -C/Users/hans/Documents/Arduino/hardware/MegaCoreX/megaavr/avrdude.conf -vvv -patmega4809 -cserialupdi -P /dev/cu.usbserial-1410 -b230400 -Uflash:r:dump.hex:i

...

avrdude: Version 6.3-20211205
Copyright (c) 2000-2005 Brian Dean, [http://www.bdmicro.com/](https://savannah.nongnu.orghttp://www.bdmicro.com/)
Copyright (c) 2007-2014 Joerg Wunsch

...

Programmer Type : serialupdi
Description     : SerialUPDI

avrdude: STCS 0x08 to address 0x03
avrdude: Sending 3 bytes [0x55, 0xc3, 0x08]
avrdude: STCS 0x80 to address 0x02
avrdude: Sending 3 bytes [0x55, 0xc2, 0x80]
avrdude: LDCS from 0x00
avrdude: Sending 2 bytes [0x55, 0x80]
avrdude: ser_recv(): programmer is not responding
avrdude: serialupdi_recv(): programmer is not responding
avrdude: Check failed
avrdude: Datalink not active, resetting...
avrdude: Sending double break
avrdude: Opening serial port...
avrdude: Opening serial port...
avrdude: STCS 0x08 to address 0x03
avrdude: Sending 3 bytes [0x55, 0xc3, 0x08]
avrdude: STCS 0x80 to address 0x02
avrdude: Sending 3 bytes [0x55, 0xc2, 0x80]
avrdude: LDCS from 0x00
avrdude: Sending 2 bytes [0x55, 0x80]
avrdude: Received 1 bytes [0x30]
avrdude: UDPI init OK
avrdude: UPDI link initialization OK
avrdude: Sending 2 bytes [0x55, 0xe6]
avrdude: Received 32 bytes [0x6d, 0x65, 0x67, 0x61, 0x41, 0x56, 0x52, 0x20, 0x50, 0x3a, 0x30, 0x44, 0x3a, 0x31, 0x2d, 0x33, 0x4d, 0x32, 0x20, 0x28, 0x30, 0x31, 0x2e, 0x35, 0x39, 0x42, 0x32, 0x30, 0x2e, 0x30, 0x29, 0x00]
avrdude: Received SIB: [megaAVR P:0D:1-3M2 (01.59B20.0)]
avrdude: Device family ID: megaAVR
avrdude: NVM interface: P:0
avrdude: Debug interface: D:1
avrdude: PDI oscillator: 3M2
avrdude: Extra information: (01.59B20.0)
avrdude: NVM type 0: 16-bit, page oriented write
avrdude: LDCS from 0x0B
avrdude: Sending 2 bytes [0x55, 0x8b]
avrdude: Received 1 bytes [0x82]
avrdude: Entering NVM programming mode
avrdude: UPDI writing key
avrdude: Sending 2 bytes [0x55, 0xe0]
avrdude: Sending 8 bytes [0x20, 0x67, 0x6f, 0x72, 0x50, 0x4d, 0x56, 0x4e]
[... 139 lines cut ...]
avrdude: ST_PTR to 0x00FF80
avrdude: Sending 4 bytes [0x55, 0x69, 0x80, 0xff]
avrdude: Received 1 bytes [0x40]
avrdude: Repeat 128
avrdude: Sending 3 bytes [0x55, 0xa0, 0x7f]
avrdude: LD8 from ptr++
avrdude: Sending 2 bytes [0x55, 0x24]
avrdude: Received 128 bytes [0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff]
Reading | ################################################## | 100% 6.17s

avrdude: Flash is empty, resulting file has no contents.
avrdude: writing output file "dump.hex"

avrdude: safemode: Fuses OK (E:FF, H:FF, L:FF)
avrdude: Leaving NVM programming mode
avrdude: Sending reset request
avrdude: STCS 0x59 to address 0x08
avrdude: Sending 3 bytes [0x55, 0xc8, 0x59]
avrdude: Sending release reset request
avrdude: STCS 0x00 to address 0x08
avrdude: Sending 3 bytes [0x55, 0xc8, 0x00]
avrdude: STCS 0x0C to address 0x03
avrdude: Sending 3 bytes [0x55, 0xc3, 0x0c]

avrdude done.  Thank you.

avrdude-admin, 2021-12-16:
Dawid Buchwald <dawidbuchwald>
Mon 13 Dec 2021 08:52:58 AM UTC

Thanks a lot for your support, I really appreciate the time and effort. I think I have a fix for you. What you have to do is:

svn update

make clean all

You expect following output:

➜  avrdude> svn update
Updating '.':
U    serialupdi.c
Updated to revision 1518.
➜  avrdude> make clean all

Try again when built. Thanks in advance!

Looking forward to your reply :)

avrdude-admin, 2021-12-16:
mcudude <mcudude>
Mon 13 Dec 2021 09:29:23 AM UTC

Thanks Dawid, reading from flash now works in the ATmega4809!

Just for curiosity and since I have all megaAVR-0 series chips, I decided to try them all.

Uploading and fuse read/write works on ATmega4809, ATmega3208, ATmega1608, and ATmega808.

On Atmega4808, ATmega3209, ATmega1609 and ATmega809 I'm getting "Segmentation fault: 11". Note that this only happens when I'm writing/reading fuses.

$ /Users/hans/Downloads/SVN/avrdude/avrdude -C/Users/hans/Documents/Arduino/hardware/MegaCoreX/megaavr/avrdude.conf -vvv -patmega3209 -cserialupdi -P /dev/cu.usbserial-1410 -b230400 -Ufuse0:w:0x00:m -Ufuse1:w:0x54:m -Ufuse2:w:0x01:m -Ufuse4:w:0x00:m -Ufuse5:w:0xC9:m -Ufuse6:w:0x06:m -Ufuse7:w:0x00:m -Ufuse8:w:0x00:m -Ulock:w:0xC5:m

avrdude: Version 6.3-20211205
Copyright (c) 2000-2005 Brian Dean, [http://www.bdmicro.com/](https://savannah.nongnu.orghttp://www.bdmicro.com/)
Copyright (c) 2007-2014 Joerg Wunsch

System wide configuration file is "/Users/hans/Documents/Arduino/hardware/MegaCoreX/megaavr/avrdude.conf"
User configuration file is "/Users/hans/.avrduderc"
User configuration file does not exist or is not a regular file, skipping

Using Port                    : /dev/cu.usbserial-1410
Using Programmer              : serialupdi
Overriding Baud Rate          : 230400
avrdude: Opening serial port...
AVR Part                      : ATmega3209
Chip Erase delay              : 0 us
PAGEL                         : P00
BS2                           : P00
RESET disposition             : dedicated
RETRY pulse                   : SCK
serial program mode           : yes
parallel program mode         : yes
Timeout                       : 0
StabDelay                     : 0
CmdexeDelay                   : 0
SyncLoops                     : 0
ByteDelay                     : 0
PollIndex                     : 0
PollValue                     : 0x00
Memory Detail                 :

Block Poll               Page                       Polled
Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
signature      0     0     0    0 no          3    1      0     0     0 0x00 0x00
Block Poll               Page                       Polled
Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
prodsig        0     0     0    0 no         61   61      0     0     0 0x00 0x00
Block Poll               Page                       Polled
Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
fuses          0     0     0    0 no          9    1      0     0     0 0x00 0x00
Block Poll               Page                       Polled
Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
fuse0          0     0     0    0 no          1    1      0     0     0 0x00 0x00
Block Poll               Page                       Polled
Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
fuse1          0     0     0    0 no          1    1      0     0     0 0x00 0x00
Block Poll               Page                       Polled
Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
fuse2          0     0     0    0 no          1    1      0     0     0 0x00 0x00
Block Poll               Page                       Polled
Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
fuse4          0     0     0    0 no          1    1      0     0     0 0x00 0x00
[... 160 lines cut ...]
avrdude: verifying ...
avrdude: 1 bytes of fuse0 verified
avrdude: reading input file "0x54"
avrdude: writing fuse1 (1 bytes):

Writing |                                                    | 0% 0.00savrdude: ST to 0x001281
avrdude: Sending 4 bytes [0x55, 0x44, 0x81, 0x12]
avrdude: Received 1 bytes [0x40]
avrdude: Sending 1 bytes [0x54]
avrdude: Received 1 bytes [0x40]
Writing | ################################################## | 100% 0.01s

avrdude: 1 bytes of fuse1 written
avrdude: verifying fuse1 memory against 0x54:
avrdude: load data fuse1 data from input file 0x54:
avrdude: input file 0x54 contains 1 bytes
avrdude: reading on-chip fuse1 data:

Reading |                                                    | 0% 0.00savrdude: LD from 0x001281
avrdude: Sending 4 bytes [0x55, 0x04, 0x81, 0x12]
avrdude: Received 1 bytes [0x00]
Reading | ################################################## | 100% 0.00s

avrdude: verifying ...
Segmentation fault: 11

avrdude-admin, 2021-12-16:
Dawid Buchwald <dawidbuchwald>
Mon 13 Dec 2021 10:01:35 AM UTC

Hey, I'm glad to hear that at least flash programming works correctly now. I assume that once uploaded, programs work as expected? Could you confirm this?

As for the atmega4808 failure during fuse setting, there are actually two problems there: the more obvious one is segmentation fault, which probably is just some invalid printf. It will take a while to find, but it's not a biggie. What bothers me more is the fact that fuse is being written with value 0x54, but read back with value 0x00:

avrdude: ST to 0x001281
avrdude: Sending 4 bytes [0x55, 0x44, 0x81, 0x12]
avrdude: Received 1 bytes [0x40]
avrdude: Sending 1 bytes [ **0x54** ]
avrdude: Received 1 bytes [0x40]
Writing | ################################################## | 100% 0.01s
avrdude: 1 bytes of fuse1 written
avrdude: verifying fuse1 memory against 0x54:
avrdude: load data fuse1 data from input file 0x54:
avrdude: input file 0x54 contains 1 bytes
avrdude: reading on-chip fuse1 data:
Reading |                                                    | 0% 0.00s
avrdude: LD from 0x001281
avrdude: Sending 4 bytes [0x55, 0x04, 0x81, 0x12]
avrdude: Received 1 bytes [ **0x00** ]

Unfortunately I don't think I can fix that without having the actual chip on my desk, and I'm currently waiting for one to arrive in couple of days.

In the meantime I will try to replicate the scenario somehow using the hardware I have on hand.

Once again, thanks a lot for all your effort!

avrdude-admin, 2021-12-16:
mcudude <mcudude>
Mon 13 Dec 2021 10:23:17 AM UTC

I can confirm that the uploaded program works as expected.

It looks like it's reading incorrect fuses on ATmega4809 as well. Tri's trying to write 0xCD, but reads back 0x40. And now I'm getting a segmentation fault when trying to set fuses on ATmega4809 as well, but not when only writing a hex file to flash...

$ /Users/hans/Downloads/SVN/avrdude/avrdude -C/Users/hans/Documents/Arduino/hardware/MegaCoreX/megaavr/avrdude.conf -vvv -patmega4809 -cserialupdi -P /dev/cu.usbserial-1410 -b230400 -Ufuse0:w:0x00:m -Ufuse1:w:0x54:m -Ufuse2:w:0x01:m -Ufuse4:w:0x00:m -Ufuse5:w:0xCD:m -Ufuse6:w:0x06:m -Ufuse7:w:0x00:m -Ufuse8:w:0x00:m -Ulock:w:0xC5:m

avrdude: Version 6.3-20211205
Copyright (c) 2000-2005 Brian Dean, [http://www.bdmicro.com/](https://savannah.nongnu.orghttp://www.bdmicro.com/)
Copyright (c) 2007-2014 Joerg Wunsch

System wide configuration file is "/Users/hans/Documents/Arduino/hardware/MegaCoreX/megaavr/avrdude.conf"
User configuration file is "/Users/hans/.avrduderc"
User configuration file does not exist or is not a regular file, skipping

Using Port                    : /dev/cu.usbserial-1410
Using Programmer              : serialupdi
Overriding Baud Rate          : 230400
avrdude: Opening serial port...
AVR Part                      : ATmega4809
Chip Erase delay              : 0 us
PAGEL                         : P00
BS2                           : P00
RESET disposition             : dedicated
RETRY pulse                   : SCK
serial program mode           : yes
parallel program mode         : yes
Timeout                       : 0
StabDelay                     : 0
CmdexeDelay                   : 0
SyncLoops                     : 0
ByteDelay                     : 0
PollIndex                     : 0
PollValue                     : 0x00
Memory Detail                 :

Block Poll               Page                       Polled
Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
signature      0     0     0    0 no          3    1      0     0     0 0x00 0x00
Block Poll               Page                       Polled
Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
prodsig        0     0     0    0 no         61   61      0     0     0 0x00 0x00
Block Poll               Page                       Polled
Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
fuses          0     0     0    0 no          9    1      0     0     0 0x00 0x00
Block Poll               Page                       Polled
Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
fuse0          0     0     0    0 no          1    1      0     0     0 0x00 0x00
Block Poll               Page                       Polled
Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
fuse1          0     0     0    0 no          1    1      0     0     0 0x00 0x00
Block Poll               Page                       Polled
Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
fuse2          0     0     0    0 no          1    1      0     0     0 0x00 0x00
Block Poll               Page                       Polled
Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
fuse4          0     0     0    0 no          1    1      0     0     0 0x00 0x00
Block Poll               Page                       Polled
Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
fuse5          0     0     0    0 no          1    1      0     0     0 0x00 0x00
[... 162 lines cut ...]
avrdude: verifying ...
avrdude: 1 bytes of fuse4 verified
avrdude: reading input file "0xCD"
avrdude: writing fuse5 (1 bytes):

Writing |                                                    | 0% 0.00savrdude: ST to 0x001285
avrdude: Sending 4 bytes [0x55, 0x44, 0x85, 0x12]
avrdude: Received 1 bytes [0x40]
avrdude: Sending 1 bytes [0xcd]
avrdude: Received 1 bytes [0x40]
Writing | ################################################## | 100% 0.01s

avrdude: 1 bytes of fuse5 written
avrdude: verifying fuse5 memory against 0xCD:
avrdude: load data fuse5 data from input file 0xCD:
avrdude: input file 0xCD contains 1 bytes
avrdude: reading on-chip fuse5 data:

Reading |                                                    | 0% 0.00savrdude: LD from 0x001285
avrdude: Sending 4 bytes [0x55, 0x04, 0x85, 0x12]
avrdude: Received 1 bytes [0xc9]
Reading | ################################################## | 100% 0.00s

avrdude: verifying ...
Segmentation fault: 11

avrdude-admin, 2021-12-16:
Dawid Buchwald <dawidbuchwald>
Mon 13 Dec 2021 10:27:00 AM UTC

Yeah, I just checked and the same issue occurs when trying to save to AVR128DB28 fuses. I think I know what this is, working on a fix now, should have it ready in a moment. Will let you know when it's ready.

avrdude-admin, 2021-12-16:
mcudude <mcudude>
Mon 13 Dec 2021 10:29:54 AM UTC

> Yeah, I just checked and the same issue occurs when trying to save to

AVR128DB28 fuses. I think I know what this is, working on a fix now, should
have it ready in a moment. Will let you know when it's ready.

Great! I'm about to leave for baby swimming with my daughter, but I'll test as soon as I'm back hoome :)

> Once again, thanks a lot for all your effort!

My pleasure! I currently maintain several Arduino "cores", that let you use pretty much any ATmega AVR with Arduino IDE while also providing way more flexibility in terms of fuse settings clock speeds etc. IMO Avrdude is the backbone of the entire AVR community, and I use it all the time, so helping you out with the testing is the least I can do! Having a reliable, FOSS uploading tool that supports pretty much all AVR programmers is such a blessing!

avrdude-admin, 2021-12-16:
Dawid Buchwald <dawidbuchwald>
Mon 13 Dec 2021 11:04:13 AM UTC

Hey, I have just posted another fix that should take care of the fuse write. Sorry for all the inconvenience. I would really appreciate if you could check it and let me know if it works correctly now.

I just noticed another issue, related to writing EEPROM, investigating it right now.

Have a great day!

avrdude-admin, 2021-12-16:
mcudude <mcudude>
Mon 13 Dec 2021 02:00:47 PM UTC

Hi! I have now tested your latest version on ATmega4809, 3209, 4808, and 3208, and everything seems to be working fine. All fuses are set and read back correctly, and I'm not getting any segment errors anymore.

I also have some tinyAVRs lying around, but they are not as convenient to experiment on as the two custom test boards I've designed for ATmegaX08 and ATmegaX09, so I was hoping to test with these when we think that the most obvious issues have been ironed out.

Let me know when you think you have resolved the EEPROM issue, and I'll be ready to test it!

(I don't know if this is related to the programmer in use or Avrdude in general, but would it be possible to show the "real" fuse values instead of this at the end of a session?)

avrdude: safemode: Fuses OK (E:FF, H:FF, L:FF)

> Have a great day!

You too!

avrdude-admin, 2021-12-16:
Dawid Buchwald <dawidbuchwald>
Mon 13 Dec 2021 02:06:30 PM UTC

Hey, thanks a lot for your help! I'm happy to hear that it's working now.

For the EEPROM issue - I found it and the fix is already in the version you have (svn update/svn info should give you revision 1520 or later). I was able to write/read back EEPROM contents, but it's extremely slow: 512 bytes take over a minute to write, due to page size of "1" in avrdude.conf.

Anyway, what I'm saying is that you should be able to test with these attiny chips, as long as you test flash and fuses.

Now, for the message at the end (lfuse/hfuse/efuse) - that's another interesting thing, but it's a part of core AVRDUDE code. I will think about approaching that in some reasonable, least invasive manner.

avrdude-admin, 2021-12-16:
mcudude <mcudude>
Mon 13 Dec 2021 02:30:38 PM UTC

I just tested reading from and writing to EEPROM on an ATmega4808 and ATmega4809, and it seems to work just fine. I've loaded various hex files, and the content read from EEPROM is always correct. It's also very fast, reading and writing take about a second, reading a little shorter perhaps.

>Anyway, what I'm saying is that you should be able to test with these attiny chips, as long as you test flash and fuses.

Great! I'll test them later tonight. I have to do some soldering in order to free up the UPDI pin.

> Now, for the message at the end (lfuse/hfuse/efuse) - that's another interesting thing, but it's a part of core AVRDUDE code. I will think about approaching that in some reasonable, least invasive manner.

Thanks for looking into it!

avrdude-admin, 2021-12-16:
mcudude <mcudude>
Mon 13 Dec 2021 06:57:04 PM UTC

Hello again Dawid!

I can confirm that I can correctly read and write to flash on both devices (I've just tested a couple of blink programs). I can also confirm that writing fuses and reading them back also works perfectly.

I've also tried to fill the EEPROMs with known content and read them back, and this also works. Great work!

Since most tiny0/tiny1 series are pretty similar, I'd expect SerialUPDI to work for all the other ones too. The tiny2 series is a little bit different and is more similar to the mega0 series.

avrdude-admin, 2021-12-16:
Dawid Buchwald <dawidbuchwald>
Mon 13 Dec 2021 07:10:39 PM UTC

Hello again! I'm really happy to hear that it works fine with other chips as well. At this point I will focus on some code refactoring and cleanup, because some of the code written today was just a quick and dirty implementation.

For the other things to do:
1. I need to figure out the whole lockbits business (I have no idea how it works, and it's not supported at the moment),
2. I need to improve error handling (especially when AVRDUDE crashes in the middle of NVM session, like with the segfault thing),
3. Reconsider the whole fuses idea, and see if that even can be changed,
4. There are other memory types like "prod_sig", and there is something about "user row" which I don't support at all yet,
5. Last but not least, look into DTR/RTS issue.

So yeah, plenty to do, but I'm glad the basic use cases (burning flash, setting fuses, writing to EEPROM) are working as expected for at least two family types (NVM 0 and NVM 2).

Whenever I reach any significant milestone I will post updates here with possible request for some testing. I have some other chips (atmega4809 and atmega4808) already in transit, so should be able to do way more testing myself.

What attiny0 chips are you using?

avrdude-admin, 2021-12-16:
mcudude <mcudude>
Mon 13 Dec 2021 08:00:47 PM UTC

> What attiny0 chips are you using?

Sorry, I forgot to tell you. I only own tiny1-series chips, so I tested on ATtiny416 and ATtiny817.

>So yeah, plenty to do, but I'm glad the basic use cases (burning flash, setting fuses, writing to EEPROM) are working as expected for at least two family types (NVM 0 and NVM 2).

Probably a noob question, but does there exist an NVM 1 family? I thought the only difference was 16-bit and 24-bit modes, where 16-bit is tiny0/1/2 and mega0, and 24-bit is AVR-DA/DB/DD/DU/EA

> Whenever I reach any significant milestone I will post updates here with possible requests for some testing.

Excellent! I'm ready when you are :)

avrdude-admin, 2021-12-16:
Dawid Buchwald <dawidbuchwald>
Mon 13 Dec 2021 09:21:29 PM UTC

> I only own tiny1-series chips, so I tested on ATtiny416 and ATtiny817.

Thanks for confirmation. I will get myself the other types as well then.

> Probably a noob question, but does there exist an NVM 1 family?

Not a noob question, it's really confusing. According to sources of pymcuprog there are actually three families: 0 (tiny), 2 (AVR Dx) and 3 (AVR EA). Family 1 has never been mentioned there.

That being said I think there are no NVM 3 chips in the wild.

I have also looked into the fuses business, and quite frankly, with my utter lack of knowledge about AVRDUDE source code (I volunteered to join the project like a week ago, for the sole reason of porting pymcuprog code to C) it would be way too risky.

See, the whole efuse/hfuse/lfuse/fuse model should be probably refactored, but we are talking about changing SIGNIFICANT part of the existing codebase affecting dozen different programmers. Without proper release model (parallel development of stable and refactored version) and considerable testing effort I wouldn't touch it.

That being said, there is an issue of safemode option. I don't want to pretend I fully understand what it does, but it seems to help to prevent accidental bricking of your chip with accidental fuse change. It supports the [e/l/h/-]fuse model and I don't think I could safely extend it to support the fuse0-8 model of new AVR chips. Sure it would be nice to have it, but I think it goes way beyond the intended SerialUPDI driver implementation. Thing is that it should work regardless of programmer used and I don't think it works for any new chips at all.

So yeah, for now I will focus on refactoring my code, improving error handling and supporting missing memory types (user row, lock) and possibly DTR/RTS support. Other things will have to wait to be prioritised.

avrdude-admin, 2021-12-16:
mcudude <mcudude>
Mon 13 Dec 2021 09:46:37 PM UTC

> Not a noob question, it's really confusing. According to sources of pymcuprog there are actually three families: 0 (tiny), 2 (AVR Dx) and 3 (AVR EA). Family 1 has never been mentioned there.

Makes me wonder why they didn't just come up with **one** mode to rule them all, instead of three.

If it turns out that the fuses and safe mode business is a deep rabbit hole, then it's probably best to leave it like it is for now. It would of course be a nice touch but by no means a deal-breaker of any sort.

avrdude-admin, 2021-12-16:
Dawid Buchwald <dawidbuchwald>
Wed 15 Dec 2021 02:14:18 PM UTC

Let me start with reply, followed by status update.

> Makes me wonder why they didn't just come up with one mode to rule them all, instead of three.

I suppose this is because they are in process of finalising the design? I have no idea, but I would rather expect even more versions in future. This, unfortunately, means that further changes to the UPDI code will be required at some point.

> If it turns out that the fuses and safe mode business is a deep rabbit hole

After some more investigation I concluded that there are actually different scenarios where Safemode is disabled, so I would really leave that for now.

**Not that short update**

So I managed to implement lockbits handling, turned out pretty easy to do. It works, I spent few hours scratching my head and trying to figure out how to unlock the device.

I have also tested potential change to avrdude.conf, related to AVR Dx family EEPROM page size - it works nicely with my programmer and size 0x20, but I'm a bit afraid it might not work with other programmers, so I won't change that. The only reason to play with this value is the efficiency of EEPROM write - it takes over a minute with 1-byte pages (there's huge overhead on UPDI communication) compared to ten seconds in 32-byte page mode.

Thing is - safety first. I don't write to EEPROM outside of my tests, so maybe better to leave it be.

So yeah, all the memories listed in AVRDUDE.CONF can be accessed now. Fuses can be configured, flash and EEPROM written, entire chip locked and unlocked. Error handling has been somewhat improved, AVRDUDE can now recover from crashing.

You can fetch latest version (revision 1522) to play with it, but it's not final, so consider carefully whether it's worth your time.

What I'm planning to do next is to do yet another refactoring. I don't like current model of error handling (inherited from pymcuprog) and I need to look into some weirder scenarios (like: lock the chip and upload flash with forced chip erase - it will work, but only the second time).

I'm also still waiting for all the tinyAVR chips. With the global shortage I had to order them abroad, and it's literally the worst time during the year to do it, so I will probably wait a bit longer. Good thing is I got all the types (0/1/2 as well as the mega ones).

MCUdude, 2022-01-07:
@dl8dtl this issue can be closed now, since SerialUPDI support has been added. BTW, can you please remove the email address at the top of the first post? Not sure why it has been exposed, but I prevent to keep it hidden due to all the spam I'm getting. Thanks!

closed 2022-01-07 by dl8dtl

mariusgreuel, 2022-01-07:
> Not sure why it has been exposed

@MCUdude: Sorry about that. The Savanne to GitHub migration script I wrote copied the data over, and I did not realize that the e-mail is hidden from public users.

MCUdude, 2022-01-07:
No worries!

Regarding issues, are there any plans to do some "house keeping" and close a bunch outdated or invalid bug reports / patches?  
There's alot that doesn't really apply anymore, like #185, #189 and a bunch of others.

dl8dtl, 2022-01-07:
> are there any plans to do some "house keeping"

So far, no plans. If you've go free CPU cycles for doing so, you're welcome. ;-)