avrdudes/avrdude #577: [patch #5655] osx intel avrispmkII hack

avrdude-admin, 2021-12-13:
Tue 02 Jan 2007 12:47:27 AM UTC

The last rev of osx-avr had avrdude5.2 bundled with it. I tested it with the mkII on the ppc and on the stk on intel. I got compalints about the mkII hanging for several minutes on the intel before programming. I dug around and found several threads where this was supposed to have been resolved with 5.2 and some which referenced the clock on the programmer. I was hoping that this would be resoved with the new version so I put it off.

Today I was finally able to sit down with this. When I tested the old version I was able to program fine on the ppc but the intel mac hung for several minutes before programming the chip. Then I ran up the new code and found the same thing. I ran across a few people who just hacked around it.

 [http://www.evilmadscientist.com/article.php/avrmac](https://savannah.nongnu.orghttp://www.evilmadscientist.com/article.php/avrmac)
 [http://article.gmane.org/gmane.comp.hardware.avr.avrdude.devel/958](https://savannah.nongnu.orghttp://article.gmane.org/gmane.comp.hardware.avr.avrdude.devel/958)

I notice that with the new version the avr-dragon seems to have the same porblem so I just copied the hack for the avrispmkII and compiled it for the intel macs.

Attached it the hack.

( As the origional code works fine on osx-ppc I think this patch only applies if building on osx for intel. )

I should have a binary rev of this out by the end of the year (in an hour or so)
Very Happy

Happy New Year.

Don.

- - - stk500v2.c.orig Wed Dec 20 15:43:34 2006--- stk500v2.c Sun Dec 31 13:09:39 2006
***************

- - - 485,490 ****--- 485,492 ----

if (pgmtype == PGMTYPE_JTAGICE_MKII)
return 0;
+ if (pgmtype == PGMTYPE_AVRISP_MKII)
+ return 0;

retry:
tries++;

- - - configure.ac.orig Sun Dec 31 16:31:19 2006--- configure.ac Sun Dec 31 16:10:22 2006
***************

- - - 24,30 ****# Process this file with autoconf to produce a configure script.

AC_PREREQ(2.57)
! AC_INIT(avrdude, 5.3.1, avrdude-dev@nongnu.org)

AC_CANONICAL_BUILD
AC_CANONICAL_HOST
--- 24,30 ----
# Process this file with autoconf to produce a configure script.

AC_PREREQ(2.57)
! AC_INIT(avrdude, 5.3.1.osxintelhack, avrdude-dev@nongnu.org)

AC_CANONICAL_BUILD
AC_CANONICAL_HOST

[file #11662: osxintelhack.diff](https://savannah.nongnu.org/patch/download.php?file_id=11662)

This issue was migrated from https://savannah.nongnu.org/patch/?5655

avrdude-admin, 2021-12-13:
Joerg Wunsch <joerg_wunsch>
Mon 29 Oct 2007 10:55:21 PM UTC

This appears to be the same as mentioned in **[bug #19882](https://savannah.nongnu.org/bugs/?19882)**.

What is not quite clear to me is why this only appears to
happen on some hardware/OS combinations but not everywhere.

Note that the JTAG ICE mkII is explicitly taken out of
consideration there since it uses its very own sign-on
sequence (way before that point), and has the AVRISP mkII
commands only wrapped up inside its own communication.

I'd really like to see some more analysis for the actual cause
of this problem before adding a hack that simply disables the
sign-on command.

MCUdude, 2022-01-10:
I'm closing this issue because there are currently no known issues related to the latest macOS version version of Avrdude and the AVRISPmkII programmer.

closed 2022-01-10 by MCUdude