avrdudes/avrdude #595: [patch #6543] serial-over-tcpip support for cygwin

avrdude-admin, 2021-12-13:
Thu 19 Jun 2008 09:25:02 PM UTC

Attached is a patch against the current CVS head (20080619) that adds serial-over-tcpip support if avrdude is built for cygwin.

This patch is required by TinyOS users who work under cygwin and want to use the MIB600 ethernet programming board from Crossbow.

[file #15878: avrdude-5.6cvs20080619-cygwin-net-serial.patch](https://savannah.nongnu.org/patch/download.php?file_id=15878)

This issue was migrated from https://savannah.nongnu.org/patch/?6543

avrdude-admin, 2021-12-13:
Joerg Wunsch <joerg_wunsch>
Tue 17 Feb 2009 01:15:25 PM UTC

Besides mixing some unrelated (and meanwhile obsolete) patches
into the actual patch, there is one fundamental issue with
this patch: it removes the -mno-cygwin option from CFLAGS
for Cygwin builds.  However, -mno-cygwin has been explicitly
added to allow Win32-API-only builds (i.e. MinGW32 builds)
even when building under Cygwin, so no dependency to
cygwin1.dll is required.  As far as I know, this feature is
crucial to the way the WinAVR build environment works.

For that reason, adding a native Cygwin build should be
implemented as a configure option (like --enable-native-cygwin),
defaulting to the current way (i.e. --disable-native-cygwin).
This is an acceptance requirement.

Also, as Cygwin basically offers a Posix-compliant API, I
think it would be better to massage the build system so
native Cygwin builds just go ahead, and use ser_posix.c
directly, rather than duplicating much of its functionality
in ser_win32.c.  This is not a requirement, but a wish that
allows for easier maintenance.

I'd kindly ask to submit a modified patch.

mcuee, 2022-05-24:
Does this still make sense now? I believe TinyOS is kind of dead now. There is also not much sense to use native Cygwin build even though that can be done.

closed 2022-10-23 by mcuee