avrdudes/avrdude #602: [patch #6824] driver for Digilent's USB programmer

avrdude-admin, 2021-12-13:
Thu 07 May 2009 06:35:28 PM UTC

I wrote a driver for Digilent's USB JTAG/ISP programming cable, and would appreciate your considering it for inclusion in avrdude. This driver handles two versions of the programmer. It is based on the code for the usbtiny, and I found the usb packet format by using the kernel logging module "usbmon". I have tested writing/reading to flash and eeprom for the mega64 and mega168 and it seems OK.

The patch also includes changes to the documentation. Feel free to edit if you like.

I needed to make a slight change in avr.c. For this programmer, the paged write command only works with paged memories, because it can't add delay between individual writes. So, I'd like to fall back to the default avr_write function for non-paged memories. In order to make this work, the counter "i" in avr_write needs to be changed from "unsigned long" to "long". I think that matches the original intent of the code:)

The patch is relative to svn trunk rev. 819

thanks
Dirk

[file #18115: usbdigilent.patch](https://savannah.nongnu.org/patch/download.php?file_id=18115)

This issue was migrated from https://savannah.nongnu.org/patch/?6824

avrdude-admin, 2021-12-13:
Joerg Wunsch <joerg_wunsch>
Tue 12 Jan 2010 02:55:58 PM UTC

> So, I'd like to fall back to the default avr_write
> function for non-paged memories. In order to make this
> work, the counter "i" in avr_write needs to be changed
> from "unsigned long" to "long". I think that matches the
> original intent of the code:)

I wonder why this never gave a warning like ``condition is
always true due to...''. ;-)

mcuee, 2022-06-09:
@dl8dtl
This is rather old, not so sure if this programmer is still offered by Digilent or not. If not probably this can be closed.
Ref: Current offerings from Digilent.
https://digilent.com/shop/boards-and-components/programmers/