avrdudes/avrdude #617: [patch #7275] Support for parallel port programmers on windows 7 64 bit and other versions of 64 bit windows

avrdude-admin, 2021-12-13:
Sun 08 Aug 2010 08:24:45 PM UTC

This patch enables avrdude to use InpOut for parallel port access on 64 bit versions of windows.
I used a modified version of InpOut for 64 bit windows from [http://www.highrez.co.uk/Downloads/InpOut32/default.htm](https://savannah.nongnu.orghttp://www.highrez.co.uk/Downloads/InpOut32/default.htm)
I modified ppiwin.c, rewriting "inb" and "outb" functions to use InpOut when runtime option "-a" is passed.
This library, InpOut seems to work both on 32 bit windows and on 64 bit windows.

Before running avrdude first time, the user should run Win32\InstallDriver.exe (from InpOutBinaries_1200.zip) which seems to install automatically the required version of driver (32 bit or 64 bit). A dll file (inpout32.dll) is also required at runtime when using option "-a".

[file #21170: avrdude_inpout.patch](https://savannah.nongnu.org/patch/download.php?file_id=21170)

This issue was migrated from https://savannah.nongnu.org/patch/?7275

avrdude-admin, 2021-12-13:
Joerg Wunsch <joerg_wunsch>
Wed 17 Aug 2011 02:28:17 PM UTC

Please don't forget to submit documentation patches.

I just added a peramble to the patch tracker submission form
explaining some details about our documentation.

Thanks for your contribution!

avrdude-admin, 2021-12-13:
Henrik Haftmann <heha>
Mon 15 Jul 2013 10:43:11 PM UTC

InpOut32.dll doesn't need any initialization as it will install its containing port access driver silently when used the first time. Obviously this will require administrative privilege. Moreover, it contains signed 64-bit drivers so it does not make any trouble with Driver Sign Enforcement. Furthermore, such a DLL can be easily replaced by one that redirects I/O to any other nonstandard device, or to nonstandard port addresses as seen for PCI/PCIexpress cards. (I have written such a DLL.) Therefore, relying to InpOut32.dll is less annoying in such cases than to giveio. The only disadvantage are the countless kernel transitions (one for each I/O operation) which will slow down programming.

Therefore, I would strongly recommend the patch from <bogdanflorin>, and possibly default to use InpOut32.dll if something fails when accessing giveio silently (= no need for a command-line option on Win64 systems).

avrdude-admin, 2021-12-13:
Joerg Wunsch <joerg_wunsch>
Sat 27 Nov 2021 09:35:37 PM UTC

Is this really still relevant, given that parallel-port computers are no longer really available?

avrdude-admin, 2021-12-13:
Joerg Wunsch <joerg_wunsch>
Tue 30 Nov 2021 05:33:14 AM UTC

See mailinglist discussion:

[https://lists.nongnu.org/archive/html/avrdude-dev/2021-11/msg00082.html](https://savannah.nongnu.orghttps://lists.nongnu.org/archive/html/avrdude-dev/2021-11/msg00082.html)

mcuee, 2022-06-11:
Ref: https://lists.nongnu.org/archive/html/avrdude-dev/2021-11/msg00082.html

@dl8dtl and @mariusgreuel 
Windows giveio.sys has been removed in the git repo and I think this can be closed. There is really no point to support Parallel Port programmer under Windows.

closed 2022-06-11 by MCUdude