avrdudes/avrdude #619: [patch #7321] usbasp: filter usb serial number

avrdude-admin, 2021-12-13:
Mon 20 Sep 2010 08:30:42 PM UTC

Recently I tried using several usbasp compatible programmers (running my
reimplemantation of the usbasp protocol([http://github.com/fd0/kahuna](https://savannah.nongnu.orghttp://github.com/fd0/kahuna))) at the same time, unfortunately
it is impossible to tell avrdude which one to use (it will just the first it
discovers, and I ended up using a usbasp programmer and my old avrisp2. **brr**).

So I added a unique usb serial number to all of my programmer devices and
patched avrdude so that if the port (-P) is given in the format usb:serialno
(like in the stk500v2 module), only devices with (prefix-) matching serial
numbers are considered.

[file #21512: 0002-prefix-matching-for-usb-serial-numbers-for-usbasp.patch](https://savannah.nongnu.org/patch/download.php?file_id=21512)
[file #21511: 0001-match-serial-number-for-usbasp-compatible-programmer.patch](https://savannah.nongnu.org/patch/download.php?file_id=21511)

This issue was migrated from https://savannah.nongnu.org/patch/?7321

avrdude-admin, 2021-12-13:
Alexander Neumann <fd0>
Thu 25 Aug 2011 08:32:10 AM UTC

Hey Thomas,

Thanks for taking over, what is the state of this patch?

Cheers,
- Alex

avrdude-admin, 2021-12-13:
Joerg Wunsch <joerg_wunsch>
Wed 17 Jan 2018 10:43:41 PM UTC

Unfortunately, Thomas never handled it, so I'll take it over
now.

Meanwhile, AVRDUDE has replaced all fprintf(stderr, ... by
avrdude_message(MSG_<severity>, ...), so the patch no longer
applies. Are you interested in filing an update patch?

Regarding a partial serial number match, I suggest to match
the **trailing** digits rather than the initial substring: usual
serial numbers commonly have a similar initial part but differ
only in the last digits.  That's how the other programmers
do it which implement a serial number match.

mcuee, 2022-06-10:
I think this is a useful feature. Therefore I create a new issue here.
* #973 

Relevant discussions:
* #718