avrdudes/avrdude #629: [patch #7594] Support for Atmel serial UART-Bootloader "FastBoot"

avrdude-admin, 2021-12-13:
Tue 09 Aug 2011 08:00:54 PM UTC

At [http://www.mikrocontroller.net/topic/73196](https://savannah.nongnu.orghttp://www.mikrocontroller.net/topic/73196) and [http://www.mikrocontroller.net/articles/AVR_Bootloader_FastBoot_von_Peter_Dannegger](https://savannah.nongnu.orghttp://www.mikrocontroller.net/articles/AVR_Bootloader_FastBoot_von_Peter_Dannegger) exists an implementation of a serial bootloader for a wide range of 8-bit Atmel devices.

The attached patch adds support to program this devices with avrdude over a serial line. It should work on linux and windows. I plan to implement the one-wire mode later.

A complete call for programming might look like:
avrdude -c pedabl -p m8 -P /dev/ttyUSB0 -b 57600 -U flash:w:main.hex -x "reset_cmd=\nR\n" -x "reset_baud=9600"

The last two extended options are optional and are used to automatically reset the device for programming.

The bootloader does not support uploading the program to the pc therefore the verify procedure has to be change in avrdude. To verify a program it has to be sent to the device again but this time the device checks it itself against the flash and returns a status code.

[file #23772: pedabl.udiff](https://savannah.nongnu.org/patch/download.php?file_id=23772)
[file #23910: fastboot.udiff](https://savannah.nongnu.org/patch/download.php?file_id=23910)

This issue was migrated from https://savannah.nongnu.org/patch/?7594

avrdude-admin, 2021-12-13:
Joerg Wunsch <joerg_wunsch>
Wed 17 Aug 2011 02:39:25 PM UTC

Please don't forget to submit documentation patches.

I just added a preamble to the patch tracker submission form
explaining some details about our documentation.

Thanks for your contribution!

avrdude-admin, 2021-12-13:
Stefan <stefan_sn>
Wed 24 Aug 2011 05:59:15 PM UTC

Ok, so I will update the patch with documentation changes in the next days. I also think about to change the name of the programmer from 'pedabl' to 'fastboot'. So please wait with the integration of the patch.

avrdude-admin, 2021-12-13:
Stefan <stefan_sn>
Tue 30 Aug 2011 06:11:39 PM UTC

Joerg

Please find attached the patch including documentation changes. As I mentioned before I also changed the programmer's name from 'pedabl' (means 'Peter Danneggers Bootloader') to 'FastBoot' which is also used on the website of mikrocontroller.net.

If you feel something is missing, please let me know.

It would be great to see the patch in the main distribution of avrdude!

(**[file #23910](https://savannah.nongnu.org/support/download.php?file_id=23910)**)

avrdude-admin, 2021-12-13:
Joerg Wunsch <joerg_wunsch>
Fri 13 Sep 2013 04:12:15 PM UTC

The patch contains unrelated changes to ser_posix.c/ser_win32.c
(different indentation level).  I won't commit these changes.
If anything needs to be changed in those files, please resubmit
that.

Also, what happens if someone e.g. starts terminal mode on
a fastboot device, and tries to use one of the byte-access
functions, or tries a -U flash:r:foo.hex:i (paged read of
entire flash)?

Finally, the page write algorithm has changed meanwhile, the
respective backend functions are now called once per page
rather than on the entire memory region.  Since I don't have
any fastboot-ready target around, I would appreciate if you
could adapt your patch to the new structure.  (I can do it
myself, but it will be a "blind flight".)

mcuee, 2022-06-04:
Reference in case people want to look for alternative host program to avrdude for this bootloader
* https://github.com/quirxi/FastBoot-Linux

mcuee, 2022-10-23:
I will close this as this issue has not been updated for so long and the bootloader does not seem to be popular.

closed 2022-10-23 by mcuee