avrdudes/avrdude #633: [patch #7610] Patch for FTDI 232H chip against AVRDUDE 5.11

avrdude-admin, 2021-12-13:
Tue 30 Aug 2011 05:49:03 AM UTC

Here is a patch to support the FTDI 232H part in AVRDUDE 5.11.  It uses libftdi 0.19 plus git patches that support the 232H part.  It has been built and tested to success in Windows32 with mingw32 and the development libftdi0.19git drivers from [http://code.google.com/p/picusb/downloads/list](https://savannah.nongnu.orghttp://code.google.com/p/picusb/downloads/list) .

$ ./avrdude -c um232h -p m1280 -v -b 3700000 -U flash:w:ATmegaBOOT_168_atmega1280.hex

avrdude.exe: Version 5.11, compiled on Aug 30 2011 at 15:21:23
Copyright (c) 2000-2005 Brian Dean, [http://www.bdmicro.com/](https://savannah.nongnu.orghttp://www.bdmicro.com/)
Copyright (c) 2007-2009 Joerg Wunsch

System wide configuration file is "C:\temp\av\avrdude-5.11-FT232HPatch\avrdude.conf"

Using Port                    : lpt1
Using Programmer              : um232h
Overriding Baud Rate          : 3700000
avrdude.exe: device: \\.\libusb0-0255--0x0403-0x6014, serial number: FTUIF780 type 0x0900 found
avrdude.exe: Using device VID:PID 0403:6014 type 0x0900(232H) and SN 'FTUIF780'.
avrdude.exe: Using USB Interface A
avrdude.exe info: reset pin value: 4
add_pins: 1: 0x0000, inv=0x0000
add_pins: 2: 0x0000, inv=0x0000
add_pin: 7: bit 0x0000 inv=0x0000
add_pin: 8: bit 0x0000 inv=0x0000
add_pin: 9: bit 0x0000 inv=0x0000
add_pin: 10: bit 0x0000 inv=0x0000
avrdude.exe info: clock divisor: 0x0007 Actual SPI ClK freq: 3750000Hz
AVR Part                      : ATMEGA1280
Chip Erase delay              : 9000 us
PAGEL                         : PD7
BS2                           : PA0
RESET disposition             : dedicated
RETRY pulse                   : SCK
serial program mode           : yes
parallel program mode         : yes
Timeout                       : 200
StabDelay                     : 100
CmdexeDelay                   : 25
SyncLoops                     : 32
ByteDelay                     : 0
PollIndex                     : 3
PollValue                     : 0x53
Memory Detail                 :

Block Poll               Page                       Polled
Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
eeprom        65    10     8    0 no       4096    8      0  9000  9000 0x00 0x00
flash         65    10   256    0 yes    131072  256    512  4500  4500 0x00 0x00
lfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
hfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
efuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
lock           0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

Programmer Type : avrftdi
Description     : FT232H based module from FTDI and Glyn.com.au

avrdude.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.10s

avrdude.exe: Device signature = 0x1e9703
avrdude.exe: safemode: lfuse reads as 8E
avrdude.exe: safemode: hfuse reads as 1B
avrdude.exe: safemode: efuse reads as FD
avrdude.exe: NOTE: FLASH memory has been specified, an erase cycle will be performed
To disable this feature, specify the -D option.
avrdude.exe: erasing chip
avrdude.exe: reading input file "ATmegaBOOT_168_atmega1280.hex"
avrdude.exe: input file ATmegaBOOT_168_atmega1280.hex auto detected as Intel Hex
avrdude.exe: writing flash (130838 bytes):

Writing | ################################################## | 100% 6.37s

avrdude.exe: 130838 bytes of flash written
avrdude.exe: verifying flash memory against ATmegaBOOT_168_atmega1280.hex:
avrdude.exe: load data flash data from input file ATmegaBOOT_168_atmega1280.hex:
avrdude.exe: input file ATmegaBOOT_168_atmega1280.hex auto detected as Intel Hex
avrdude.exe: input file ATmegaBOOT_168_atmega1280.hex contains 130838 bytes
avrdude.exe: reading on-chip flash data:

Reading | ################################################## | 100% 5.01s

avrdude.exe: verifying ...
avrdude.exe: 130838 bytes of flash verified

avrdude.exe: safemode: lfuse reads as 8E
avrdude.exe: safemode: hfuse reads as 1B
avrdude.exe: safemode: efuse reads as FD
avrdude.exe: safemode: Fuses OK

avrdude.exe done.  Thank you.

[file #23901: avrdude-5.11-232H.patch](https://savannah.nongnu.org/patch/download.php?file_id=23901)
[file #23905: ATmegaBOOT_168_atmega328.hex](https://savannah.nongnu.org/patch/download.php?file_id=23905)

This issue was migrated from https://savannah.nongnu.org/patch/?7610

avrdude-admin, 2021-12-13:
Jason Hecker <roklobster>
Tue 30 Aug 2011 05:53:12 AM UTC

I will add I have tried this patch in Ubuntu 11.04 with the same AVRDUDE 5.11 tarball.  The results to Windows are different - the speed is about 5x faster (does Windows libusb have a bottleneck?) but the validation for the hex file fails.  libftdi0.19git was built and used as well.  At this point I haven't investigated why this patch fails in Linux.

blah@Myth:~/avrdude/avrdude-5.11$ sudo ./avrdude -C ./avrdude.conf -c um232h -p m1280 -v -b 3700000 -U flash:w:ATmegaBOOT_168_atmega1280.hex

avrdude: Version 5.11, compiled on Aug 30 2011 at 15:00:29
Copyright (c) 2000-2005 Brian Dean, [http://www.bdmicro.com/](https://savannah.nongnu.orghttp://www.bdmicro.com/)
Copyright (c) 2007-2009 Joerg Wunsch

System wide configuration file is "./avrdude.conf"
User configuration file is "/home/blah/.avrduderc"
User configuration file does not exist or is not a regular file, skipping

Using Port                    : /dev/parport0
Using Programmer              : um232h
Overriding Baud Rate          : 3700000
avrdude: device: 003, serial number: FTUIF780 type 0x0900 found
avrdude: Using device VID:PID 0403:6014 type 0x0900(232H) and SN 'FTUIF780'.
avrdude: Using USB Interface A
avrdude info: reset pin value: 4
add_pins: 1: 0x0000, inv=0x0000
add_pins: 2: 0x0000, inv=0x0000
add_pin: 7: bit 0x0000 inv=0x0000
add_pin: 8: bit 0x0000 inv=0x0000
add_pin: 9: bit 0x0000 inv=0x0000
add_pin: 10: bit 0x0000 inv=0x0000
avrdude info: clock divisor: 0x0007 Actual SPI ClK freq: 3750000Hz
AVR Part                      : ATMEGA1280
Chip Erase delay              : 9000 us
PAGEL                         : PD7
BS2                           : PA0
RESET disposition             : dedicated
RETRY pulse                   : SCK
serial program mode           : yes
parallel program mode         : yes
Timeout                       : 200
StabDelay                     : 100
CmdexeDelay                   : 25
SyncLoops                     : 32
ByteDelay                     : 0
PollIndex                     : 3
PollValue                     : 0x53
Memory Detail                 :

Block Poll               Page                       Polled
Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
eeprom        65    10     8    0 no       4096    8      0  9000  9000 0x00 0x00
flash         65    10   256    0 yes    131072  256    512  4500  4500 0x00 0x00
lfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
hfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
efuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
lock           0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

Programmer Type : avrftdi
Description     : FT232H based module from FTDI and Glyn.com.au

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e9703
avrdude: safemode: lfuse reads as 8E
avrdude: safemode: hfuse reads as 1B
avrdude: safemode: efuse reads as FD
avrdude: NOTE: FLASH memory has been specified, an erase cycle will be performed
To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: reading input file "ATmegaBOOT_168_atmega1280.hex"
avrdude: input file ATmegaBOOT_168_atmega1280.hex auto detected as Intel Hex
avrdude: writing flash (130838 bytes):

Writing | ################################################## | 100% 1.47s

avrdude: 130838 bytes of flash written
avrdude: verifying flash memory against ATmegaBOOT_168_atmega1280.hex:
avrdude: load data flash data from input file ATmegaBOOT_168_atmega1280.hex:
avrdude: input file ATmegaBOOT_168_atmega1280.hex auto detected as Intel Hex
avrdude: input file ATmegaBOOT_168_atmega1280.hex contains 130838 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 1.65s

avrdude: verifying ...
avrdude: verification error, first mismatch at byte 0x1f000
0x0c != 0xff
avrdude: verification error; content mismatch

avrdude: safemode: lfuse reads as 8E
avrdude: safemode: hfuse reads as 1B
avrdude: safemode: efuse reads as FD
avrdude: safemode: Fuses OK

avrdude done.  Thank you.

avrdude-admin, 2021-12-13:
Jason Hecker <roklobster>
Tue 30 Aug 2011 08:19:43 AM UTC

I have added the hex file I have been using for testing.  I'd appreciate it if anyone else could test uploading to:

- Arduino Mega or other AtMega1280 device
- use the 'avrftdi' driver
- 4232H or 2232H or 232C/D
- the release version of AVRDUDE 5.11
- Linux or FreeBSD

and see if you get the same verification errors.
The hex file is the bootloader from the Arduino 0.22 package.

(**[file #23905](https://savannah.nongnu.org/support/download.php?file_id=23905)**)

avrdude-admin, 2021-12-13:
Joerg Wunsch <joerg_wunsch>
Tue 10 Jan 2012 03:22:39 PM UTC

Delayed until libftdi-1 has been released.

Also, a snippet in configure.ac is needed to detect the
presence of libftdi-1, and to conditionalize the respective
code parts on that (so it will continue to build on
libftdi-0.x systems).

avrdude-admin, 2021-12-13:
Rene Liebscher <rliebscher>
Thu 01 Nov 2012 08:52:39 AM UTC

See also **[patch #7729](https://savannah.nongnu.org/patch/?7729)** "Updated support for FTDI-232H"

MCUdude, 2022-03-19:
I'm closing this issue because there hasn't been any activity i the last ten years in this thread, and Avrdude 6 and the upstream version works fine with FT232.

closed 2022-03-19 by MCUdude