avrdudes/avrdude #659: [patch #7729] Updated support for FTDI-232H

avrdude-admin, 2021-12-13:
Fri 24 Feb 2012 02:13:17 PM UTC

I was having some problems with certain .hex files not being verified correctly in my old 5.11-patch7610 of AVRDUDE.

I have since checked out SVN1068 and applied **[patch #7610](https://savannah.nongnu.org/patch/?7610)** making the necessary changes so that it compiled.

The new binary now verifies the problematic hex files correctly.

This patch adds support for the FTDI-232H part which will be properly available in the upcoming libftdi-0.20.  In addition it enables the 30MHz clock in the FTDI chip for all H suffix components.  This will obviously allow SPI rates up to 30MHz (as opposed to the default 6MHz) - not that any AVR can do that.  It effectively adds some finer frequency granularity for use with the AVR.

I have built the binary using MinGW and the headers and libs from [http://code.google.com/p/picusb/downloads/detail?name=libftdi_0.19git_devkit_mingw32_27August2011.zip](https://savannah.nongnu.orghttp://code.google.com/p/picusb/downloads/detail?name=libftdi_0.19git_devkit_mingw32_27August2011.zip)

It's only been tested on Win7-64 bit with libusb-win32 1.2.5.0.

I have also added a new definition for the C232HM MPSSE cable from FTDI though it's just a copy of the UM232H.

I'll update my site helix.air.net.au with the new Windows binary too.

[file #25155: avrdude-fdi232h.patch](https://savannah.nongnu.org/patch/download.php?file_id=25155)
[file #25162: avrdude-fdi232h_1.patch](https://savannah.nongnu.org/patch/download.php?file_id=25162)

This issue was migrated from https://savannah.nongnu.org/patch/?7729

avrdude-admin, 2021-12-13:
Hannes Weisbach <hweisbach>
Fri 24 Feb 2012 08:40:32 PM UTC

Hi,

I had a quick look and it looks fine to me. I have just one question. You changed:
if (pgm->pinno[PIN_AVR_RESET] < 4 || pgm->pinno[PIN_AVR_RESET] == 0) {
to:
if (pgm->pinno[PIN_AVR_RESET] < 5 || pgm->pinno[PIN_AVR_RESET] == 0) {
and I don't understand why. Using pin 4 (TMS/CS) as reset-pin is a valid choice and indeed the default. Four lines below pgm->pinno[PIN_AVR_RESET] actually gets set to 4.

Cheers,
Hannes

avrdude-admin, 2021-12-13:
Jason Hecker <roklobster>
Sat 25 Feb 2012 01:21:33 PM UTC

You are right.  I used pin 5 for some reason.  I have corrected this and changed it all to pin 4 and tested it.  New patch below.

(**[file #25162](https://savannah.nongnu.org/support/download.php?file_id=25162)**)

avrdude-admin, 2021-12-13:
Hannes Weisbach <hweisbach>
Wed 25 Sep 2013 11:10:43 AM UTC

So, I've looked over the patch in detail (finally, sorry for the delay).
I'm not convinced on the merits of the patch. As you've stated, the increased speed does not matter much for AVRs. On the finer frequency resolution I would argue, that
a) you still have only 0xffff distinct frequencies to choose from
b) this was (in my opinion) never a problem in the first place.
Let me elaborate: If I program an AVR part, I choose my SPI frequency fairly arbitrarily and rather conservatively, Let's say 100kHz. Whether I can choose 101.6949kHz (12MHz) or 100.3344kHz (for the 60MHz version) does not matter to me at all (the question is, if this applies to all/the majority of AVRdude users).
In contrast to that feature is the increased code complexity and maintenance (Think a new FTDI device). Essentially, it would be necessary (more or less) to copy the switch-statement from avrftdi_open() (handling the different device types) into set_frequency(). I think that is a bad idea. I could live with having set_frequency() called after the switch and the maximum frequency be passed in as a parameter, determined within said switch-statement. This still leaves the command to switch off the divisior - I'm not sure how to handle this in a "good" way, although one could always hope that FTDI builds new devices backwards-compatible (as they already did).

Other opinions/comments/suggestions/arguments?

Programmer declarations for UM232H and C232H are included in r1248.

mcuee, 2022-05-29:
As mentioned in the previous comments and looking at the codes, other than set_frequency changes, the other parts have been merged (eg: the current codes already differentiate different FTDI chips).

So the question is whether the 30MHz is useful or not. It does not seem to help much as per the comments. In that case, maybe this can be closed as there are no real benefits.

```
 
+/*
+ * The *232H series can clock up to 30MHz.  Older chips like the 232C clock up to 6MHz.
+ * This function will select which mode is appropriate based on the device.
+ * It won't neccessrily mean you can program the chip any faster - it just means the
+ * SPI frequency can be tuned a bit finer.
+ */
 static int set_frequency(avrftdi_t* ftdi, uint32_t freq)
```

mcuee, 2022-06-08:
@dl8dtl and @mariusgreuel 
I will propose that we close this for now since 30MHz does not help much.

mcuee, 2022-06-08:
On the other hand, not sure if 30MHz will result in faster Flash read/write speed.

It takes me about 1 minutes to read the Flash contents of a AVR Mega2560 on the Arduino Mega2560.

```
PS C:\work\avr\avrdude_test\avrdude-v7.0-windows-x64>  .\avrdude.exe -p m2560 -c jtagkey -U flash:r:mega2560_blink_ftdi2232h_read.hex:i -vv

avrdude.exe: Version 7.0
             Copyright (c) Brian Dean, http://www.bdmicro.com/
             Copyright (c) Joerg Wunsch

             System wide configuration file is "C:/work/avr/avrdude_test/avrdude-v7.0-windows-x64/avrdude.conf"

             Using Port                    : usb
             Using Programmer              : jtagkey
I avrftdi_open(714): Using device VID:PID 0403:cff8 and SN '(null)' on interface A.
I set_frequency(210): Using frequency: 150000
I set_frequency(211): Clock divisor: 0x0027
I avrftdi_pin_setup(645): Pin direction mask: 001b
I avrftdi_pin_setup(646): Pin value mask: 0010
             AVR Part                      : ATmega2560
             Chip Erase delay              : 9000 us
             PAGEL                         : PD7
             BS2                           : PA0
             RESET disposition             : dedicated
             RETRY pulse                   : SCK
             Serial program mode           : yes
             Parallel program mode         : yes
             Timeout                       : 200
             StabDelay                     : 100
             CmdexeDelay                   : 25
             SyncLoops                     : 32
             PollIndex                     : 3
             PollValue                     : 0x53
             Memory Detail                 :

                                               Block Poll               Page                       Polled
               Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
               ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
               eeprom                 65    10     8    0 no       4096    8      0  9000  9000 0x00 0x00
               flash                  65    10   256    0 yes    262144  256   1024  4500  4500 0x00 0x00
               lfuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
               hfuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
               efuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
               lock                    0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
               calibration             0     0     0    0 no          1    1      0     0     0 0x00 0x00
               signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00

             Programmer Type : avrftdi
             Description     : Amontec JTAGKey, JTAGKey-Tiny and JTAGKey2

avrdude.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.01s

avrdude.exe: Device signature = 0x1e9801 (probably m2560)
avrdude.exe: reading flash memory:

Reading | ################################################## | 100% 57.53s

avrdude.exe: writing output file "mega2560_blink_ftdi2232h_read.hex"
I avrftdi_close(807): LED Mask=0x0000 value =0x0018 &=0x0000

avrdude.exe done.  Thank you.
```

MCUdude, 2022-06-08:
I don't own a 2232-based programmer, but a rule of thumb is to not use an ISP clock faster than 1/4 of the microcontroller clock speed. For a 20 MHz device, this means that a 5 MHz clock is the highest we can go. Not sure what we would need 30 MHz...

mcuee, 2022-06-08:
> I don't own a 2232-based programmer, but a rule of thumb is to not use an ISP clock faster than 1/4 of the microcontroller clock speed. For a 20 MHz device, this means that a 5 MHz clock is the highest we can go. Not sure what we would need 30 MHz...

In that case, probably this can be closed.

MCUdude, 2022-06-08:
I haven't really read the entire thread, but what's the highest clock speed currently supported? And can the 2232 be used to program xmegas too?

mcuee, 2022-06-08:
BTW, the patch in #474 fixed my issue with reading/writing of ATmega2560. JTAGKey-2 is at least quite a bit faster than AVR Dragon when reading the ATmega2560, although not as fast at the wiring bootloader.

Ref:
* https://github.com/avrdudes/avrdude/issues/474#issuecomment-1149951393

mcuee, 2022-06-19:
Close this for now. Please re-open if there is a need later.

closed 2022-06-19 by mcuee