avrdudes/avrdude #671: [patch #8130] Start programming/verifying address for usbasp programmer

avrdude-admin, 2021-12-13:
Fri 26 Jul 2013 01:49:10 PM UTC

This patch add global support for start address of programming/verifying regions and usage of this feature for usbasp programmer.

[file #28661: avrdude_start_address.patch](https://savannah.nongnu.org/patch/download.php?file_id=28661)
[file #28662: avrdude_start_address.patch](https://savannah.nongnu.org/patch/download.php?file_id=28662)

This issue was migrated from https://savannah.nongnu.org/patch/?8130

avrdude-admin, 2021-12-13:
Kostya V. Ivanov <warmcat>
Fri 26 Jul 2013 02:05:01 PM UTC

Sorry! Previous patch file is not clean...

(**[file #28662](https://savannah.nongnu.org/support/download.php?file_id=28662)**)

avrdude-admin, 2021-12-13:
Joerg Wunsch <joerg_wunsch>
Mon 02 Sep 2013 09:15:50 PM UTC

Despite some minor things I don't agree (like initializing
an unsigned int to 0xffffffff, rather than UINT_MAX), I'm
not quite sure how this is actually meant to work.

I'm afraid the more general feature that has been added some
time ago where AVRDUDE tracks all the initialized data from
input files would have made this patch obsolete.

Do you agree with this?

mcuee, 2022-06-10:
@dl8dtl and @MCUdude 
Since the OP has never come back for so many years, I think this can be closed.

closed 2022-06-10 by MCUdude