avrdudes/avrdude #672: [patch #8176] butterfly.c (AVR109 protocol implementation) clean-up and bug-fixing

avrdude-admin, 2021-12-13:
Sun 08 Sep 2013 09:08:22 PM UTC

While writing my own AVR109 compliant (SRAM) buffered bootloader I needed the attached butterfly.c changes.

Three bug fixes:

1. Word flash addresses (already fixed in master).

2. Absent butterfly_page_erase()
Seg fault when trying to program the eeprom only.

3. Terminal mode uses butterfly_read_byte_flash() to dump flash content. Original got byte order wrong.

Clean-up:

1. Skips address setting if the bootloader auto increments the address.

2. Allows multi-page load/write (if just the calling avr.c would take advantage of!).

For your perusal. Thanks, Enoch.

[file #29087: patch](https://savannah.nongnu.org/patch/download.php?file_id=29087)

This issue was migrated from https://savannah.nongnu.org/patch/?8176

avrdude-admin, 2021-12-13:
Joerg Wunsch <joerg_wunsch>
Fri 13 Sep 2013 07:40:22 PM UTC

Bug fixes #1 and #3 are (in my opinion) already present.

I adopted your fix for **[bug #2](https://savannah.nongnu.org/bugs/?2)**.

I'm not yet convinced about the cleanups (as partially already
mentioned).

mcuee, 2022-07-08:
In this case, I think this issue can be closed as the bug fixes 1/2/3 are already incorporated and the cleanups are not followed up for so many years.

closed 2022-07-08 by mcuee