avrdudes/avrdude #673: [patch #8214] Patch for Devantech USB-ISS bit-banging support

avrdude-admin, 2021-12-13:
Sun 20 Oct 2013 10:43:18 AM UTC

This patch implements support for the Devantech USB-ISS device in bit-banging mode. The USB-ISS is a small SPI/I2C/Digital IO/Analog in device for the USB port.

The SPI mode of this module currently (firmware version 6) does not make it possible to control the #CE line for AVR-RESET. Devantech/Robot electronics has been approached, but they will not implement a second mode of SPI-operation with manually controlled #CE, and they will not provide the firmware source...

So bit-banging it is, unfortunately.

[file #29425: avrdude-6.0.1_usbiss.patch](https://savannah.nongnu.org/patch/download.php?file_id=29425)

This issue was migrated from https://savannah.nongnu.org/patch/?8214

mcuee, 2022-06-06:
I am not so sure if this is till work pursuing even though this USB-ISS module is still widely available.

mcuee, 2022-06-25:
I will close this for now as it is rather old and there are many cheap programmers available now (eg: usbasp), in case people want to revive the patch, please reopen. Thanks.

closed 2022-06-25 by mcuee