avrdudes/avrdude #676: [patch #8272] rfcomm devices can't connect in time to upload

avrdude-admin, 2021-12-13:
Sun 12 Jan 2014 05:17:00 AM UTC

When using bluetooth serial devices for uploading, and using `rfcomm bind` to connect those devices when the /dev/rfcomm0 file is opened, avrdude doesn't allow enough time for the device to connect before failing.

If avrdude reads first, then the device has time to connect, but it seems that avrdude performs a ser_send first.  This patch adds 10 retries when ENOTCONN (107) "Transport endpoint is not connected" is returned as the errno for the send

[file #30248: rfcomm_wont_upload.patch](https://savannah.nongnu.org/patch/download.php?file_id=30248)

This issue was migrated from https://savannah.nongnu.org/patch/?8272

avrdude-admin, 2021-12-13:
Rob Riggs <rriggs>
Sun 23 Mar 2014 04:48:07 PM UTC

The problem is with the Linux Bluetooth stack and not with avrdude.  Pyserial has the same problem.  The root cause is that RFCOMM sends ENOTCONN for non-blocking connections. POSIX requires either EAGAIN or EWOULDBLOCK if the connection may eventually succeed.  ENOTCONN indicates a permanent failure.

See also [https://bugzilla.redhat.com/show_bug.cgi?id=1060457](https://savannah.nongnu.orghttps://bugzilla.redhat.com/show_bug.cgi?id=1060457) and [http://comments.gmane.org/gmane.linux.bluez.kernel/42303](https://savannah.nongnu.orghttp://comments.gmane.org/gmane.linux.bluez.kernel/42303)

The fix is to use a Linux kernel with a non-broken Bluez stack.

avrdude-admin, 2021-12-13:
Joerg Wunsch <joerg_wunsch>
Sat 27 Nov 2021 09:43:16 PM UTC

The audit trail refers to a problem inside Linux.

Is that still an issue these days?

mcuee, 2022-06-22:
I will close this issue as it seems to be an old Linux kernel issue. Please re-open if this pops up again.

closed 2022-06-22 by mcuee