avrdudes/avrdude #678: [patch #8380] adds 500k 1M 2M baud to ser_posix.c

avrdude-admin, 2021-12-13:
Thu 06 Mar 2014 10:34:36 AM UTC

Hi,

This patch add three faster baud rates for posix systems.
500k, 1M, and 2M are the missing "standard" baud rates supported by most newer AVRs.
Please consider to apply this patch to the trunk version of avrdude so it can be part of the next release.

Regards
Georg von Zengen

[file #30791: baudrate.patch](https://savannah.nongnu.org/patch/download.php?file_id=30791)
[file #30874: custom_baud.patch](https://savannah.nongnu.org/patch/download.php?file_id=30874)
[file #30875: custom_baud_linux_only.patch](https://savannah.nongnu.org/patch/download.php?file_id=30875)
[file #30876: custom_baud_linux_only_under.patch](https://savannah.nongnu.org/patch/download.php?file_id=30876)

This issue was migrated from https://savannah.nongnu.org/patch/?8380

avrdude-admin, 2021-12-13:
Joerg Wunsch <joerg_wunsch>
Thu 06 Mar 2014 10:49:20 AM UTC

Curious, what's the point of adding more baudrates?

The code is supposed to allow arbitrary baudrates already as
it is.  The historical Bxxxx constants are mainly there to
support historic code; on all modern systems, Bxxxx is just
the numerical value xxxx anyway.  (The Single Unix Specification
only allows for the use of Bxxxx constants as the portable
method; but then, baud rates beyond 38400 Bd are inherently
non-portable anyway.)

avrdude-admin, 2021-12-13:
Joerg Wunsch <joerg_wunsch>
Thu 06 Mar 2014 12:26:18 PM UTC

After some more review, it appears that (at least some versions
of) Linux don't follow the idea of using Bxxxx == xxxx, so I'm
probably going to accept your patch.

avrdude-admin, 2021-12-13:
Georg von Zengen <oni>
Thu 06 Mar 2014 10:19:54 PM UTC

I also had some more research about this.
Here my results:
-no linux version (had a look at debian derivatives, fedora, arch)     follows the idea of Bxxxx = xxxx
-freebsd, openbsd, and os x follow this idea until B38400, everything above is handled differently in each of the three as far as I saw.
-What did not found was a system where -b 1000000 is supposed to work correctly

During the test for my last comment the baud rate was accidentally  set to 1000000 by a previous test with my patched version of avrdude.

So I think for Linux we are fine with this patch.
For OS X I may have a look tomorrow.
And for XXXXbsd, I do not know anyone using it on the development machine so I have no chance to test it.

avrdude-admin, 2021-12-13:
Joerg Wunsch <joerg_wunsch>
Fri 07 Mar 2014 07:18:04 AM UTC

As an additional note, the calls to cfsetispeed() /
cfsetospeed() should really check their return values, so
attempts to set baud rates which are not supported by the
underlying system are caught.

avrdude-admin, 2021-12-13:
Georg von Zengen <oni>
Wed 12 Mar 2014 10:22:31 AM UTC

The last days I put a bit more work into this and think I did some progress.
The custom_baud.patch adds not only the three rate mentioned in the topic. It enables avrdude to set "every" custom baud rate.
It is "every" because of kernel/hardware limitations.

The patch supports mac os x and linuxs.
For mac os x 921600 is the maximum baud rate, it seem to be a kernel/hardware limitation.
For linux I tested it up to 2M baud on different debian and arch systems and a fedora virtualbox.

For all not os x or linux systems the patch does not change anything, here the standard posix way is used.
This is mostly because of my missing knowledge of BSDs.

The Patch also fixes the issue of checking the return values of cfsetospeed() and  cfsetispeed() you mentioned before.

Custom baud rates are useful for avrdude because it makes programming this serial bootloaders faster even if the maximum baud rate of the avr is kind of odd.

So this patch makes baudrate.patch obsolete.

(**[file #30874](https://savannah.nongnu.org/support/download.php?file_id=30874)**)

avrdude-admin, 2021-12-13:
Joerg Wunsch <joerg_wunsch>
Wed 12 Mar 2014 11:23:57 AM UTC

Thank you for the patch!

Some remarks:

. I don't think you have to special-case OSX in any way.  It does
already work the way it is now, just the same like *BSD (which it is
derived from in that respect); the Bxxxx names are just aliases for
number xxxx there, so cfsetispeed() and cfsetospeed() can be fed the
raw baudrate numbers as well as the Bxxxx defines.

. Your patch for Linux is probably fine, though I cannot test it
(lacking any device with a non-standard baudrate).

. Please don't add unrelated changes to a patchfile (indentation
changes in line 96 ff.).

avrdude-admin, 2021-12-13:
Georg von Zengen <oni>
Wed 12 Mar 2014 12:54:39 PM UTC

And one more patch file.

- removed OSX stuff
- linux still works (tested)
- hopefully removed all unrelated changes

(**[file #30875](https://savannah.nongnu.org/support/download.php?file_id=30875)**)

avrdude-admin, 2021-12-13:
Joerg Wunsch <joerg_wunsch>
Wed 12 Mar 2014 12:58:54 PM UTC

Thanks again, looks good!

Curious, would it also work with #ifdef **_linux_**?

I think the macro without a leading underscore won't be defined
if you put the compiler into strict standards compliance mode
(like, -std=c99) since this macro collides with the application
namespace.

avrdude-admin, 2021-12-13:
Georg von Zengen <oni>
Wed 12 Mar 2014 02:44:04 PM UTC

Ok just to clarify it.
Are you ok with the last patch or should I change anything else?

avrdude-admin, 2021-12-13:
Joerg Wunsch <joerg_wunsch>
Tue 25 Nov 2014 09:33:38 PM UTC

Thanks, applied in SVN r1351.

avrdude-admin, 2021-12-13:
Joerg Wunsch <joerg_wunsch>
Tue 09 Feb 2016 03:23:38 PM UTC

Reopening this patch, as the fix has been causing too much
collateral damange, and thus been backed out again.

See **[bug #46610](https://savannah.nongnu.org/bugs/?46610)**.

Patch needs a major overhaul before it can be accepted again.

MCUdude, 2022-03-20:
I think we can close this issue. Custom baud rates has been supported on all platform except for MacOS, and with PR #898, custom baud rates now work on all platforms.

closed 2022-03-20 by MCUdude