avrdudes/avrdude #686: [patch #8463] typo in rev 1305

avrdude-admin, 2021-12-13:
Sun 18 May 2014 09:58:53 AM UTC

Hi,

in rev 1305 there is a typo which prevents avrdude from compiling.
It is in avrftdi.c.
Please check before commiting!

I appended a patch, please apply it.

regards
Georg

[file #31398: typo.patch](https://savannah.nongnu.org/patch/download.php?file_id=31398)

This issue was migrated from https://savannah.nongnu.org/patch/?8463

avrdude-admin, 2021-12-13:
Axel Wachtler <awachtler>
Sun 18 May 2014 11:33:00 AM UTC

Patch is applied rev 1306, thanks for reporting.

avrdude-admin, 2021-12-13:
Rene Liebscher <rliebscher>
Wed 09 Jul 2014 07:25:40 AM UTC

This was not only a typo, at this place there was a vprintf before so it should not replaced by avrdude_message which behaves like printf.

I get an segmentation fault when use for example 'avrdude -cavrftdi -pm8' without a connected programmer, as avrftdi_open fails and it tries to output the error message using this function.

MCUdude, 2022-01-07:
This issue can probably be closed, because `vfprintf(stderr, ...)` solved the issue for now.

closed 2022-01-08 by dl8dtl