avrdudes/avrdude #692: [patch #8717] pattch for mcprog and libhidapi support

avrdude-admin, 2021-12-13:
Sun 16 Aug 2015 01:36:18 PM UTC

This implements support for our open hardware "mcprog" interface. EAGLE files available at:
 [http://rhodecode.bytewerk.org/Mikrocontroller/mcprog](https://savannah.nongnu.orghttp://rhodecode.bytewerk.org/Mikrocontroller/mcprog)

mcprog uses the MCP2210 USB-to-SPI interface.
This code was successfully tested on Linux and MAC.

The patch started as a copy of the pickit2 code already shipped with avrdude, but as the libusb backend seems to be dysfunctional on MAC for quite some time, I replaced libusb by libhidapi.

Therefore, this patch also brings support for libhidapi to avrdude. It's probably a good idea to use libhidapi instead of libusb for other programmers, that use the USB HID protocol.

[file #34656: mcprog_libhidapi.patch](https://savannah.nongnu.org/patch/download.php?file_id=34656)

This issue was migrated from https://savannah.nongnu.org/patch/?8717

mcuee, 2022-06-22:
Reference here: https://wiki.bytewerk.org/view/MCProg

Patch file is also attached here for reference.

[mcprog_libhidapi.patch.txt](https://github.com/avrdudes/avrdude/files/8957096/mcprog_libhidapi.patch.txt)

mcuee, 2022-06-22:
I will still close this issue as the OP has not come back for many years and the paticular programmer does not seem to be popular. Please re-open if someone wants to further work on this topic.

closed 2022-06-22 by mcuee