avrdudes/avrdude #695: [patch #8790] Add support for Visual Studio Builds

avrdude-admin, 2021-12-13:
Fri 13 Nov 2015 03:43:41 PM UTC

Per our previous discussion, this is the first patch submission that includes just the Visual Studio build support.

[file #35448: windows-vs.zip](https://savannah.nongnu.org/patch/download.php?file_id=35448)
[file #35447: diff](https://savannah.nongnu.org/patch/download.php?file_id=35447)

This issue was migrated from https://savannah.nongnu.org/patch/?8790

avrdude-admin, 2021-12-13:
Joerg Wunsch <joerg_wunsch>
Thu 19 Nov 2015 09:59:18 PM UTC

OK, the new diff (**[patch #8796](https://savannah.nongnu.org/patch/?8796)**) is fine except for #ifdefing out the
linuxgpio programmer.  This programmer's implementation only compiles
into real code if HAVE_LINUXGPIO is set anyway, otherwise it simply
issues a message when trying to access it.  Without that, FreeBSD or
Solaris would not be able to compile either.

What I don't understand is why you #ifdef the calls to setvbuf() away.
This function has been in the C standard for more than 25 years.  If
this is disabled, user experience will be poor in some situations for
MSC users.

Is there really no Windows counterpart for isatty()?  It is used to
find out whether the program is called interactively (for Windows,
this means started on a cmd.exe commandline).

If you don't have a way to handle inb() and outb() direct parallel
port IO (ppiwin.c), then all those historic parallel-port attached
programming dongles simply cannot be supported.  Instead of adding
fake early return values in inb() and outb() for the MSC compilation,
it makes more sense to wrap that entire file in #if HAVE_PARPORT, and
set HAVE_PARPORT to 0 for the MSC compilation (same as in ppi.c for
the Unix counterpart).

The new files getopt.c and getopt.h need a suitable copyright.  Please
suggest what should be added there.  Please don't use Brian Dean's or
my name there :), use your own name or whoever wrote these files.
Also, these files should be protected by #ifdef **MSC** so they can
become a normal sourcefile as anything else.  (That's important for
the source code distribution.)

I think that entire autoconf replacement (prebuild.cmd) nees a little
more thought.  What autoconf/automake basically does is:

1) generate ac_cfg.h based on the autodetected features
2) generate the Makefiles
3) generate avrdude.conf from avrdude.conf.in

As for #1, you supply your own ac_cfg.h file, but it would be
clobbered by anyone running the real configure script.  Instead, I
suggest giving that file a different name (e.g. ac_cfg_msc.h), and
copy it over to ac_cfg.h in prebuild.cmd.

#2, you don't need, as you've got your own project files.

As for #3, it's about replacing @DEFAULT_SER_PORT@ and
@DEFAULT_PAR_PORT@ by sensible values, and kicking out everything
between @HAVE_PARPORT_BEGIN@ and @HAVE_PARPORT_END@.  prebuild.cmd
could perform something similar using sed (configure.ac also uses sed
for that purpose).

If prebuild.cmd can perform these actions, it might serve as a
complete replacement for the Unix-style configure script.

yegorich, 2022-01-09:
@mariusgreuel this issue can be closed as we now officially support Visual Studio

mariusgreuel, 2022-01-09:
Included in v7.0

closed 2022-01-09 by mariusgreuel