avrdudes/avrdude #696: [patch #8796] Changes to support building AVRDUDE using VS on Windows

avrdude-admin, 2021-12-13:
Tue 17 Nov 2015 04:31:17 PM UTC

[file #35480: windows.vs.zip](https://savannah.nongnu.org/patch/download.php?file_id=35480)
[file #35479: diff](https://savannah.nongnu.org/patch/download.php?file_id=35479)

This issue was migrated from https://savannah.nongnu.org/patch/?8796

avrdude-admin, 2021-12-13:
Joerg Wunsch <joerg_wunsch>
Thu 19 Nov 2015 09:27:07 PM UTC

Well, these files belong to **[patch #8790](https://savannah.nongnu.org/patch/?8790)** rather than here.

MCUdude, 2022-04-04:
Thanks to @mariusgreuel it's now possible to build Avrdude on Windows using Visual Studio.

https://github.com/avrdudes/avrdude/wiki/Building-AVRDUDE-for-Windows-using-Visual-Studio

closed 2022-04-04 by MCUdude