avrdudes/avrdude #704: [patch #8954] Enable usage of libusb-1.0 for stk500v2 (aka avrispmkii)

avrdude-admin, 2021-12-13:
Thu 24 Mar 2016 12:32:46 PM UTC

I tried using an AVRISPmkII in Windows 10 with avrdude 6.3, but libusb couldn't find the device ("avrdude.exe: usbdev_open(): did not find any USB device "usb" (0x03eb:0x2104)").

Since libusb 0.1 is marked deprecated in cygwin, I installed libusb-1.0 and had a look at usbasp.c, where v1.0 is already in use. I then added support for libusb-1.0 in usb_libusb.c, and expanded the corresponding preprocessor conditions in jtag3.c, jtagmkII.c and stk500v2.c.

After replacing the ATMEL driver of the AVRISP by the WinUSB driver using Zadig avrdude is once more able to flash my AVRs using AVRISPmkII.

I also successfully applied and tested my patch with said AVRISPmkII in a virtual machine running xubuntu 14.04.

[file #36745: libusb.patch](https://savannah.nongnu.org/patch/download.php?file_id=36745)

This issue was migrated from https://savannah.nongnu.org/patch/?8954

avrdude-admin, 2021-12-13:
Joerg Wunsch <joerg_wunsch>
Sun 21 Jan 2018 10:21:05 PM UTC

This does unfortunately not compile under FreeBSD where the OS
offers a libusb-1.0 API compatible USB library:

```
usb_libusb.c:265:17: warning: implicit declaration of function 'libusb_has_capability' is invalid in C99
[-Wimplicit-function-declaration]
if (libusb_has_capability(LIBUSB_CAP_SUPPORTS_DETACH_KERNEL_DRIVER))
^
usb_libusb.c:265:39: error: use of undeclared identifier 'LIBUSB_CAP_SUPPORTS_DETACH_KERNEL_DRIVER'
if (libusb_has_capability(LIBUSB_CAP_SUPPORTS_DETACH_KERNEL_DRIVER))
^
usb_libusb.c:723:62: warning: passing 'char *' to parameter of type 'uint8_t *' (aka 'unsigned char *') converts between
pointers to integer types with different sign [-Wpointer-sign]
errcode = libusb_interrupt_transfer(udev, fd->usb.wep, (char *)bp, tx_size, &rv, 10000);
^~~~~~~~~~
/usr/include/libusb.h:549:87: note: passing argument to parameter 'data' here
int     libusb_interrupt_transfer(libusb_device_handle * devh, uint8_t endpoint, uint8_t *data, int length, int *trans...
^
usb_libusb.c:725:57: warning: passing 'char *' to parameter of type 'uint8_t *' (aka 'unsigned char *') converts between
pointers to integer types with different sign [-Wpointer-sign]
errcode = libusb_bulk_transfer(udev, fd->usb.wep, (char *)bp, tx_size, &rv, 10000);
^~~~~~~~~~
/usr/include/libusb.h:548:82: note: passing argument to parameter 'data' here
int     libusb_bulk_transfer(libusb_device_handle * devh, uint8_t endpoint, uint8_t *data, int length, int *transferre...
^
usb_libusb.c:792:51: warning: passing 'char [512]' to parameter of type 'uint8_t *' (aka 'unsigned char *') converts between
pointers to integer types with different sign [-Wpointer-sign]
errcode = libusb_interrupt_transfer(udev, ep, usbbuf, maxsize, &rv, 10000);
^~~~~~
/usr/include/libusb.h:549:87: note: passing argument to parameter 'data' here
int     libusb_interrupt_transfer(libusb_device_handle * devh, uint8_t endpoint, uint8_t *data, int length, int *trans...
^
usb_libusb.c:794:46: warning: passing 'char [512]' to parameter of type 'uint8_t *' (aka 'unsigned char *') converts between
pointers to integer types with different sign [-Wpointer-sign]
errcode = libusb_bulk_transfer(udev, ep, usbbuf, maxsize, &rv, 10000);
^~~~~~
/usr/include/libusb.h:548:82: note: passing argument to parameter 'data' here
int     libusb_bulk_transfer(libusb_device_handle * devh, uint8_t endpoint, uint8_t *data, int length, int *transferre...
^
usb_libusb.c:897:57: warning: passing 'char [512]' to parameter of type 'uint8_t *' (aka 'unsigned char *') converts between
pointers to integer types with different sign [-Wpointer-sign]
errcode = libusb_bulk_transfer(udev, fd->usb.eep, usbbuf,
^~~~~~
/usr/include/libusb.h:548:82: note: passing argument to parameter 'data' here
int     libusb_bulk_transfer(libusb_device_handle * devh, uint8_t endpoint, uint8_t *data, int length, int *transferre...
^
usb_libusb.c:923:64: warning: passing 'char [512]' to parameter of type 'uint8_t *' (aka 'unsigned char *') converts between
pointers to integer types with different sign [-Wpointer-sign]
errcode = libusb_interrupt_transfer(udev, fd->usb.rep, usbbuf,
^~~~~~
/usr/include/libusb.h:549:87: note: passing argument to parameter 'data' here
int     libusb_interrupt_transfer(libusb_device_handle * devh, uint8_t endpoint, uint8_t *data, int length, int *trans...
^
usb_libusb.c:926:59: warning: passing 'char [512]' to parameter of type 'uint8_t *' (aka 'unsigned char *') converts between
pointers to integer types with different sign [-Wpointer-sign]
errcode = libusb_bulk_transfer(udev, fd->usb.rep, usbbuf,
^~~~~~
/usr/include/libusb.h:548:82: note: passing argument to parameter 'data' here
int     libusb_bulk_transfer(libusb_device_handle * devh, uint8_t endpoint, uint8_t *data, int length, int *transferre...
^

```
So it seems the libusb_has_capability() needs an explicit
probing in configure.ac as it has not always been part of
the libusb-1.0 API.  (Unfortunately, LIBUSB_CAP_HAS_CAPABILITY
is part of an enum rather than a #define, so it cannot be
tested at compile-time.  Seems someone didn't really think
about that detail before.)

The warnings about char vs. uint8_t * can probably safely
be silenced using typecasts.

mcuee, 2022-05-22:
I believe this issue has been fixed in later version of FreeBSD.
https://svnweb.freebsd.org/base?view=revision&revision=362224

MCUdude, 2022-05-22:
@dl8dtl can you confirm?

dl8dtl, 2022-05-23:
Well, it's supposed to be in the latest 11.x release (11.4, 11.x is discontinued), and since 12.2 in the 12.x release line (and then all ore recent ones).

So yes, with any supported FreeBSD version, it compiles now. It causes a bunch of warnings though:

```
[  6%] Building C object src/CMakeFiles/libavrdude.dir/usb_libusb.c.o
/home/joerg/src/avrdude-official/src/usb_libusb.c:723:62: warning: passing 'char *' to parameter of type 'uint8_t *' (aka 'unsigned char *') converts between pointers to integer types where one is of the unique plain 'char' type and the other is not [-Wpointer-sign]
      errcode = libusb_interrupt_transfer(udev, fd->usb.wep, (char *)bp, tx_size, &rv, 10000);
                                                             ^~~~~~~~~~
/usr/include/libusb.h:569:87: note: passing argument to parameter 'data' here
int     libusb_interrupt_transfer(libusb_device_handle * devh, uint8_t endpoint, uint8_t *data, int length, int *transferred, uint32_t timeout);
                                                                                          ^
```
In the quoted case, there's even a `(char *)` typecast (which is completely wrong here), in some other cases, there's a warning without the cast as the parameter passed is obviously of type `char[]`.

Just reviewed the header file at `https://github.com/libusb/libusb/blob/master/libusb/libusb.h`, they also use `unsigned char *` for these functions.

Thus, the codes needs a bit of cleanup at first, and thorough testing against all affected devices (AVRISPmkII, JTAGICEmkII, JTAGICE3, AtmelICE, STK600, AVR Dragon – did I forget any?) afterwards, as it changes the underlying transport library for all these devices.

mcuee, 2022-05-24:
> Thus, the code needs a bit of cleanup at first, and thorough testing against all affected devices (AVRISPmkII, JTAGICEmkII, JTAGICE3, AtmelICE, STK600, AVR Dragon – did I forget any?) afterwards, as it changes the underlying transport library for all these devices.

I have the AVR Dragon so I can help out testing.

Related topic: https://github.com/avrdudes/avrdude/issues/974

mcuee, 2023-04-09:
Just post the patch code here for easier view.
```
Index: avrdude/jtag3.c
===================================================================
--- avrdude/jtag3.c	(revision 1387)
+++ avrdude/jtag3.c	(working copy)
@@ -1314,7 +1314,7 @@
   LNODEID usbpid;
   int rv = -1;
 
-#if !defined(HAVE_LIBUSB) && !defined(HAVE_LIBHIDAPI)
+#if !defined(HAVE_LIBUSB) && !defined(HAVE_LIBUSB_1_0) && !defined(HAVE_LIBHIDAPI)
   avrdude_message(MSG_INFO, "avrdude was compiled without USB or HIDAPI support.\n");
   return -1;
 #endif
@@ -1353,7 +1353,7 @@
   }
   if (rv < 0) {
 #endif	/* HAVE_LIBHIDAPI */
-#if defined(HAVE_LIBUSB)
+#if defined(HAVE_LIBUSB) || defined(HAVE_LIBUSB_1_0)
     serdev = &usb_serdev_frame;
     for (usbpid = lfirst(pgm->usbpid); rv < 0 && usbpid != NULL; usbpid = lnext(usbpid)) {
       pinfo.usbinfo.flags = PINFO_FL_SILENT;
@@ -1366,7 +1366,7 @@
       strcpy(pgm->port, port);
       rv = serial_open(port, pinfo, &pgm->fd);
     }
-#endif	/* HAVE_LIBUSB */
+#endif	/* HAVE_LIBUSB || HAVE_LIBUSB_1_0 */
 #if defined(HAVE_LIBHIDAPI)
   }
 #endif
Index: avrdude/jtagmkII.c
===================================================================
--- avrdude/jtagmkII.c	(revision 1387)
+++ avrdude/jtagmkII.c	(working copy)
@@ -1502,7 +1502,7 @@
    * search for.
    */
   if (strncmp(port, "usb", 3) == 0) {
-#if defined(HAVE_LIBUSB)
+#if defined(HAVE_LIBUSB) || defined(HAVE_LIBUSB_1_0)
     serdev = &usb_serdev;
     pinfo.usbinfo.vid = USB_VENDOR_ATMEL;
     pinfo.usbinfo.flags = 0;
@@ -1554,7 +1554,7 @@
    * search for.
    */
   if (strncmp(port, "usb", 3) == 0) {
-#if defined(HAVE_LIBUSB)
+#if defined(HAVE_LIBUSB) || defined(HAVE_LIBUSB_1_0)
     serdev = &usb_serdev;
     pinfo.usbinfo.vid = USB_VENDOR_ATMEL;
     pinfo.usbinfo.flags = 0;
@@ -1606,7 +1606,7 @@
    * search for.
    */
   if (strncmp(port, "usb", 3) == 0) {
-#if defined(HAVE_LIBUSB)
+#if defined(HAVE_LIBUSB) || defined(HAVE_LIBUSB_1_0)
     serdev = &usb_serdev;
     pinfo.usbinfo.vid = USB_VENDOR_ATMEL;
     pinfo.usbinfo.flags = 0;
@@ -1659,7 +1659,7 @@
    * search for.
    */
   if (strncmp(port, "usb", 3) == 0) {
-#if defined(HAVE_LIBUSB)
+#if defined(HAVE_LIBUSB) || defined(HAVE_LIBUSB_1_0)
[... 655 lines cut ...]
@@ -515,7 +946,7 @@
 		    usb_strerror());
 	  return -1;
 	}
-
+#endif  /* USE_LIBUSB_1_0 */
       if (rv <= nbytes)
 	{
 	  memcpy (buf, usbbuf, rv);
@@ -534,7 +965,7 @@
 /*
  this ends when the buffer is completly filled (nbytes=0) or was too small (nbytes< 0)
  or a short packet is found.
- however we cannot say for nbytes=0 that there was really a packet completed, 
+ however we cannot say for nbytes=0 that there was really a packet completed,
  we had to check the last rv value than for a short packet,
  but what happens if the packet does not end with a short packet?
  and what if the buffer is filled without the packet was completed?
@@ -612,4 +1043,4 @@
   .flags = SERDEV_FL_NONE,
 };
 
-#endif  /* HAVE_LIBUSB */
+#endif  /* HAVE_LIBUSB || HAVE_LIBUSB_1_0 */
```

mcuee, 2023-04-09:
From the direction given by @dl8dtl in the following issue, avrdude will probably continue using libusb-0.1 API in the foreseable future. The main advantage of libusb-1.0 vs libusb-0.1 API is the support of asynchronous API. But it may not bring too much speed improvement for avrdude. And then the above patch does not use asynchronous API at all.

* https://github.com/avrdudes/avrdude/issues/974
* https://github.com/avrdudes/avrdude/issues/906#issuecomment-1120433920

mcuee, 2023-06-02:
I will close this one for now.

closed 2023-06-02 by mcuee