avrdudes/avrdude #706: [patch #8967] Change ft2xx library to ftdi_syncbb instead of libusb

avrdude-admin, 2021-12-13:
Mon 04 Apr 2016 12:15:05 PM UTC

FT2XX is more stable and drivers for windows is signed.
This patch changes libusb to ft2xx to ftdi_syncbb (FT245R and FT232R).

To compile you need:
1. patch sources
2. execute aclocal
3. autoconf
4. update config.guess to the newest
5. configure && make

[file #36833: avrdude-6.3-diff.tar.gz](https://savannah.nongnu.org/patch/download.php?file_id=36833)

This issue was migrated from https://savannah.nongnu.org/patch/?8967

avrdude-admin, 2021-12-13:
Joerg Wunsch <joerg_wunsch>
Mon 04 Apr 2016 08:59:27 PM UTC

Well, from a licensing point of view, it's somewhat questionable
whether we'd really like to introduce a dependency on a
closed-source library in code licensed under GPL.  Such
dependencies are always tolerable if the library qualifies
as a "system library" (otherwise, you could not use GPL
software on a closed-source OS), but ftd2xx would certainly
not qualify as such.

Second, there are a number of platforms AVRDUDE supports
where libftd2xx is not available at all (*BSD, Solaris).

I'd like to see the real benefit of such an implementation
before having that in the code.  However, it makes more sense
to discuss that on the mailing list rather than in a patch
tracker.  In particular, your submission suggests the
existing implementation were less stable, so we certainly
ought to fix that.  (OK, the signed driver for Windows is
accepted as a benefit.)

Finally, some stylistic issues:

- If at all, the use of ftd2xx must be completely optional,so people who cannot or don't want to install it don't lose
any functionality.  Replacing the existing ft245r.c
implementation is thus not acceptabel.

```
#elif defined(HAVE_LIBFTDI)&&(defined(HAVE_USB_H)||defined(HAVE_LUSB0_USB_H))

```
- Spaces and braces are welcome. ;-)- If you wrote the file, please add your name in the copyrightheader.

- In the comment in ftd2xx_initialize, there are non-ASCII,non-ISO8859-1, non-UTF-8 characters. ('\x93', '\x94')

mariusgreuel, 2022-01-18:
The new Windows MSVC version uses a modified libftdi library that works with ftd2xx.dll.

closed 2022-01-18 by mariusgreuel

mcuee, 2022-06-12:
> Well, from a licensing point of view, it's somewhat questionable
> whether we'd really like to introduce a dependency on a
> closed-source library in code licensed under GPL.  Such
> dependencies are [...]

@dl8dtl
So you are also holding a more strict view of the "system libary" definition. In this case, the current official Windows binary's usage of avrdude-libftdi is also a bit questionable. The good thing is that  avrdude-libftdi uses dynamic linking to D2XX DLL file, so it is probably still arguable.

Ref: avrdude-libftdi is basically a wrapper of D2XX.
https://github.com/avrdudes/libftdi
https://github.com/avrdudes/libftdi/issues/1