avrdudes/avrdude #714: [patch #9127] Support for CH341A programmer.

avrdude-admin, 2021-12-13:
Fri 30 Sep 2016 05:03:34 PM UTC

CH341A or CH341H SPI Flash/EEPROM programmer.
This code was successfully tested on Windows with libusb driver.

[file #38632: 0001-CH341A-Programmer.patch](https://savannah.nongnu.org/patch/download.php?file_id=38632)

This issue was migrated from https://savannah.nongnu.org/patch/?9127

avrdude-admin, 2021-12-13:
Max Gorb <maxxproff>
Sat 03 Dec 2016 05:15:34 AM UTC

Compiled at Raspbian-Jessie some warnings
[code]
gcc -DHAVE_CONFIG_H -I.  -DCONFIG_DIR=\"/usr/local/etc\"  -Wall -Wno-pointer-sign -g -O2 -MT libavrdude_a-ch341a.o -MD -MP -MF .deps/libavrdude_a-ch341a.Tpo -c -o libavrdude_a-ch341a.o `test -f 'ch341a.c' || echo './'`ch341a.c
ch341a.c: In function 'ch341a_spi_transfer':
ch341a.c:272:14: warning: unused variable 'bytestransferred' [-Wunused-variable]
int ret, bytestransferred;
^
ch341a.c:272:9: warning: unused variable 'ret' [-Wunused-variable]
int ret, bytestransferred;
^
ch341a.c: In function 'ch341a_initpgm':
ch341a.c:381:25: warning: assignment from incompatible pointer type
pgm->spi            = ch341a_spi;
^
ch341a.c: At top level:
ch341a.c:269:12: warning: 'ch341a_spi_transfer' defined but not used [-Wunused-function]
static int ch341a_spi_transfer(PROGRAMMER * pgm, const unsigned char **cmd, unsigned char**res) {
^
[/code]

Succesful maked it works with UsbAsp but not works with ch341a.

avrdude-admin, 2021-12-13:
Max Gorb <maxxproff>
Sat 03 Dec 2016 05:23:21 AM UTC

Incompatible pointer type was passed:
"...int size" instead "...unsigned int size" at
ch341a.c:113 and ch341a.c:298.
Unused-variables was commented but it not helps.

avrdude-admin, 2021-12-13:
Max Gorb <maxxproff>
Wed 14 Dec 2016 09:24:34 AM UTC

Sorry, I have tested it more and more, and it really works!
Patch needs some cleanup, but not serious.
BUT it works only with fast devices,
it needs LOW SCK option.
For example, it fail with the atmega8L default fuses (0xD9E1 internal osc)
but works fine with the same atmega8L overclocked to 12Mhz (0xC9FF external crystal).
I learn more for fix it...

avrdude-admin, 2021-12-13:
Alexey Sadkov <alx2000y>
Mon 19 Dec 2016 04:00:57 PM UTC

For LOW SCK option requires bitbang. It is not done because it was not required for me. I decided that the only FastSPI better than none.

avrdude-admin, 2021-12-13:
Joerg Wunsch <joerg_wunsch>
Thu 18 Jan 2018 10:32:35 PM UTC

I can understand that you are mainly interested in fast
programming - but if it fails to operate on AVRs the way
they are shipped (i.e., clocked with 1 MHz), it doesn't make
much sense to include it. You'd need something else then to
reprogram the fuses for higher clock, but if you've got
"something else", why not also use it for actual programming?

mcuee, 2022-05-23:
github repo here for reference:
https://github.com/Alx2000y/avrdude_ch341a

mcuee, 2022-06-05:
Some more info here.
https://github.com/maxxproff/avrdude-ch341

mcuee, 2022-06-25:
@maxxproff
Just wondering if you want to create a pull request against latest git head? Thanks.

mcuee, 2022-06-25:
@Alx2000y 
Just wondering if you want to address the questions raised by @dl8dtl and create a pull request against latest git head. Thanks.

mcuee, 2022-06-26:
@maxxproff
avrdude has moved to github, please comment here and not in https://savannah.nongnu.org/. Thanks.

maxxproff, 2022-06-27:
> Joerg Wunsch <joerg_wunsch> Thu 18 Jan 2018 10:32:35 PM UTC
>
> I can understand that you are mainly interested in fast programming - but if it fails to operate on AVRs the way they are shipped (i.e., [...]

@mcuee @dl8dtl 
CH341A has pre-build Fast-SPI commands, which have been sniffered and reverse-engineered by @setarcos and some other people.
For slow SPI it is need to use bitbang mode, all other people can't find examples of bitbang mode for CH341A (or someone found it, but doesn't want to do this work).
Due this fact all opensource software for CH341 can use fast SPI only. (this is answer the questions raised by @dl8dtl)
I found bitbang mode examples in CH341DPS.ZIP (it is source code of CH341DP.EXE - programmer utility for AVR from WCH).
I pasted it into @Alx2000y 's patch and add native win-drivers support.
I'm not programmer, I have done pre-built exe, brought it to some forums and shelve, pigeonhole, dismiss it.
 It was interest useful amusement, short-dated hobby.
My code is very bad and very dirty, but it works.
It was made in 2017, I don't remember all points of that work.
I'm too lazy and I don't have enough competence to clean up this code for normal pull request, maybe later... 
(2022-2017 = It took 5 years to copy my sources from my old RaspberyPi-programming-toy-PC to here. Cleaning may take a some more years yet...)
@mcuee, you can do it, if you want.
or maybe @Alx2000y 
(Алексей, привет! Есть желание запилить нормальный пул-реквест? С битбангом и кул-гёрлами)))) )
All points of bitbang code stored in [ch341a_bitbang.c](https://github.com/maxxproff/avrdude-ch341/blob/main/ch341a_bitbang.c)
It is for libusb driver.
.
For using wch win driver [ch341a_bitbang_wch.c](https://github.com/maxxproff/avrdude-ch341/blob/main/ch341a_bitbang_wch.c)
That's all, thank you for your attention.!

dl8dtl, 2022-06-28:
> С битбангом и кул-гёрлами

ROTFL – It took me quite a moment to grasp it. ;-)

mcuee, 2023-04-07:
Please take note #1215 has been merged.

mcuee, 2023-04-07:
I will close this issue for now and create another issue to address the remaining issue -- for example, to add bitbang SPI support.

#1215 only has fast SPI mode with single fixed SPI frequency) so that it will not support many raw AVR chips running on internal oscillator at about 1MHz to 1.6MHz.

mcuee, 2023-04-07:
We will continue the discussion in #1336.

closed 2023-04-07 by mcuee