avrdudes/avrdude #718: [patch #9254] Adds support for selecting USB devices to USBasp

avrdude-admin, 2021-12-13:
Thu 09 Feb 2017 04:49:47 PM UTC

The documentation for USBasp indicates that it supports -p usb:bus:dev
style USB device selection, however this is not actually supported. This
commit adds support for this in the same style as USBTiny.
If a dev is not included (e.g. -p usb:xxx), then it will print all the
devices that match one of the vendor and product IDs supported by the
USBasp protocol.

[file #39712: 0001-Adds-support-for-selecting-USB-devices-to-USBasp.patch](https://savannah.nongnu.org/patch/download.php?file_id=39712)
[file #39719: 0001-Adds-support-for-selecting-USB-devices-to-USBasp.patch](https://savannah.nongnu.org/patch/download.php?file_id=39719)

This issue was migrated from https://savannah.nongnu.org/patch/?9254

avrdude-admin, 2021-12-13:
Joerg Wunsch <joerg_wunsch>
Thu 18 Jan 2018 10:53:09 PM UTC

Somehow, it doesn't seem to work for me.

The USBasp can be found (no target connected, thus fails):

```
./avrdude -C avrdude.conf -p m16 -c usbasp -P usb

avrdude: error: usbasp_transmit: Unknown error
avrdude: error: usbasp_transmit: Unknown error
avrdude: warning: cannot set sck period. please check for usbasp firmware update.
avrdude: error: usbasp_transmit: Unknown error
avrdude: error: program enable: target doesn't answer. 1
avrdude: initialization failed, rc=-1
Double check connections and try again, or use -F to override
this check.

avrdude done.  Thank you.

```
However, when passing -P usb:xxx as suggested, I do not get
a list of devices but just:

```
./avrdude -C avrdude.conf -p m16 -c usbasp -P usb:xxx
avrdude: Error: Invalid -P value: 'usb:xxx'
Use -P usb:bus:device

avrdude done.  Thank you.

```
Nonsensical :bus:dev values are not rejected but cause a
stupid error message:

```
./avrdude -C avrdude.conf -p m16 -c usbasp -P usb:foo:bar
avrdude: error: could not find USB device with vid=0x16c0 pid=0x5dc vendor='www.fischl.de' product='USBasp'

avrdude done.  Thank you.

```
If there is no USBasp connected, it dumps core:

```
Program received signal SIGSEGV, Segmentation fault.
[Switching to Thread 802806400 (LWP 100762/avrdude)]
0x0000000801acc7a0 in strcasecmp_l () from /lib/libc.so.7
(gdb) up
#1  0x0000000000419bcd in usbasp_open (pgm=0x80286d000, port=<value optimized out>) at usbasp.c:604
604           if(strcasecmp(port, "nibobee") == 0) {
Current language:  auto; currently minimal

```

mcuee, 2022-06-10:
Relevant discussion:
* https://github.com/avrdudes/avrdude/issues/973

mcuee, 2022-06-22:
Since the patch did not work and the author has not come back for many years, I will close this one.

closed 2022-06-22 by mcuee