avrdudes/avrdude #720: [patch #9315] Add PRUSS programmer type for xmega pdi

avrdude-admin, 2021-12-13:
Tue 25 Apr 2017 07:42:28 AM UTC

This patch adds XMEGA PDI programming support using the Programmable Realtime Unit (PRUSS) available on AM335x based boards like Beaglebone Black. You can program an XMEGA in-circuit without requiring any external hardware, with only a few extra components in your circuit. Note that the programmer needs a custom firmware that needs to be loaded to the PRU (also attached here). Avrdude loads the firmware and communicates with it to be able to program the firmware on XMEGA. E.g.

avrdude -p atxmega16d4 -c pruss -e -U flash:w:my-firmware.hex

[file #40496: uio_pdi-data-1.bin](https://savannah.nongnu.org/patch/download.php?file_id=40496)
[file #40495: uio_pdi-text-1.bin](https://savannah.nongnu.org/patch/download.php?file_id=40495)
[file #40494: add-support-for-am335x-pruss-programmer.patch](https://savannah.nongnu.org/patch/download.php?file_id=40494)

This issue was migrated from https://savannah.nongnu.org/patch/?9315

avrdude-admin, 2021-12-13:
Joerg Wunsch <joerg_wunsch>
Tue 06 Jun 2017 12:11:39 PM UTC

Basically, a good idea.

Documentation patches are missing (either to the .texi or the
man page file, preferrably for both).

I see a couple of problems with the firmware files though:

First, we are guests on the FSF's servers, and they only accept
opensource projects.  Thus, we are not allowed to host a
closed-source firmware for the programmer.

Second, hardcoding a path name in the application is completely
a "don't ever do this".  This should preferrably be handled
using autoconf parameters, depending from --prefix.  However,
on Windows, this is usually completely different.  See
winconf.c for how this is implemented wrt. avrdude.conf.

avrdude-admin, 2021-12-13:
Enric Balletbo i Serra <eballetbo>
Tue 06 Jun 2017 01:00:47 PM UTC

Joerg, thanks for you answer.

Some comments below.

About the documentation missing I'll prepare a second version including the .texi and man page files. About hardcoding the path name I have another version of this patch that uses and extra parameter to specify the location of the firmware files, but seeing your answer might be not enough, I'll try to investigate how to handle this part with autoconf parameter and see how it works on Windows

Finally, the firmware is open [1] so that shouldn't be a problem, I guess

[1] [https://github.com/eballetbo/pdi-pruss](https://savannah.nongnu.orghttps://github.com/eballetbo/pdi-pruss)

mcuee, 2022-06-25:
@[eballetbo
Do you still want to persue this? If yes, please follow up as per the requests. If not, I will close this issue in one month time.

mcuee, 2022-06-25:
@eballetbo
Basically please create a pull request against latest git master and include the requests by @dl8dtl. Thanks.

eballetbo, 2022-06-27:
Hi @mcuee, sorry, I don't have the hardware to continue pursuing this so I don't expect work on this. I think you can close it. Thanks.

mcuee, 2022-06-27:
Thanks for the update.